MiczFlor/RPi-Jukebox-RFID #1: gpio buttons

ole-seelenmeyer, 2017-06-08:
Hey there,
maybe you can help me out. what could be the best way to use gpio buttons for forward and rewind tracks using the gpio buttons. i read a lot about it but i am not able to do it right now. maybe you can give me a hint.

hankman13, 2017-06-08:
Hi Ole,
maybe i can help you, i didn't solve it right now but i'm intressted in.
Whats your general problem about gpio?

ole-seelenmeyer, 2017-06-08:
its mostly a general question. how to control these jukeboxscript with gpio buttons. how to wireing the button is not the problem but how to implent it in the script. or write a second script and put it in the autostart by crontab?

hankman13, 2017-06-08:
I think a second script should work, if you only use it for forward or
backward your songs. The biggest issue could be to control VLC. I tried to
use the slave mode on VLC so that i can control the instance itself. But in
my setup i have an headless server without Desktop Environment so slave
mode did not work. So i switched to the mpv player.

MiczFlor, 2017-06-08:
The way this package works, using VLC, "next" and "previous" tracks are not an option. So the gpio buttons are not needed or doable. 
There is another, similar project where the author is using XMMS instead of VLC. The "Howto" is in German, but auto translation might do the trick:

[Kindgerechter Audioplayer mit dem Raspberry Pi , Part4](http://www.raspberry-pi-geek.de/Magazin/2014/03/Kindgerechter-Audioplayer-mit-dem-Raspberry-Pi/(offset)/4)

I will eventually close this "issue" because it's not really an issue. But will leave it open for now, because it looks like it will be helpful for others.

ole-seelenmeyer, 2017-06-08:
ok i just see the Jukebox project at adventures in raspberry pi that also uses vlc controlled by gpio buttons but without rfid support. my intention was to connect both projects. to have the rfid trigger to start a audiobook and have some buttons to skip trough the chapters or adjust volume. its hard for a python newby to understand.
thanx a lot.

hankman13, 2017-06-08:
In my Solution i used also rfid cards for skiping the Playlist and adjust
the Volume... i will Upload my work and clean up the docs in my fork in the
next days

ole-seelenmeyer, 2017-06-08:
Yes I know. That already works fine for me.but I want buttons cause it's
looks better and I can press stop whenever I want. I know my child's :)

Alwiz79, 2017-08-17:
Hi Ole, have you somehow come to point to combine these projects?
There are a lot jukebox projects out there. But they are all a bit different.
This is by far the best documented project. Anyhow, I'd also to combine the RFID tags with button controls for volume and next/previous track. I simply don't know how.

tullm, 2017-10-07:
hi.
Just built a prototype for my daughter thanks to your work and great instructions.

I also wanted to control vlc with next or pause and so on. Today I found a working solution.

Some little tweaking to the trigger script is needed.

First you need the definition of the command you want. Mapped to a RFID key on my side:
CMDPAUSE="000123456"

Then you need to add this to the other commands:
elif [ "$CARDID" == "$CMDPAUSE" ]
then
    # pause VLC media player
    echo "pause" | nc.openbsd -w 1 localhost 4212

Then you need to edit the cvlc call on the end of the script:
 (cvlc -I rc --rc-host localhost:4212 $PATHDATA/../playlists/$FOLDERNAME.m3u &)

So it starts vlc with the rc interface on localhost with port 4212. So vlc listens on this port for rc commands. The command we set (pause in my example) echoes pause per nc over the rc interface on the localhost port 4212 and quits after 1.

So we can use all other commands to control vlc.

Maybe that helps with the buttons too?!

Maybe so it is possible to add the standard functions to the Webinterface?!

Maybe also a check it the RFID used is already at play an do nothing?

Also it would be better if only musik files will be added to the playlist. Great would be the possibility to jump to a specific track in the playlist ?!

Thank you very much for your work an I hope this info might be useful for somebody..

Cheers

Tullm

MiczFlor, 2017-10-09:
@tullm thanks for the hint. This is what I was looking for. Included it in the script:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/45616b8161eac3a15639753fc883a619f8ce6132
I have not much time at the moment, but will also look into the other things you listed, like adding it to the web interface.

What do you mean with "Also it would be better if only musik files will be added to the playlist."?

All the best, micz

tullm, 2017-10-09:
No problem. You're welcome.

What I mean Ist that there are all files from a folder are put into playlist and are shown in the folder info in the Webinterface. For example are there folder.jpg in my folders. These are also in the playlist.
So I think it would be better  to filter for audio files only to put them into the playlist.

Hope that cleared things up.

Cheers

Geliras, 2017-11-16:
Is it possible to use buttons now?

Geliras, 2017-11-16:
@MiczFlor @tullm
Is it possible to use buttons now?

tullm, 2017-11-17:
Yes, should work as long as you can link them to the card in the script. I#m not aware of how to script this right.

Geliras, 2017-11-17:
Thank you. I will start buying this weekend and then we will see. If I get it working (I think so), I will write it here.

MiczFlor, 2017-11-17:
@tullm @Geliras I have not made any changes that would support the GPIO buttons. It would be great if you give this a try and report back. Many people contact me about this feature and - after all - this is on github, so it is open to suggestions, forks and different branches.
What I am working is a version where the RPI serves as a wifi access point. That way, the installation could be easier because rather than connecting the RPi to the local Wifi network, it will its own SSID that devices can connect to.

Geliras, 2017-11-17:
Controls are working with RFID, so it cannot be that hard to trigger the same command when pressing a button. Right? I never used own buttons on RPi yet.

MiczFlor, 2017-11-17:
Skipping to the next and previous track on VLC you can see in the bash script
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/rfid_trigger_play.sh.sample
on line 109:
` echo "next" | nc.openbsd -w 1 localhost 4212`
and line 113:
`echo` "prev" | nc.openbsd -w 1 localhost 4212`

The volume change is done with the command line in the bash script. So the command to trigger with volume buttons should be something like this:
Volume up: `amixer sset 'PCM' 5%+`
Volume down: `amixer sset 'PCM' 5%-`

Good luck and please report back :)

Geliras, 2017-11-22:
Have not recieved my buttons yet, so i cant try. But wouldn't it be possible to just send one this commands when pressing a button:

echo "play" | nc.openbsd -w 1 localhost 4212
echo "pause" | nc.openbsd -w 1 localhost 4212
echo "next" | nc.openbsd -w 1 localhost 4212
echo "prev" | nc.openbsd -w 1 localhost 4212

We could just edit one of these:

https://webcache.googleusercontent.com/search?q=cache:nKcEN5uWTqUJ:https://www.element14.com/community/docs/DOC-78055/l/adding-a-shutdown-button-to-the-raspberry-pi-b+&cd=1&hl=de&ct=clnk&gl=de

https://raspberrypi.stackexchange.com/a/13214

MiczFlor, 2017-12-04:
Hi everybody, I just got a mail that somebody has the Jukebox working with the buttons. Here is the thread in German (sorry, will try to incorporate a "howto" in English soon).

Let me know if this is of any help:
https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=312257#post312257

stefansiegl, 2017-12-08:
Hi @MiczFlor, this is a tremendously well-documented and easy to setup project. Thanks man, had this running in no time on my raspberry. What a good job.

Regarding the buttons I really think that this would improve this project even more. Having a similar documentation on how to attach the buttons, resistors, ... would just be perfect (especially for somebody like me with little to none practical tinkering experience) ;)

jackwidu, 2018-04-13:
Sorry, if I capture this topic. It seemed inappropiate for me to open a new topic for such a small problem:
I cannot change the volume over the GPIO Buttons. No problem skipping or pausing. I also switched the assigned buttons, so I'm sure the buttons are working.
When in SSH "amixer sset Speaker 20%+" will do what it is suppose to do, and in the
gpio-buttons.py it is:
"def def_volU():
    check_call("amixer sset Speaker 20%+", shell=True)"
(Speaker is my USB-Soundcard)

There is a thing I suspect, as I did something not according to the GPIO-buttons.md manual:
in the "/etc/rc.local" I had to add "python " before "/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py &"
because otherwise the buttons wouldn't work at all. 
Any hints or ideas?
Thanks, Jackwidu

manolllo, 2018-04-16:
@jackwidu I had the same issue that the GPIO buttons were not doing anything. My solution to the problem was the following. Instead of using PCM 1.5db+/- I used Master 3+ and 3-

I think the change to Master is only needed if you use USB Audio which I am using.

After that everything worked like a charm. 

I think this is the code I used. I will verify once I am at home.

def def_volU():
    check_call("amixer sset Master 3+", shell=True)

def def_volD():
    check_call("amixer sset Master 3-", shell=True)

MiczFlor, 2018-04-18:
Hi @jackwidu 
This might help to unify your code (webapp, cards, gpio buttons):
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/playout_controls.sh.sample
I added this in the latest release and you can set the variable for `sset` in line 35.
Then from PHP, Bash or Python, calling this script should do the same action across all ways of interacting with the box.
You can see it in action already in the webapp:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/htdocs/index.php
and the bash script:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/rfid_trigger_play.sh.sample

If you can test it and update the code for the GPIO buttons:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/misc/GPIO-button-sample.py
and test if it works, that would be awesome.

All the best, micz

jackwidu, 2018-04-19:
Well..
BEFORE: GPIO were working, except for the volume controll.
the webinterface was working, including volume control, except for starting a folder manually (no reaction at all).
AFTER: GPIO buttons are working, except for the volume controll.
webinterface, neither volume nor starting a folder manually works. But Pause button works after starting via nfc card.
after changing rfid_trigger_play.sh , htdocs/index.php, and rfid_trigger_play.sh creating (changing PCM to Speaker, setting chmod +x) 
in ssh this works:  ./playout_controls.sh -c=setvolume -v=80
gpio-buttons-sample.py is changed to 
def def_volU():
    check_call("./playout_controls.sh -c=setvolume -v=80", shell=True)

if I call python gpio-buttons-sample.py I can change volume via the gpio-buttons.

Any ideas? I am not even sure, if this stil is my personal error and your change works perfectly for others.

Jakob

MiczFlor, 2018-04-20:
Hi @jackwidu 

I read your post a few times and am not quite sure what works and what doesn't work now :)
~~~
> after changing rfid_trigger_play.sh , htdocs/index.php, 
> and rfid_trigger_play.sh creating (changing PCM to Speaker, setting chmod +x)
~~~
This is of course necessary. And for a fresh install it is documented in CONFIGURE.md.
You say that it works on the command line, like `./playout_controls.sh -c=setvolume -v=80`
but does it also work for the web interface? If not, have you made sure the settings for the webapp are correct?
~~~~
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared
~~~~
From the sound of your mail, it seems that you have been updating your system rather than installed from scratch. And you are right, there were a couple of changes that might affect the system.

Please tell me in short, after your changes: what does not work?

thanks

jackwidu, 2018-04-20:
Sorry, that my description remained unclear. 
It's true, for the reason that 95% work perfectly I did an update and not a clear installation.
Things that are not working now (also after setting chmod as proposed by you):
Volumecontroll via GPIO after starting a file via nfc-card (play/pause, next Button work). Same in the webinterface. In the webinterface I also cannot start any file, I can only use the function play, pause, next track, reboot, reload, shutdown and "last Chip ID". 
Interestingly, if I call the GPIO file directly in SSH with "python /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py" I CAN controll the volume of a playing file via the buttons (though not the play-pause-Button which otherwise works!?). Therefore I think, it might have to something with the way, this file is called in the /etc/rc.local. There it is: "python /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py&" (next and last line "exit 0"). 
That doesn't make any sense to me :-(

Thank you so much!

MiczFlor, 2018-04-23:
It looks to me as if the GPIO script is not running after boot. Why that would be the case, I don't know. And I still don't have a GPIO control running myself. Right now I am working on an update for stretch and will also set up a GPIO Pi for testing.

What I would do: reinstall from scratch.

To make this easier, you can take a look at the helperscripts to create a backup of the shortcuts installed on your machine:
https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/scripts/helperscripts

jackwidu, 2018-04-23:
The GPIO Script is working, as the other buttons (play/pause, next track) do their job nicely, but I think you're right and I should do a reinstall. Thanks for your effort!

MiczFlor, 2018-04-28:
Hi @jackwidu,
I have to give @keinkurt a nudge, because I have no experience with the MRFC522 reader. Did you get in touch with him?
The keinkurt repo is quite a few commits behind this repo, could that be the problem?

jackwidu, 2018-04-30:
Do you have any idea about not being able to start any file via the webinterface?
The link "http://192.168.178.31/index.php?play=/home/pi/RPi-Jukebox-RFID/shared/audiofolders/..whateverfolder" won't start playback. Is there a way to find out, where exactly the problem lies?
In SSH I can trigger the exact mentioned file with cvlc /home/pi/RPi-Jukebox-RFID/shared/audiofolders/102pinguinei
greetings, Jakob

MiczFlor, 2018-05-02:
Hi @jackwidu 
the two things that come to mind are:
* did you [patch the VLC player](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/INSTALL-jessie.md#install-the-media-player-vlc)? 
* did you [add lighttpd to sudoer list](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/INSTALL-jessie.md#configuring-lighttpd)?

Because: the webserver must be operating as sudo to launch VLC. And VLC needs to be patched to allow sudo to launch it.

This would explain why you can not launch it from the web app, but can launch it from the terminal. Can you try to launch it like this?
~~~
sudo cvlc /home/pi/RPi-Jukebox-RFID/shared/audiofolders/102pinguinei
~~~

jackwidu, 2018-05-02:
That gives me..
pi@raspberrypi:~ $ sudo cvlc /home/pi/RPi-Jukebox-RFID/shared/audiofolders/102pinguinei
VLC media player 2.2.6 Umbrella (revision 2.2.6-0-g1aae78981c)
[013aa130] pulse audio output error: PulseAudio server connection failure: Verbindung verweigert
[013b0ce0] core interface error: no suitable interface module
[012fc918] core libvlc error: interface "globalhotkeys,none" initialization failed
[013b0ce0] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[013b0ce0] core interface error: no suitable interface module
[012fc918] core libvlc error: interface "dbus,none" initialization failed
[013b0ce0] dummy interface: using the dummy interface module...
libdvdnav: Using dvdnav version 5.0.3
libdvdread: Encrypted DVD support unavailable.
************************************************
**                                            **
**  No css library available. See             **
**  /usr/share/doc/libdvdread4/README.css     **
**  for more information.                     **
**                                            **
************************************************
libdvdread: Couldn't find device name.
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.IFO failed
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.BUP failed
libdvdread: Can't open file VIDEO_TS.IFO.
libdvdnav: vm: failed to read VIDEO_TS.IFO
[013a3588] core playlist: stopping playback

I uninstalled and reinstalled Linux, patched the binary code and added lighttpd to the sudoer list. Before and after.

MiczFlor, 2018-05-03:
Hi @jackwidu 
Looks like the sudo patch for VLC did not work, because the error seems to be the same. 

Can you try 
~~~
/usr/bin/sudo /usr/bin/cvlc --no-video -I rc --rc-host localhost:4212 '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/102pinguinei'
~~~
and post the results here? (gee, I hope there is no typo, I am not on my box to test the above command...)

I just installed a box myself and it worked 'out of the box'. I will try to do it again and test your setup. Are you working with stretch or jessie?

jackwidu, 2018-05-03:
No typo, but no obvious errors difference :-(
I don't eben know, if the error is of amy relevance..
pi@raspberrypi:~ $ /usr/bin/sudo /usr/bin/cvlc --no-video -I rc --rc-host localhost:4212 '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/102pinguinei'
VLC media player 2.2.6 Umbrella (revision 2.2.6-0-g1aae78981c)
[01648b28] pulse audio output error: PulseAudio server connection failure: Connection refused
[01642618] core interface error: no suitable interface module
[015b2918] core libvlc error: interface "globalhotkeys,none" initialization failed
[01665b88] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[01665b88] core interface error: no suitable interface module
[015b2918] core libvlc error: interface "dbus,none" initialization failed
libdvdnav: Using dvdnav version 5.0.3
[01642618] [cli] lua interface: Listening on host "localhost:4212".
libdvdread: Encrypted DVD support unavailable.
************************************************
**                                            **
**  No css library available. See             **
**  /usr/share/doc/libdvdread4/README.css     **
**  for more information.                     **
**                                            **
************************************************
libdvdread: Couldn't find device name.
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.IFO failed
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.BUP failed
libdvdread: Can't open file VIDEO_TS.IFO.
libdvdnav: vm: failed to read VIDEO_TS.IFO
[015c25e8] core playlist: stopping playback
pi@raspberrypi:~ $ /usr/bin/sudo /usr/bin/cvlc --no-video -I rc --rc-host localhost:4212 '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/102pinguinei'
VLC media player 2.2.6 Umbrella (revision 2.2.6-0-g1aae78981c)
[01648b28] pulse audio output error: PulseAudio server connection failure: Connection refused
[01642618] core interface error: no suitable interface module
[015b2918] core libvlc error: interface "globalhotkeys,none" initialization failed
[01665b88] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[01665b88] core interface error: no suitable interface module
[015b2918] core libvlc error: interface "dbus,none" initialization failed
libdvdnav: Using dvdnav version 5.0.3
[01642618] [cli] lua interface: Listening on host "localhost:4212".
libdvdread: Encrypted DVD support unavailable.
************************************************
**                                            **
**  No css library available. See             **
**  /usr/share/doc/libdvdread4/README.css     **
**  for more information.                     **
**                                            **
************************************************
libdvdread: Couldn't find device name.
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.IFO failed
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.BUP failed
libdvdread: Can't open file VIDEO_TS.IFO.
libdvdnav: vm: failed to read VIDEO_TS.IFO
[015c25e8] core playlist: stopping playback

Neponator, 2018-05-08:
Hey guys, 

I face a similar issue like jackwidu described - every button works (next, previous, VolUp, VolDown) except the "pause" button. If I run 

echo "pause" | nc.openbsd -w 1 localhost 4212 

in the _console_ it perfectly pauses the current music file playing. However if beeing run with a button via "gpio-buttons.py" in the "playout_controls.sh" this exactly same line (echo "pause" | nc.openbsd -w 1 localhost 4212 ) does not work. You have an idea why? I exchanged the "pause" in the "playout_controls.sh" with "stop", that works - so it's not the button and not the code... I am at my wit's end.

MiczFlor, 2018-05-09:
Hi @Neponator hi @jackwidu 
Looks like `sudo` fixed it for jackwidu (see above)
~~~
check_call("sudo amixer sset PCM 300+", shell=True)
~~~
However, it seems that this is not the latest gpio buttons script, which is now using the bash for settings chnages. Now this will say: 
~~~
check_call("./scripts/playout_controls.sh -c=volumeup", shell=True)
~~~
As you can see here: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/misc/GPIO-button-sample.py

Could the two of you figure out if sudo will fix the issue here, too. In which case: please file a pull request. Thanks.
P.S.: hi @itkevin - I quickly ping you here, too, because you committed the original pull request and I believe you know more about the gpio buttons than I do :/ ;)

itkevin, 2018-05-09:
Hi guys,
as @MiczFlor has already mentioned, check out the new GPIO-button-sample.py and copy it into your script folder. Then run the script as root or with `sudo`. You can also check out the `gpio-buttons.service`. It will run it as root on startup.
The Pause button should then work as Pause and Resume when pressed again.

Neponator, 2018-05-09:
Thanks for the quick reply, didn't solve issue directly but the gpio-buttons.service hint brought me on the right track: 
My gpio-buttons.py was running two times

This is caused if you follow the installation guide step-by-step and include the optional parts:
In the [CONFIGURE-stretch.md](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/CONFIGURE-stretch.md#auto-start-the-jukebox) it is written in the "Auto-start the jukebox"-section to enable 

`"sudo systemctl enable gpio-buttons (optional)"`

**--> This starts the gpio-buttons.py for the first time**

Secondly, in the [GPIO-BUTTONS.md](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/GPIO-BUTTONS.md#install-gpio-software) installation guide the last point is to 
```
"sudo nano /etc/rc.local"
"/home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py &"
```
**--> Which starts the gpio-buttons.py for the second time.**

Starting this script twice will pause/unpause that fast, that you can't hear it. Similarly the volume is increased two times and the next buttons jumps two songs further. 

If you face similar issues, check your tasks via `ps -ef` or `top` if the gpio-buttons.py runs twice and kill one / take it off the /etc/rc.local

MiczFlor, 2018-05-09:
Thank @itkevin and @Neponator for the quick reaction time and input. I updated the documentation to be more clear: https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/0ed39e05e68a13870285e2e03d91d006b07fc3b1

(unrelated question: should I close this ticket? It's become more of a forum than a feature :) ?

closed 2018-05-14 by MiczFlor