MiczFlor/RPi-Jukebox-RFID #4: Controlling by RFID

rebellsport, 2017-11-26:
Hi Micz,
yesterday and today  I installed twice your system with jessie to a new sd-card.
Now I start a song by rfid without problems.
But when I prepare the rfid_trigger_play.sh to control 
**pause, shutdown, next etc with rfid** 
it does not respond. 
With stretch and without webinterface everthing works well. So I am not sure where the problem is.
Best regards
Ulli

MiczFlor, 2017-12-04:
Hi Ulli,
just to make sure I understand the issue. You have added the IDs to this script:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/rfid_trigger_play.sh.sample
Between line 31 and 40 you can add IDs to skip next previous, shutdown, etc.
You have to have the IDs within the "" characters (that is my first thought to find the problem).

rebellsport, 2017-12-08:
Hi Micz,
the problem is solved.
I made two faults. First in the app the 50% Volume was simply to silent and my RFID Reader didn‘t work. He beeped but I had a wrong configuration.
Today I installed everything again and it works. Great job Micz, thanks a lot

closed 2017-12-08 by rebellsport

phreakadelle, 2018-11-17:
Is this feature documented somewhere @MiczFlor ?

I think thats very useful as i was exactly for a function like that to control the box without doing the GPIO button stuff. For me this works pretty well.