MiczFlor/RPi-Jukebox-RFID #5: Some Pics of my Jukebox

hailogugo, 2017-12-22:
Here are some pics of my Jukebox.
Thank you Micz for your support.

[http://www.directupload.net/file/d/4944/efjq27ax_jpg.htm](url)
[http://www.directupload.net/file/d/4944/t4occoii_jpg.htm](url)

[http://www.directupload.net/file/d/4944/jbmuylk2_jpg.htm](url)
[http://www.directupload.net/file/d/4944/89sce4u6_jpg.htm](url)
[http://www.directupload.net/file/d/4944/ipe27uz6_jpg.htm](url)

Video
[https://youtu.be/hzfOC9PD-tI](url)

Geliras, 2017-12-22:
Something is wrong with the links. But copied url of the video. Looks great. How did you made the rfid cards? Printed? You could answer in german if this is allowed :)
I designed my cards with photoshop, printed them ("Foto SOFORTDRUCK" at DM) and laminated it together with the rfid card.

hailogugo, 2017-12-23:
I have bought https://www.amazon.de/gp/product/B001CPF3FE/ref=oh_aui_detailpage_o05_s00?ie=UTF8&psc=1 (Klebe Etiketten)

To laminate them was not possible for me because the cards were to hight for my machine.

KingKahn123, 2017-12-29:
[image]
[image]

Geliras, 2017-12-31:
https://www.youtube.com/watch?v=DbpXD0Y3a-Q

[image]
[image]
[image]

tullm, 2018-01-17:
Hi,
wow. these are looking amazing.

Mine is a very simple setup, but works. No Buttons, controlled over chips and music via cards:

[image]

It has the RFID Jukebox, but also has Mopidy installed for files on my NAS and Spotify. Also Airplay Receiver and Spotify Connect is working.

cheers

MiczFlor, 2018-01-17:
@tullm looke nice. Adding it to the gallery now. If you want to share a little howto regarding the modipy installation and configuration, please do so. Others will benefit. And connecting Spotify is a neat addition.

tullm, 2018-01-18:
Hi,
thanks.
I don't think that I get it all together now. But here are some hints:

To add **Airplay Support** to the Box I used: shairport-sync [https://github.com/mikebrady/shairport-sync](url)
Here is a tutorial to install it: [https://thepi.io/how-to-set-up-a-raspberry-pi-airplay-receiver/](url)

To add **spotify-connect** I used: [https://github.com/dtcooper/raspotify](url)
tutorial to install is on github there.

To add the **Mopidy Music Server** (for local or networkfiles, spotify and other music services) I used: [https://www.mopidy.com/](url)
tutorials: 
english: [https://maex.me/2017/10/raspberry-pi-as-a-music-server/](url)
german: [https://braspi.de/blogs/braspi-blog/raspberry-pi-als-musik-server-mit-mopidy](url)
or the very good documentation from mopidy: [https://docs.mopidy.com/en/latest/](url)

I used extensions: spotify, local-images (cover database for local or network files), iris (webinterface), local files.

This was a bit of fiddeling with all I neede with mopidy an sambashare to my NAS and the extensions. But it was worth it.

Hope that helps a bit. Not a Tutorial, but with some reading it should be no problem.

Cheers

TUllm

bgraeb, 2018-05-15:
Finally adding mine. As it was the first IOT'ish project I made, I'm quite proud of it. Only local files right now, as my kid has lots of unheard CD's flying around. When they're thru, I'll add Spotify to it.

[image]

[image]

Thanks for sharing your projects, @MiczFlor. I learned a lot, implementing it and tinkering with the parts included.

pheck, 2018-05-15:
Nice cable management!
I'll be using nearly the same kind of box; just need a weekend of spare time.

Where did you get these grids to protect the openings for the boxes?

bgraeb, 2018-05-16:
These are from a spare mesh paper bin. But beware the orientation of the mesh... looks a bit strange, if you mess it up ;-)

MiczFlor, 2018-05-16:
Very nice indeed. I would love to add this also to the gallery, if I may? Two questions from me:
* Looks like you are running the USB speakers from the power coming from the Pi USB. Is there enough power coming of these?
* With the wooden box: I once had the problem that the material was too thick for the cards to be read. Did you have that problem?

bgraeb, 2018-05-16:
Feel free to add it to the gallery.

Power is enough to run the thing. I run it with a wall plugged usb thingy, or if mobile: I can store a power bank in the box and run the cable through the handle hole. Powerinput with micro usb is on the bottom right corner (you might see the cable ;-) ).

The thickness of the wood is about 1cm (1/3 inch? dunno). It's still enough to go thru the wood. The other side with the fiber glas is to far away. So one can safely store all the cards inside. 

The only thing, that could be problematic is the amount of flickering and brightness emitted by the 4 LEDs.

WalterWHeisenberg, 2018-05-16:
Hello, everybody,
now that I have finished my two boxes, I would like to show you some pictures of them.
But first of all, many thanks to **MiczFlor** for all the software development work, without which I would never have been able to get it up and running. Also many thanks to everyone for the many tips and tricks around the Spotify integration. After a little trial and error, it works very well.

I have built two completely identical boxes for my children, they only differ in colour.
In addition to the functions of the jukebox, I have integrated the respective Spotify Family Account, so that the children can hear the stories etc. independently of each other.
So far, I have only implemented the entire control system using RFID cards. I don't know if I will install any buttons in the future.
Alternatively, I can control the jukebox via the web interface. For Spotify I prefer the surface Iris, which also runs under Mopidy.
The complete power supply, including that of the small speakers, only works via a small USB power supply and the USB ports of the Raspberry.
In the event that there is no power supply, everything could also be operated with a battery. But I haven't tried that yet.

I built the speakers myself from plywood. Since I had no grids for the opening of the speakers, I made a template and drilled all the holes with a 5mm wood drill. By the way, the RFID reader functions completely smoothly  even through the wood.

You are welcome to add the photos and the information to the gallery.

Here is the list of the devices I have installed:
Raspberry pi 3
Simple Case for the pi
UGREEN USB sound card
Neuftech USB RFID Reader 
Trust Leto 2.0 USB Speaker Set
Aukru Micro USB 5 V 3000 mA Charger

[image]
[image]
[image]

MiczFlor, 2018-05-18:
Hi @WalterWHeisenberg, thanks for your pictures. You are an experienced dad: two *identical* boxes for the two kids. In my mind even changing the colour a bit might be triggering outrage ;)
And I really want to say again: I am so impressed about the different shapes and forms these jukeboxes get. This really is the open source and DIY maker spirit. Truely creative and inspiring.

gruenler, 2018-06-18:
[image]

**Thank you for the great work!**

My phoniebox comes with this specs:

- [Wooden box](https://www.rig-tig.com/de/woodstock-kleine-p-3855) 
- [Phat Beat (DAC and VU-Meter)](https://shop.pimoroni.de/products/phat-beat)
- Old RaspberryPi2
- Old PC Speaker (2 speaker)
- RFID Card-Reader via USB
- USB cable with switch
- iPad Power Adapter
- ... some small things

Everything besides changing volume works (acceptable for me ;-)

Are there any news about spotify support out of the box?

MiczFlor, 2018-06-19:
Hi @grnlr 
looks great and big ... and confusing with the carpeting. Combined with the LED, it might well be a small pet disco... ;)

grnlr, 2018-06-20:
@MiczFlor The "carpeting" is damping material for speaker. But i'm no pro - speaker sound design still has room for improvements. But yes, you're right - pet disco is the best description!

MiczFlor, 2018-07-23:
Hi @webjochen, nice one. Looks like a serious MIT PhD prototype :) can I include your work in the gallery? And if so, can I add a link tour side (which URL?) All the best, micz 

P.S.: i am not near my pc and will only get to this after my holidays

webjochen, 2018-07-23:
hi @MiczFlor yes you can use my pics.
link tour side url ?? i do not understand, what do you mean by that

genkigenki, 2018-08-04:
Thanks a lot for the project, updated recently to the latest version.
Here a picture:
[image]

I bought this: 
[Speaker](https://www.amazon.de/gp/product/B076KF1RR9)
[Voltage meter](https://www.amazon.de/gp/product/B0788KT8TN)
[External USB port](https://www.amazon.de/gp/product/B00HE5MUWK)

With the latest version the application starts up a bit slow. Also there is stuttering at the beginning of the songs. Im hoping for the next version without VLC

suedbadner, 2018-08-05:
[
[image]
](url)

MiczFlor, 2018-08-06:
Hi @genkigenki and @suedbadner for the latest contributions :) awesome. 
@genkigenki did you fit the Pi into the box or is this what you are using the external USB port for?
And @suedbadner: what's the difference between the two boxes? (or are they they same, batmobil style detaching the motorbike from the car?)

genkigenki, 2018-08-06:
Hi,

i used this wiring:

Usb-charging port on outside of the box
|
Battery (10000mAh USB battery)
|
ON/OFF switch  ---------------+                              
|               |                          |
Raspi        Amplifier/Auna     Voltage display
|
USB-RFID-reader

Inside the box, i wired:
   Raspi -> Amplifier Aux input (soldered)
   USB battery output -> Amplifier original battery (was only 1000mAh).

So yes, all is internal, the RFID cards work through the wood of the box, if you put them on the top right of the box.

JuCarax, 2018-08-09:
[image]

Caliandroid, 2018-08-09:
this is our father-and-3y-old-son selfbuilt wooden phoniebox with gpio buttons
[image]

morivox, 2018-08-19:
Hi @MiczFlor et al,

what a great project, many thanks for all the support and tutorials! This finally brought me back to do handicrafts and soldering - it had been >20 years! 

Attached is my first version using an old radio and attaching 5 GPIO buttons, which I built with my 4year old son. Since I have a 2nd one, I guess I have to build a 2nd (similar) one too...

[image]
[image]
[image]

Next I want to include fast forwarding, and Napster/Rhapsody support - any hints for that are highly welcome.

MiczFlor, 2018-08-21:
Hi @JuCarax @Caliandroid @morivox 
thanks for the contributions. And also keep the code coming :) I have been planning to start the wiki and migrate the Phoniebox gallery there. There have been so many astonishing additions.Looks like @JuCarax and @morivox added the feature "start playing when RFID is near the Phoniebox". Could you create a thread under issues and document this install? I want to include it in the docs. Thanks!

morivox, 2018-08-21:
Thanks @MiczFlor, unfortunately I am not there, yet, being able to code "start playing when RFID is near the Phoniebox". I still wanted to have that feature so used the Neuftech USB RFID Reader, which does allow for it. It's also a bit cheaper, but does only come with 5 RFID coins. (https://smile.amazon.de/gp/product/B018OYOR3E/). Maybe it's worth referring to that product as well in your "shopping list"?

genkigenki, 2018-08-24:
@morivox : I have the same USB reader, but i dont see any option to know wether the RFID card is near. You can leave the card on the reader and it will continue to play. But there is no way to know that the card is removed, even the LED doesnt change (it initially only blinks once in red and then turns to green as if no card is near). So you remove the card, but it will not stop playing. 

For me, this behaviour is however good, i like it like that.

axelslade, 2018-08-25:
My beta version... For testing and while the real version is in the making... 
Not nice but I had to connect another rfid reader via the breadboard and did not want to solder

[image]
[image]
[image]

axelslade, 2018-08-25:
And my first version, but the box was to unstable and I was to 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ddbkva, 2018-08-28:
i just finished my first phoniebox! Thank you very much !!!!
[image]

MiczFlor, 2018-08-30:
Hi @ddbkva 
quite an impressive Interface-Face you made there, reminds me of "Nightmare before Christmas" ;)

MiczFlor, 2018-08-30:
@axelslade you really use a Jabra for the Phoniebox. Wow, that ups the budget :)

ddbkva, 2018-08-30:
unaware, 
I think my design is a bit friendlier (i hope so it`s for a 3 year old person..)
i built a second box for my daughter, same design

ddbkva, 2018-09-03:
so finished my first (not my last!) phoniebox! it comes with rpi2, wlan, hama usb speakers, intensio powerbank, acarde button and an multilplex and plastic case! For charging there is an usb c 
recessed socket (for the kids its easier than micro usb)!
[image]

ekorn, 2018-09-05:
Mine uses a old jewel case and the Logitech V20 USB Speaker with integrated sound card and Media keys:
[image]

In the first drawer is a rpi zero w and the rfid reader in the second the kids have their cards.

ddbkva, 2018-09-06:
how did you connect the speakers with the raspberry pi? 
I also bought the speakers, but i`m not sure that the power of the raspberry pi is enough. 
are the media keys working ?

hailogugo, 2018-09-06:
Hi 

I have connect the speaker directly per USB at the Pi and on Pin 5V and GND on the Pi.

With the New Rev. 1. I do not add any keys up to now.

But look up in this files

 

 <https://github.com/MiczFlor/RPi-Jukebox-RFID> RPi-Jukebox-RFID/ <https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/misc> misc/ <https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/misc/sampleconfigs> sampleconfigs/gpio-buttons.py.sample

 <https://github.com/MiczFlor/RPi-Jukebox-RFID> RPi-Jukebox-RFID/ <https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/docs> docs/GPIO-BUTTONS.md 

 

Regards

 

Andreas

Snible, 2018-09-13:
Here is my PhonieBox, thanks for this great project.
[image]
[image]
[image]
[image]
[image]
[image]

toblinga, 2018-09-14:
@Snible: Looks verry professional! Well done!
I'm planning also to laser cutting the front and back side. Which cind of color did you use before you lasered the wood? I thought it would not be possible to laser colored wood.

Second question: which box did you use as basic?

Thank you!

Snible, 2018-09-17:
@toblinga 
I've used this paint before laser engraving: [https://www.bauhaus.info/buntlackspray/dupli-color-color-lackspray-ral-9010/p/15071041](url)
The box I've used is this one [https://www.bauhaus.info/holzkisten/stapelbox/p/10750020](url)

axelslade, 2018-09-21:
@MiczFlor was the only speaker I had laying around... But in the new version I have used the cheap ones you have in your buy list. But I have a constant noise (brummschleife) which is not going away

axelslade, 2018-09-21:
> @axelslade you really use a Jabra for the Phoniebox. Wow, that ups the budget :)

@MiczFlor was the only speaker I had laying around... But in the new version I have used the cheap ones you have in your buy list. But I have a constant noise (brummschleife) which is not going away

axelslade, 2018-09-21:
[image]
[image]
[image]
[image]
[image]
[image]

Here is the new and final version...

ddbkva, 2018-09-21:
@ekorn how do you connect the v20 speakers and the rfid reader on one usb micro port at the rpi zero?
i bought some otg 4 port hub, i hope there is enough power for the speakers. if not i could take the usb power from another usb port. i hope it works.

ekorn, 2018-09-24:
No Magic there just a decent USB-Charger (I think 2400 mA) and a old USB-Hub which is way bigger than the rpi zero. Let me know if your otg 4 port hub works and which one you bought, than I would do a upgrade.

ddbkva, 2018-09-25:
i ordered the wondershare 4 port usb hub I'm curious if everything works out
https://www.ebay.de/itm/Waveshare-4-Port-USB-HUB-HAT-for-Raspberry-Pi-3-Model-B-Zero-V1-3-Zero-W/272676340552?_trkparms=aid%3D555017%26algo%3DPL.CASSINI%26ao%3D1%26asc%3D20161121140619%26meid%3D6678f1b9652347d196be0f57af0d9de6%26pid%3D100642%26rk%3D1%26rkt%3D1%26%26itm%3D272676340552&_trksid=p2045573.c100642.m3226

pechart, 2018-09-30:
Here's ours. The inside is not that beatiful, but everything works. Maybe I'll add some buttons on the next version, but for my kids (5/8) the volume control via RFID works good. And also I would need to find some nice buttons to keep the simple design. Thanks for this great project, very inspiring! We're mainly using it for kids news podcasts by WDR5 or some older audiobooks that are not easily available via streaming services.

And I opted for a breakout of the power supply on this, so we can run it on a powerbank or wired. Also I split the audio output with a simple 3,5" Y cable so the kids can attach the device to their boombox (or headphones) in addition to the weak sound of the trust speakers.
[image]

toblinga, 2018-09-30:
Thanks for the answer!

MiczFlor, 2018-10-01:
Hi @pechart 
it is difficult to guess how big or small the box is. It looks huge at first glance, then the speaker next to it seems to dwarf it. First I thought this is a really full sized speaker box, but it's smaller, isn't it?

pechart, 2018-10-01:
Yes, it is smaller.
It's an Ikea vildapel (heise online wrote about IT some time ago) https://www.google.de/shopping/product/1?lsf=seller:17064391,store:7941523779002247195&prds=pid:12738995181875638352,oid:10865875941863359546&q=ikea+bambus+blumentopf&hl=de&ei=SXmyW4HsJ4SWkwXq1qroDQ&lsft=gclid:Cj0KCQjwi8fdBRCVARIsAEkDvnJeclUGOnFiDaw-SbieUt5O_hD3fVrfJJIwh7mF3tx7Hen9o5HEi7gaAhJ6EALw_wcB

ddbkva, 2018-10-01:
so second version
[image]

Alwiz79, 2018-10-07:
And here is my version of the phoniebox.
This is now my third version, and after a “classic” wooden box and a portable version in a small lunch box, I thought it was time for a “daddy version”.
I used an old tube radio which I cleaned, put some new fabrics and some lacquer on it.
All the old electronics are in there but not in use (except for the speaker). I guess I kept everything in there for my karma:-)
It is powered on by pushing the original TA button on the front (which was used to switch to an external audio source (vinyl). The original volume knob can also be used but obviously is now soldered to a new amplifier. Pretty happy with the result.

[image]
[image]
[image]
[image]
[image]

elalemanyo, 2018-10-07:
@Snible can you explain or show more how you did yours? I like a lot how the circuits are looking and I would like to do something similar. Any help you can give me would be great.

Thanks!

Franzformator, 2018-10-11:
@webjochen 
I realy like the Buttons (Play, Stop...) you have used for your Phoniebox. Can you post a link for this? Do this Buttons already has the printing or have you used sticker for this?

Thank You!

webjochen, 2018-10-12:
[Arcade Taster Ebay](https://www.ebay.de/itm/Arcade-Taster-Nr-3-Jamma-Mame-Aktionstaster-Neu-Push-Button-Mikroschalter-/321971550601?var=&hash=item4af6fff989)

no the symoble does not have the buttons,
I had extra adhesive symbols printed by a business colleague

Snible, 2018-10-17:
> @Snible can you explain or show more how you did yours? I like a lot how the circuits are looking and I would like to do something similar. Any help you can give me would be great.
> 
> Thanks!

I've created the board in Autodesk Eagle and a friend of mine created it with a cnc, thats it. I can check if I still have the file and upload it.

marcohorstmann, 2018-10-27:
This is the "Lottibox". I've build it for my daugther Carlotta.
Its 17x17x17cm in size. 

[image]

Franzformator, 2018-10-28:
@marcohorstmann 
Nice Box!
Where did get these stickers on the buttons?

[100 more replies]

closed 2021-03-05 by ZyanKLee