MiczFlor/RPi-Jukebox-RFID #6: WebInterface

KingKahn123, 2017-12-23:
I have reinstalled the JukeBox three days ago. After the new installation the WebInterface does not show the available Audio Files and Audio Folders. Have you changed something on the scripts?

MiczFlor, 2017-12-23:
Hm. I have been changing the scripts. Possibly time to do a proper "release" rather than push to master all the time :)

Inside the htdocs folder are the files for the web interface. I have tried the install at a frech RPi just a few weeks ago. And it did work for me. 

A good start would be: do a `git pull` to get your repo to the latest stage. See if that fixes the issue.

And a question from me would be:

* Have you done the install "out of the box", meaning the folder with all the code is at this location: `/home/pi/RPi-Jukebox-RFID`? If you have installed the code at a different location, you need to change the path in the config.php file

hailogugo, 2017-12-23:
Hi Micz, 
yesterday i wrote this problem to you by mail. I have the same problem.

MiczFlor, 2017-12-24:
I can not reproduce the problem. Please, send me information about the setup.
* Which OS are you using? I tested only Jessie.
* Can you send me a list of files in the `htdocs`folder, please?
* And send me the `config.php` file / post the content here, should only be two or three lines.

KingKahn123, 2017-12-29:
Sorry for the late reply... I did a git pull. But this did not solve the issue for me.

--> OS: PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
--> Files in htdocs: _assets  config.php  config.php.sample  func.php  index.php  page_home.php
--> config.php:
$conf               = array();
$conf['base_url']   = ""; // usually => "", generally: end with trailing slash
$conf['base_path']  = "/home/pi/RPi-Jukebox-RFID"; // absolute path to folder

Remark: I have done an out-of-the-box installation. Therefore I have the same location like in the config file.

RollatorOma, 2017-12-31:
Hi first of all sry for my bad english,
but i have the same problem with the webinterface. There is nothing under Availible audio and nothing under chipID. Ich have made 2 fresh installations one with jessie php5 and one with stretch php7. In both installations the same problem. I ahve the same folders and files as KingKahn123.

Please help me i am an bloody newbee.

greetings

MiczFlor, 2017-12-31:
Hi guys, 

I am offline on holidays and will look into this next week. (I type on my phone :) 

I wish you all a happy new year and will fix and improve in 2018. 

All the best, micz

MiczFlor, 2018-01-05:
Hi,

I tried to reproduce the error on two RPi installs (1 and 3) and could not reproduce it. I can see the folders, the containing files and the matching Chip ID for each folder and the new feature where the Chip ID is shown in a modal.

I am also sure that you have these problems and I would like to reproduce it to fix it :/

Here is a list of things to check:

Inside the `htdocs` folder the access rights are as such:
```
drwxr-xr-x 6 pi pi 4096 Jul 22 22:11 _assets
-rw-r--r-- 1 pi pi  198 Jan  5 08:17 config.php
-rw-r--r-- 1 pi pi  198 Jan  5 08:17 config.php.sample
-rw-r--r-- 1 pi pi 3431 Dec 14 20:09 func.php
-rwxr-xr-x 1 pi pi 5398 Dec 14 20:09 index.php
-rw-r--r-- 1 pi pi 6475 Dec 14 20:09 page_home.php
```

Does the webserver have root access?
```
$ sudo cat /etc/sudoers
```
Gives a result which contains the (probably last) line:
```
www-data ALL=(ALL) NOPASSWD: ALL
```

That's all I can think of right now. Can anybody else help out?

KingKahn123, 2018-01-05:
I checked everything and everything is configured as described above... But no luck. For me the WebInterface is still not working. Does the WebInterface script produces any logs which I can check for any error message?

hailogugo, 2018-01-05:
Hi at all,
in my installtion i think i have not the right laws by the folder

pi@raspberrypi:~ $ ls -ldh /home/pi/RPi-Jukebox-RFID/htdocs/
drwxr-xr-x **3** pi pi 4,0K Jan  5 14:19 /home/pi/RPi-Jukebox-RFID/htdocs/
-rw-r--r-- 1 pi pi 198 Jan  5 14:19 /home/pi/RPi-Jukebox-RFID/htdocs/config.php
-rw-r--r-- 1 pi pi 198 Jan  5 14:19 /home/pi/RPi-Jukebox-RFID/htdocs/config.php.sample
-rw-r--r-- 1 pi pi 3,4K Jan  5 14:19 /home/pi/RPi-Jukebox-RFID/htdocs/func.php
-rwxr-xr-x 1 pi pi 5,3K Jan  5 14:19 /home/pi/RPi-Jukebox-RFID/htdocs/index.php
-rw-r--r-- 1 pi pi 6,4K Jan  5 14:19 /home/pi/RPi-Jukebox-RFID/htdocs/page_home.php

How can i change it?

KingKahn123, 2018-01-07:
Hi hailugo, why do you think that the folder rights are not correct?

MiczFlor, 2018-01-08:
Hi,

I created a file called `test.php` inside the folder `htdocs`. You can either download it manually or get it with a git pull.

When you have the RPi Jukebox running, please go to the web interface and inside the folder where the index.php file is, call the text.php file. The URL might look something like this:

`192.168.123.199/test.php`

Please copy the content of the page and paste it here - or send it to me directly via mail.

This is another stept to get to the bottom of this :) more might follow. Weird....

KingKahn123, 2018-01-08:
Hi, seems to be that the audiofolder and shortcut folder cannot be found:

```
File 'config.php' found and included.

Variable $conf['url_abs'] = http://192.168.33.226/test.php

$conf array:
Array
(
    [base_url] => 
    [base_path] => /home/pi/RPi-Jukebox-RFID
    [url_abs] => http://192.168.33.226/test.php
)

File 'func.php' found and included.

Function 'html_bootstrap3_createHeader' exists.

Calling 'html_bootstrap3_createHeader("en","RPi Jukebox",);'

File 'page_home.php' found.

Folder '/home/pi/RPi-Jukebox-RFID/shared/shortcuts/' not found.

Shortcuts temporary file:
Array
(
)

Shortcuts resolved file:
Array
(
)

Folder '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/' not found.

Audio folders:
Array
(
)

```

However, the folder is exactly at the path mentioned above:

```
pi@JukeBox4Kids:~/RPi-Jukebox-RFID/shared $ ls -l
insgesamt 12
drwxr-xr-x 23 pi pi 4096 Dez 24 10:33 audiofolders
-rw-r--r--  1 pi pi  147 Jan  7 16:12 latestID.txt
drwxr-xr-x  2 pi pi 4096 Dez 29 20:56 shortcuts
```

MiczFlor, 2018-01-08:
Hm, thanks. Looks like the `latestID.txt` is being written by the bash script. Can you confirm this and also confirm that you can open the shared folders via samba?

Could you try to run the following on your RPi via ssh and post the resulting output here. And afterwards see if that makes a difference?

```
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared
```

KingKahn123, 2018-01-08:
That was it... Now it is working. I can see the audiofolders and the last ID used ;-) Thank you very much!

MiczFlor, 2018-01-08:
I reopened this issue again, to ask a question: have you checked this?
Does the webserver have root access?
`
$ sudo cat /etc/sudoers
`
Gives a result which contains the (probably last) line:
`
www-data ALL=(ALL) NOPASSWD: ALL
`

KingKahn123, 2018-01-08:
Yes, this is the last line:

```
#includedir /etc/sudoers.d
www-data ALL=(ALL) NOPASSWD: ALL
```

MiczFlor, 2018-01-08:
Ok, then please @KingKahn123 check one more thing: if you create a new folder and push a file in there via samba and create the shortcut now, without running 
`
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared
`
... are they still visible through the web interface and can you play the files via the web interface?
(I know it's a lot to ask, but we are only one step away from stepping on this *bug*)

KingKahn123, 2018-01-08:
I am happy to help... I have created a new folder "test" and copied a new music file into the folder. The folder is visible in the Webinterface and I can play the file by pushing the "play" button.

Dellex66, 2018-01-09:
Ich weiß nicht, ob hier Englisch Pflicht ist, aber jeder der bisherigen Teilnehmer kann perfekte Beiträge in Deutsch schreiben.
Auch ich kann freudig berichten, dass es nun auch bei mir funktioniert!
Wenn ich alles richtig verstanden habe, sollte der Eintrag 
www-data ALL=(ALL) NOPASSWD: ALL
in der sudoers ausreichen.
Folgendes habe ich im Web gelesen:
„Die Datei /etc/sudoers sollte immer mit dem Befehl visudo bearbeitet werden, da so eine Syntaxprüfung gewährleistet ist. Die letzte Zeile der Sudoers-Datei muss zudem immer leer sein! Bei der direkten Bearbeitung ohne Prüfung kann der kleinste Tippfehler dazu führen, dass man sich aus dem System aussperrt und nur über den Recovery Modus wieder Zugang erhält. Es ist dabei korrekt, die Änderung in sudoers.tmp zu speichern, denn visudo überprüft nun die Syntax und man hat die Möglichkeit, Fehler zu korrigieren.“
Unabhängig davon habe ich auch mal im Internet von einem Problem gelesen, was sich der User auch nicht erklären konnte. Nachdem mit dem Befehl visudo die sudoers editiert wurde, konnte das Problem behoben werden.
Es ist wahrscheinlich nicht die Lösung, aber meiner Meinung nach ein wichtiger Hinweis, um die Anleitungen anzupassen.

RollatorOma, 2018-01-09:
Hallo guten Abend ich hoffe man versteht mich wenn ich auf deutsch schreibe. Ich habe das Problem welches ich am Anfang hatte (ich kann die Available audio Ordner nicht sehen) jetzt nicht mehr. Meine normalen Play Pause Vol  etc. funktionieren. Der Play Button bei Available audio geht aber leider bei mir nicht. Auch nicht wenn ich ein einen Ordner über puti erstelle und eine Datei hineinkopiere.

Hoffentlich kann mir jemand helfen.

Schonmal vielen DANK

Hier noch meine Test.php:

File 'config.php' found and included.

Variable $conf['url_abs'] = http://192.168.178.46/test.php

$conf array:
Array
(
    [base_url] => 
    [base_path] => /home/pi/RPi-Jukebox-RFID
    [url_abs] => http://192.168.178.46/test.php
)

File 'func.php' found and included.

Function 'html_bootstrap3_createHeader' exists.

Calling 'html_bootstrap3_createHeader("en","RPi Jukebox",);'

File 'page_home.php' found.

Folder '/home/pi/RPi-Jukebox-RFID/shared/shortcuts/' found.

Shortcuts temporary file:
Array
(
    [0] => /home/pi/RPi-Jukebox-RFID/shared/shortcuts/0005387556
    [1] => /home/pi/RPi-Jukebox-RFID/shared/shortcuts/0005475267
    [2] => /home/pi/RPi-Jukebox-RFID/shared/shortcuts/0006342511
    [3] => /home/pi/RPi-Jukebox-RFID/shared/shortcuts/placeholder
)

Shortcuts resolved file:
Array
(
    [0005387556] => test
    [0005475267] => test2
    [0006342511] => test4
    [placeholder] => 
)

Folder '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/' found.

Audio folders:
Array
(
    [2] => /home/pi/RPi-Jukebox-RFID/shared/audiofolders/test
    [3] => /home/pi/RPi-Jukebox-RFID/shared/audiofolders/test2
    [4] => /home/pi/RPi-Jukebox-RFID/shared/audiofolders/test3
    [5] => /home/pi/RPi-Jukebox-RFID/shared/audiofolders/test4
)

MiczFlor, 2018-01-10:
Hallo,
ich schreib mal kurz auf Deutsch zurück - und denke, dass ich den Thread dann zusammenfassen - und offen lassen werde. Das mach ich dann in Deutsch und Englisch.

Um die Audiodateien aus dem Browser abspielen zu können MUSS der web server `lighttpd` Zugriff auf sudo haben. Im file `htdocs/index.php` wird beim Play Button folgendes ausgeführt:

```
$exec = "/usr/bin/sudo /usr/bin/cvlc -I rc --rc-host localhost:4212 ".$urlparams['play']." > /dev/null 2>/dev/null &";
```
D.h. lighttpd versucht als sudo (also superuser) den VLC zu starten. 

Ich habe mein sudoer config file von jessie RPi 1st generation im github repo. Das könnte man manuell so aktivieren:
```
sudo cp /home/pi/RPi-Jukebox-RFID/misc/sampleconfigs/sudoers.jessie-default.sample /etc/sudoers
sudo chown root:root /etc/sudoers
sudo chmod 440 /etc/sudoers
```
ACHTUNG: es kann sein, dass das schief geht (siehe auch diesen Thread) Dellex66 Beitrag. Bei mir geht es so. Ich arbeite an einem universellen install script. Aktuell habe ich nur folgendes: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/installscripts/jessie-install-default-01.sh
... das stellt meine box her, mit meinem WiFi und meiner IP adresse. Ist also nicht universell, aber funktioniert. Das sudoer file müsste universell sein.

RollatorOma, 2018-01-11:
Hallo,

erst mal vielen DANK für deine tolle Arbeit! Ich habe einen neuen Pull gemacht und dann alle Befehle deines Script ausgeführt. Mit dem Ergebniss, Oft passiert nichts und mal gehts beim 5 Drücken auf den Button. Oder auch mal mit Pause beim Ersten. Beim RFID Chip gehts  sofort. 
Zur Hardware ich verwende einen Raspi3 mit einem HiFiBerry MiniAmp, Neuftech USB RFID Reader.

Mich würde es sehr freuen wenn jemand eine Lösung fenden könnte.

Vielen Dank

Gruß

Markus

hailogugo, 2018-01-12:
Hallo habe deine E-Mail nur zufällig gesehen.

 

Du kannst auf Github ein Request eröffnen damit dir andere helfen können.

Das mit den buttons häng von den Buttuns ab. Wenn die kippeln und der Kontakt wird beim drücken mehrfach geöffnent und geschlossen dann macht der mehrfach natürlich den Befehl. Das kannst du verhindern indem du eine Bouncezeit hinzufügst. Ähnlich wie bei der Halt taste. Dann muss der knopf z.B. 0,1 Sekunden gedrückt werden.

 

MFG

RollatorOma, 2018-01-13:
Hallo erstmal Vielen Dank für deine Antwort. Es liegt nicht an der Hardware Buttons. Es sind die im Browser.

Gruß

Markus

MiczFlor, 2018-01-16:
Hi everybody,
this thread has been opened some time ago and I keep it open, because we are still looking for some answers for some dark magic that happens in the web interface. But first some facts, then the known issues:
## web server can access audio files and folders?
The web server needs to able to see and read the folders and files with audio content and shortcuts for RFID chips. Type the following:
~~~~
sudo chown -R pi:www-data /home/pi/RPi-Jukebox-RFID/shared
sudo chmod -R 775 /home/pi/RPi-Jukebox-RFID/shared
~~~~
## root access for the web server?
Does the webserver have root access?
~~~~
$ sudo cat /etc/sudoers
~~~~
Gives a result which contains the (probably last) line:
~~~~
www-data ALL=(ALL) NOPASSWD: ALL
~~~~
Because otherwise the problem could be that the new files added via samba are also not visible to the webserver. And/or the webserver can not start playback on VLC.

Edit the `sudoers` file using the [visudo](https://www.digitalocean.com/community/tutorials/how-to-edit-the-sudoers-file-on-ubuntu-and-centos) editor
## known issues - can you help?
* Sometimes, audio folders can not be played via the web interface while they work flawlessly using the RFID chip. We need to learn more about this, because I can not reproduce this bug.
   * Does this happen all the time for the same audio folder?
   * Does the audio folder contain special characters like white spaces or Ä type UTF8 chars?
   * If the behaviour only occurs sometimes and sometimes not, does it occur sometimes for all audio folders or only for a few?
   * If this happens to you, please share your theory here, no matter how bizarre it might sound.

hailogugo, 2018-02-28:
Hi Micz, 
i install it all again for a secound Jukebox now. My System is Rasbian Stretch. If i use the play button in the webinterface of any dictionary no one will start to play. The button on top will work if i use a rfid kart and something is playing.

What info`'s do you need to help me?

Regards

MiczFlor, 2018-03-18:
@hailogugo the code was now updated to work with php7 and there was another thread related to the vlc player with --no-video option. Do you still have the problems you describe here? And if not, how did oyu solve them? Please share. thanks, m

hailogugo, 2018-03-27:
Hi
I will Test It again. At the Moment i work on Other things. I have at in a ky-040 for vol control. I will Send you the howto.
Micz please look up in the Forum again. They have Add a Display too.

https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?pageNo=19

hailogugo, 2018-03-28:
Hi Micz,

yes it works. THX

MiczFlor, 2018-03-28:
@hailogugo nice! I would like to see more pics :) I will take a look into the XML status file. I had no idea that this existed.

closed 2018-04-16 by MiczFlor