MiczFlor/RPi-Jukebox-RFID #11: RC522 module

B5GeK8A2YqH9, 2018-01-23:
First of all thank you for this awesome tutorial. Even as bloody beginner I was able to get most of my own project - a jukebox for my little son - running. The only thing which suddenly stopped me was, that you are using an USB RFID reader instead of the RC522 module, which I am using.
If somebody can give me a hint how to integrate this one instead of an USB reader, this would be great!

MiczFlor, 2018-01-24:
(I leave this issue open, because I get similar questions quite often)

This code base only works with USB RFID readers and I can not give any advice how to use this code with other models (like using pins). If anybody else has ideas and pointers, feel free to post them in this issue thread.

USB RFID readers usually are set up in a way that they act "like a keyboard". The Chip ID is pushed to the RPi as if you were typing it in a keyboard and pressed "enter". This input is what the python daemon is waiting for and then it pipes the result into the bash script.

keinkurt, 2018-03-05:
I have adjust the code to work with the RC522 module over GPIOs. Have a look at https://github.com/keinkurt/RPi-Jukebox-RFID/tree/MFRC255

BTW @MiczFlor Great project, thank you very much. I'm now beginning to build my own Jukebox.

MiczFlor, 2018-03-05:
@keinkurt thanks for the scripts. Would it be ok, if I copy them over in a subfolder to this project?

keinkurt, 2018-03-06:
@MiczFlor yes, no problem for me. Don't forget to copy the MFRC522.py, too.

jackwidu, 2018-03-24:
both of you, great project! 
@keinkurt , can you give me further advices as how to activate and use it with the mfrc522? I know how to read my uids (https://github.com/pimylifeup/MFRC522-python) but have no clue how to register the device as in configure.md or proceed even further in the manual.
Thanks a lot! Jakob@Kiel

MiczFlor, 2018-03-26:
@keinkurt @jackwidu If either of you or both of you would write a quick "howto" for the troubleshooting section on "how to (try to) add any RFID reader" - that would be awesome. I don't plan to buy a couple to play with them, but it seems I don't need to, because there is great talent here in the (small) community to fix this issue. Thanks!

jackwidu, 2018-03-26:
Ok, finally I got it working. There were bugs that of course weren't your fault. 
To help others:
1) go throught the installation manual. Be sure, that you have the mfrc522.py file in your scripts folder at the end. If you encounter errors, replace all the "php" installation with "php5". I have no idea what the linus-headers are for, but for my raspPi it didn't work (couldn't find 4.14)
2) make sure your NFC Chip works and is properly connected. Try it with https://github.com/pimylifeup/MFRC522-python or the more conclusive description on https://pimylifeup.com/raspberry-pi-rfid-rc522/. When you manually start the Python Scripts it should be able to tell you the UID. I encountered problems, which were related to missing tabstops (always on the last line). Be careful and correct your python scripts if necessary.
3) From the configure.md you can totally ignore the passages "Identify and configure your RFID reader" and "Register your USB device for the jukebox". To be sure, there are no errors, it might be a good idea to start the daemon_rfid_reader.py file manually (python /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py). If this again gives you the uid back, everything looks fine. By now, you should also be able to visit the web app and folder via smb.
Thats it, all credit to @MiczFlor and @keinkurt !!!

ottk3, 2018-04-04:
Works like a charm on my RPi3 with the RFIO-RC522 module. Thanks for all your effort!
I used the *jessie-install-default-01.sh* for the initial installation. 

Additional steps I took:

1. Connect the RC522 module properly
2. Enable the SPI (raspi-config or /boot/config.txt)
3. Install python-spidev (`sudo apt-get install python-spidev`)
4. Copy the scripts from @keinkurt into the scripts folders

5. Enjoy!

MiczFlor, 2018-04-16:
Hi @keinkurt @ottk3 @jackwidu @B5GeK8A2YqH9 thanks for your input. 

Personally, I haven't yet gotten my hands dirty with anything beyond the USB plugin. Thanks for your input and I will link to this thread from the manual. (until I will get my hands dirty, build one with the RC522, verify your great work, include it and finally close the issue ;)

raymonvdm, 2018-04-17:
Strange i cannot find the SPI module
 
```

root@JukeBox:/home/pi/RPi-Jukebox-RFID/scripts# dpkg -l | grep spidev
ii  python-spidev                   20170223~145721-1                 all          Bindings for Linux SPI access through spidev (Python 2)
ii  python3-spidev                  20170223~145721-1                 all          Bindings for Linux SPI access through spidev (Python 3)
root@JukeBox:/home/pi/RPi-Jukebox-RFID/scripts# python Reader.py
Traceback (most recent call last):
  File "Reader.py", line 1, in <module>
    import MFRC522
  File "/home/pi/RPi-Jukebox-RFID/scripts/MFRC522.py", line 5, in <module>
    import spi
ImportError: No module named spi
root@JukeBox:/home/pi/RPi-Jukebox-RFID/scripts# 

```

raymonvdm, 2018-04-18:
Is it possible to add a buzzer to the RC522 / Pi to get the beep when reading a TAG. I have found a working example but i don`t no how to translate this to the current project

https://github.com/eoghan-c/team_track/blob/master/team_track.py

keinkurt, 2018-04-18:
@raymonvdm it's a good idea. I will try it and after that I can tell you what to do.

MalteHST, 2018-07-06:
Hello. First of all thank you for this nice project. 
Currently I'm trying to get it work with a Zero W and a RC522 module, which I have left from another project. 
The reader is already properly working with my Zero. (tested with a simple MFRC522 script to check if the module can read and write). Audio output with the help of the web server works fine, too. 

Unfortunately I don't really get it at which stage the scripts from @keinkurt must be integrated/ copied/ combined with the scripts from @MiczFlor. Thatswhy I can't read any RFIDs over the web server. 
Maybe anyone can give a foolproof step description to me. 

( I think my problem is that I don't have the mfrc522.py file in my script folder???)

MiczFlor, 2018-07-09:
Hi @MalteHST,

I am not sure I can be of help here, actually, I am sure I can be of no
help here, because I don't use the RC522 module. Hopefully @keinkurt
can jump to the rescue?

If you find a solution, please share it here, so I can add it to the
documentation. Don't forget :)

Thanks, micz

MalteHST, 2018-07-15:
Hi @MiczFlor,
fortunately I could help me with the troubleshooting section of you. I had the problem that the rfid reader works perfectly if I communicate per terminal, but not in the webserver. But I added the delay at reboot as you described it. Thanks.

MalteHST, 2018-07-24:
Has anyone tried to implement the "Stop on removal" - Function for the RC522 Module? 
Maybe this gives a sufficient approach:
https://github.com/miguelbalboa/rfid/issues/188

But I'm not so deep in this project, that I know where to add code and if it is really that simple like described in the link above.

keinkurt, 2018-08-09:
@MalteHST Please try this adjustment on the rfid daemon and RC522 Reader: https://github.com/keinkurt/RPi-Jukebox-RFID/commit/0b48c4d841d51657ad6ca1602a4abe3b39e6f7f3

MiczFlor, 2018-08-10:
Hi @keinkurt 
how about adding both flavours of your script (as pull requests :) to the code base here? I will then add them to the documentation.
All the best, micz

MalteHST, 2018-08-10:
Many thanks @keinkurt. Stop on removal - Feature works perfectly with this adjustment. Also has a good reaction time.

MalteHST, 2018-08-17:
The RC522 module works also with the new 1.0 release as desired. Except that 'Latest_RFID' in the settings folder did not update with a new RFID tag. So actually you can only register new id cards manually in the shared folder. Because in the shortcuts folder the new id card appears when swiping over the reader. Then you manually edit the corresponding audio file.

leachiM2k, 2018-09-10:
Hi @keinkurt,
thank you for providing this implementation for RC522 Reader. 
Do you think it would be a valid option to integrate your script and the script from https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/175 that way, that enables us users to choose from USB, RC522 and RDM6300? Maybe in the installscript?

Ardios, 2018-09-24:
I can't get the reader running under version 1.1.2 anymore does anyone else have the problem?

Ardios, 2018-09-25:
Hey @keinkurt,

is there a way to use the reader normally or is it just possible to use the stop on removal method?

Thank you
Ardios

keinkurt, 2018-09-26:
@Ardios I'm working on a solution to integrate the RC522 to main repository and installation process. Please wait for this.

keinkurt, 2018-10-03:
this PR #235 should now working with usb, rc522 and rdm6300 devices, dynamically. @Ardios please test it!

MiczFlor, 2018-10-03:
Hi @keinkurt @Ardios 
I merged the pull request into the `develop` branch - if you want to test it, please checkout that branch. I will only find time to test on the weekend, and only on a USB device.
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/235

varac, 2018-11-04:
@MiczFlor First of all, thanks for this amazing project !

I'm trying to set up my MFRC522 reader, but I fail.
Using current master or develop, `RegisterDevice.py` is unable to detect my rfid card reader.
Only when I use the `Version-1.1.5-dynamicRFIDreader` branch, my reader gets recognized:

```
pi@musicbox:~/RPi-Jukebox-RFID/scripts $ python RegisterDevice.py
Choose the reader from list
..

pi@musicbox:~/RPi-Jukebox-RFID/scripts $ git checkout develop
Branch develop set up to track remote branch develop from origin.
Switched to a new branch 'develop'

pi@musicbox:~/RPi-Jukebox-RFID/scripts $ python RegisterDevice.py
Choose the reader from list

pi@musicbox:~/RPi-Jukebox-RFID/scripts $ git checkout Version-1.1.5-dynamicRFIDreader
Switched to branch 'Version-1.1.5-dynamicRFIDreader'
Your branch is up-to-date with 'origin/Version-1.1.5-dynamicRFIDreader'.

pi@musicbox:~/RPi-Jukebox-RFID/scripts $ python RegisterDevice.py                                                                                                                        
Choose the reader from list:
0 MFRC522
1 RDM6300
…
```

Can you please give details about the current state of usage of non-USB readers, or point me to current docs how to set it up ? Thx !

varac, 2018-11-04:
In the `Version-1.1.5-dynamicRFIDreader` branch I could register and use the reader successfully, `daemon_rfid_reader.py` shows `Card found` when reading a RFID tag.

Using `daemon_rfid_reader.py` from master branch fails:

```
pi@musicbox:~/RPi-Jukebox-RFID/scripts $ python /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py                                                                                  
Could not find the device MFRC522
. Make sure is connected

```

martinclausen2, 2018-11-25:
I also had to install pyserial. Otherwise it is working fine following the posts from above.

internetfan420, 2018-11-28:
> The only problem with the `Version-1.1.5-dynamicRFIDreader` branch seems to be that the webinterface settings still show `RFID Reader Disabled`, and `Switch ON` don't have any effect (where are the logs from the webinterface, `logs/debug.log` doesn't show any output from trying to switch it on) ?

Also have the exact same issue. 
when registering new cards tho it autopopulates with RFID card scanned. even tho the 
'RFID Reader Disabled' and the 'Switch ON' doesnt do anything

alex-voigt, 2018-12-08:
> Thanks for the tutorial.
> Unfortunately, I can't get it running.
> Following the howto from jackwidu I can read and write cards via pimilifeup.
> But Phoniebox won't recognize any card ID anyway. Starting the daemon_rfid_reader.py manually, he won't do anything as well. cancelling it he gives me feedback as follows:
> 
> Traceback (most recent call last):
> File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 31, in 
> cardid = reader.reader.read_card()
> File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 36, in read_card
> select([self.dev], [], [])
> File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 20, in end_read
> reader.reader.cleanup()
> AttributeError: 'UsbReader' object has no attribute 'cleanup'
> 
> What am I doing wrong? Sorry, I'm not that experienced in this. Thanks

Did you already register the device? Just run:
`python ~/RPi-Jukebox-RFID/scripts/RegisterDevice.py`
and choose the right device number for MFRC522

fapoo, 2018-12-09:
> Did you already register the device? Just run:
> `python ~/RPi-Jukebox-RFID/scripts/RegisterDevice.py`
> and choose the right device number for MFRC522

I tried to. But this is all I get from the list:
_Choose the reader from list
0 C-Media Electronics Inc.       USB PnP Sound Device
Device Number:_

pyriand3r, 2018-12-26:
I have tried to use an MFRC522 with phoniebox. Got it working with the 1.1.5 branch but had reinstall clean. So i am just doing a reinstall and saw this during installation. Maybe that is the reason why it is not working in the master branch? Not shure, don't have the time to investigate further.

```
  Downloading https://files.pythonhosted.org/packages/84/f2/e3d02257949e9caa9bff26044e0185e743ff45d3e0e099a93880e10bf718/pi-rc522-2.2.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-tLkqiZ/pi-rc522/setup.py", line 11, in <module>
        from pirc522 import __version__  # flake8: noqa
      File "pirc522/__init__.py", line 4, in <module>
        from .rfid import RFID
      File "pirc522/rfid.py", line 2, in <module>
        import spidev
    ImportError: No module named spidev
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-tLkqiZ/pi-rc522/
```

martinclausen2, 2018-12-27:
Hi @pyriand3r I guess you need to install the spidev python lib manually.

EsraDan, 2018-12-30:
@pyriand3r: Joining your problem.
Seems to be one of the "well known" python2<->python3 issues.
Couldn't install the missing spidev manually.
But installing _apt-get install python3-pip_ gives me a _pip3_.
With _pip3 install -r requirements.txt_ the script runs to the end.

Lets see if I make it work...

fhuebner2, 2019-01-18:
Hi, I have the same issue and could install the requirements with `pip3 install -r requirements.txt`. The card reader is working (I tested it with MFRC522 Read.py).
`sudo systemctl status phoniebox-rfid-reader` shows `active (running)` but if I swipe the card the web interface is not popping up the card id.
For testing, I stopped the `service sudo systemctl stop phoniebox-rfid-reader` and started `python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py` manually but it's not showing any information except
`pi@raspberrypi:~ $ python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py
/home/pi/RPi-Jukebox-RFID/scripts/pirc522/rfid.py:78: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(pin_rst, GPIO.OUT)
/home/pi/RPi-Jukebox-RFID/scripts
`
Any idea if it is a problem that the channel is already in use? And how can I solve my issue? Any ideas?

MiczFlor, 2019-01-18:
Hi @fhuebner2 
Could you add that to the wiki please?
All the best, micz 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

fhuebner2, 2019-01-18:
Hi @MiczFlor 
I wouldn't add this as long as my issue is not solved :-),

EsraDan, 2019-01-18:
Hi folks!
At last I fixed my problems by the typical way: rtfm

I downloaded the latest stretch image.
Made a update&upgrade.
And it work out of the box by using installation string & batch file.

My mistake was to install an older stretch via pybakery to get the headless setup. Update&Upgrade didn't work for me. (on Pi 3)

I also set up the hole thing on zero w to shrink it.
Even there, with the latest image, the installations failed, because I didn't update&upgrade.
After update&upgrade everything worked perfectly.

So in my opinion the hole installation tool chain is very sensitive to the environment.
But that's quite common...

MiczFlor, 2019-01-18:
@EsraDan thanks for the input. The install script might well be more up to date than the step by step guide. I haven‘t updated that for some time. 
Possibly I should add more infos in the comments of the install script and limit the wiki to tweaks and additional info. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

EsraDan, 2019-01-18:
@fhuebner2
I can't remember exactly which steps I have done by installing everything on the second device.(Was quick&dirty in the middle of the night thing)

But I remember a strange feeling that update&upgrade->installation-script throwed an exception.
Because I was almost sleeping and my thoughts were not so clear, I just type upgrade&update (what is "wrong") and tried it again. 

So in my opinion that fits with your experience.

fhuebner2, 2019-01-20:
Sorry, unfortunately it's still not working. :(

fapoo, 2019-01-20:
Same here. Can't get it running.

fhuebner2, 2019-01-21:
I did a complete new installation with Normal Raspbian stretch (but without recommended software).
Still not get it running :(

Ardios, 2019-01-21:
Hey, at first sorry for my broken english.

a german guide from me u can find here: https://raspi.mobi/viewtopic.php?f=10&t=4

i had problems in earlier versions so now i always install 1-2 things before i start with the one line script

1.) Start with a Fresh Installation of raspian lite
2.) Then we searching for updates `sudo apt update && sudo apt upgrade`
3.) Now install few packages `sudo apt install python-dev python-pip git` (I have no idea if it's necessary or not, but that's how it works for me.)
4.) Now we install spidev `sudo pip install spidev`
5.) Now we install pi-rc522 `sudo pip install pi-rc522`
6.) After this we install the phoniebox with the One line Script from here https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#one-line-install-command (in my case i used the classic Edition)
7.) After the Installation we have to copy two files `cp /home/pi/RPi-Jukebox-RFID/scripts/Reader.py /home/pi/RPi-Jukebox-RFID/scripts/Reader.py.old`
8.) And `cp /home/pi/RPi-Jukebox-RFID/scripts/Reader.py.experimental /home/pi/RPi-Jukebox-RFID/scripts/Reader.py`
9.) Now we must Register our Reader with `python /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py` and select 0 MFRC522
10.) Go in the webgui to Settings and switch the RFID Reader button to on

Finish

i hope it works and again sorry for my bad broken english ;)

fapoo, 2019-01-21:
> Hey, at first sorry for my broken english.
>
> a german guide from me u can find here: https://raspi.mobi/viewtopic.php?f=10&t=4
>
> i had problems in earlier versions so now i always install 1-2 things [...]

That's exactly what I was looking for. I'll try it later.
Besten Dank.

Ardios, 2019-01-21:
@fhuebner2 
I just remembered something...How did you wire the reader? 7 or 8 wires? u need to wire the IRQ pin too for the pi-rc522 package…

https://github.com/ondryaso/pi-rc522   scroll a litte bit down to "Connecting"

fhuebner2, 2019-01-21:
@Ardios I can't believe it! That did it! Thank you so much. Later I will try to get it running with stretch lite. Shouldn't be a problem anymore.

Ardios, 2019-01-22:
Here a fritzing layout for wire a RC522 with the IRQ Pin, 2 different Version (RPi3 and RPi Zero..only visual nothing else)

RPi Zero:
http://raspi.mobi/up/RPi_Zero_RC522.png

RPi 3:
http://raspi.mobi/up/RPi3_RC522.png

georgbachmann, 2019-01-25:
@Ardios wonderbar!!! It works now for me! Awesome! :) Thank you so much!!! Vielen lieben Dank!!!

Ardios, 2019-01-26:
@georgbachmann 

have you wire the IRQ Pin from RC522 to Raspberry Pin18/GPIO24?

See the fritzing Picture above...you must soldering all 8 wires from the Reader...MFRC522-python works without the IRQ PIN but you need it for the Phoniebox

georgbachmann, 2019-01-26:
Awesome!!!! I was following a tutorial where that pin was left empty... thank you very much!!! Now it's working like it should! Great!!! :)

MiczFlor, 2019-02-01:
Hi @Ardios @georgbachmann 
I added your wiring and info to the wiki. Thanks for sorting this out here in the issue thread - for all to benefit.
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Wiring_for_RC522_card_reader

domu83, 2019-02-16:
Hi!
really great project... I think I did the newbie mistake and try the have all approach (Display, rotary switch, RC522, Buttons,...)
I just try to get the basic done. To keep it simple initially by using the classic version. I have followed Ardios how to, all 8 pins soldered. Reader LED is on and config has worked, incl. enabling in the WebUI. However a swipe does literally nothing and I do not even get the latestID.txt in my shared folders.
Maybe the last one is a result of having the service not started?

pi@raspberrypi:~ $ sudo systemctl status phoniebox-rfid-reader
● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/phoniebox-rfid-reader.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-02-16 12:42:37 GMT; 1s ago
  Process: 1171 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py (code=exited, status
 Main PID: 1171 (code=exited, status=1/FAILURE)

Feb 16 12:42:37 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Feb 16 12:42:37 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Service hold-off time over, sch
Feb 16 12:42:37 raspberrypi systemd[1]: Stopped Phoniebox RFID-Reader Service.
Feb 16 12:42:37 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Start request repeated too quic
Feb 16 12:42:37 raspberrypi systemd[1]: Failed to start Phoniebox RFID-Reader Service.
Feb 16 12:42:37 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Unit entered failed state.
Feb 16 12:42:37 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
lines 1-13/13 (END)...skipping...
● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/phoniebox-rfid-reader.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-02-16 12:42:37 GMT; 1s ago
  Process: 1171 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py (code=exited, status=1/FAILURE)
 Main PID: 1171 (code=exited, status=1/FAILURE)

Feb 16 12:42:37 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Feb 16 12:42:37 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Service hold-off time over, scheduling restart.
Feb 16 12:42:37 raspberrypi systemd[1]: Stopped Phoniebox RFID-Reader Service.
Feb 16 12:42:37 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Start request repeated too quickly.
Feb 16 12:42:37 raspberrypi systemd[1]: Failed to start Phoniebox RFID-Reader Service.
Feb 16 12:42:37 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Unit entered failed state.
Feb 16 12:42:37 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.

any hint would be really helpful! Thanks a lot!

pseeger, 2019-02-17:
> really great project... I think I did the newbie mistake and try the have all approach (Display, rotary switch, RC522, Buttons,...)
> ...
> any hint would be really helpful! Thanks a lot!

Which version of scripts/Reader.py are you using? Did you replace the default one with Reader.py.experimental? RC522 does need the experimental verison.

domu83, 2019-02-17:
yes, I have replaced the default with the experimental, so Reader.py is the RC522 version

Ardios, 2019-02-19:
@domu83 

sorry for the late answer.

Can we test a few things?

Try this:

`git clone https://github.com/ondryaso/pi-rc522.git`

Then start the example Read Script with:

`python /home/pi/pi-rc522/examples/Read.py`

Now you can teat the Reader, is this working for you?

When not, i think you have to check the wire

As always, excuse my bad English. ;)

Ardios

domu83, 2019-02-19:
Hi Ardios,

no problem, what language is your preferred?
I get the following when I execute your guidance:

pi@raspberrypi:~ $ python /home/pi/pi-rc522/examples/Read.py
Traceback (most recent call last):
  File "/home/pi/pi-rc522/examples/Read.py", line 10, in <module>
    rdr = RFID()
  File "/usr/local/lib/python2.7/dist-packages/pirc522/rfid.py", line 47, in __init__
    self.spi.open(bus, device)
IOError: [Errno 2] No such file or directory

Ardios, 2019-02-19:
@domu83 

i'm from Germany ;)

Try this:

`sudo raspi-config`

Then:

`5 Interfacing Options`

Then:

`P4 SPI`

Then:

Would you like the SPI interface to be enabled?
`Select Yes`

Now Reboot and test again

Ardios

domu83, 2019-02-19:
lol... ich auch 
jetzt hängt er nachdem
 python /home/pi/pi-rc522/examples/Read.py
/usr/local/lib/python2.7/dist-packages/pirc522/rfid.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(pin_rst, GPIO.OUT)
Starting

[41 more replies]

closed 2023-12-11 by s-martin