MiczFlor/RPi-Jukebox-RFID #13: INSTALL.md-Suggestions after successful build

noestreich, 2018-02-13:
First, thanks a bunch for the effort you put in this project and for the detailed instructions. It worked out flawlessly for me and will supply my daughter with a great piece of easy-to-use hardware. 
Hats off to you. 

I just wanted to suggest a couple of minor-changes for the [INSTALL.md](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/INSTALL.md)-Tutorial, on passages which caught me off-guard: 

- **Installscript**
I did not see (or better: i missed) the well commented installscripts in [/scripts/installscripts/](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/scripts/installscripts) - maybe they ought be linked in the tutorial for their time-saving nature? 

- **Configuring lighttpd**
In the last step of configuring lighttpd, we set the rights for the shared-folder to make sure ist is accessible by the web server. But at this point in the build-process the folder does not yet exist. We are cloning the repo a couple of steps down the line. maybe the chmod-action should be placed under the ["Install git"-section](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/INSTALL.md#install-git)?

- **Using a USB soundcard and Volume**
I made use of the linked stackexchange-article regarding the configuration of the usb soundcard [via ~/.asoundrc](https://raspberrypi.stackexchange.com/questions/39928/unable-to-set-default-input-and-output-audio-device-on-raspberry-jessie). 
It worked without a problem. But after finishing up everything else, i was not able to set the volume video the Web-App. 
I had to edit [Line 60 in /htdocs/index.php](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/htdocs/index.php#L60) and replace 'PCM' with 'Speaker' - this might be helpful for others ordering from your proposed shopping-list (like i did). The line in question looks like this now: 

`exec("/usr/bin/sudo amixer sset 'Speaker' ".$urlparams['volume']."%");`

- **PHP7**
Since we are living in PHP7-Times i used the following Line to install the php-dependencies

`sudo apt-get install lighttpd php7.0-common php7.0-cgi php7.0`

i can report, it all seems to work just fine.

MiczFlor, 2018-02-23:
Hi, thank you for these suggestions. I'll give them a try and will incorporate them in the scripts. And, true, sometimes there are tweaks that make it into the manual based on memory - never a reliable source ;)

I will leave this open for now, until I included your work.

MiczFlor, 2018-03-26:
@noestreich I fixed the inconsistency regarding the git install and chmod. I also changed the manual to support "php" in general and not "php5" (what was I thinking :). One thing where I could really benefit from your input:

`Using a USB soundcard and changing the volume`

If you take a look here: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/20 it seems you are not alone with your tweaks. However, your's seems to be more general. Can you create a pull request for your solution (if it works globally) or give me a list of changes that I can add to the manual? Highly appreciated. Thanks.

MiczFlor, 2018-04-16:
Changed the install back to php5 after all, because the newer version with Stretch is in the making.

closed 2018-04-16 by MiczFlor