MiczFlor/RPi-Jukebox-RFID #14: RFID reader working properly ??

mtacki, 2018-02-26:
Hi all,

I followed the steps above with the exact same hardware...
Unfortunately, my RFID-Reader doesn't seem to work properly.
It's making it's "BEEP"-sound on power-on - but there's no beep when I'm holding a NFC Card next to it... and YES - they are the nfs cards that came with the reader and should therefore be the compatible types.

I could not find any setup steps regarding the reader during the tutorial !?

Shouldn't there be any drivers installed for the reader ??

Thanks and many greets !!

Mark

mtacki, 2018-02-26:
add:
sudo lsusb -v gives me:

Bus 001 Device 004: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x7523 HL-340 USB-Serial adapter
  bcdDevice            2.62
  iManufacturer           0 
  iProduct                2 USB2.0-Serial
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      1 
      bInterfaceProtocol      2 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

MiczFlor, 2018-02-27:
Hi @mtacki 
I wish I could help you, but I can't help you much. The USB card readers that I came across all work like this:

- Plug it in: the PC will recognize it as a keyboard
- Swipe a card / key fob: the USB reader will act like a keyboard and "type in the number and press enter"

Therefore: swiping a RFID chip in front of a USB reader will always "enter the number" no matter where you are. If you have the command line open or a text editor, it will literally type the numbers and start a new line.

The ones I know all beep when you swipe. But that might differ from model to model. And I know that not all cards work on all readers. There are differences.

However, as I said, I can't help you. There is the one in a million chance your reader does not work. That might be the case if the wires are broken.

What I do know for sure: the Raspberry Pi does not need any drivers to work with a USB reader.

I hope this helps. All the best, micz

mtacki, 2018-02-27:
Thank you so much for your answer !!
Since my reader doesn't seem to work properly, I contacted the shop and asked for a refund / exchange... curious how it goes !
Have a nice week !

Mark

MiczFlor, 2018-02-28:
Hi @mtacki 
good luck with the project. As I said, I can not say if the USB reader is faulty or not. The readers which you attach to Arduino devices for example make no sound at all. So I told you all I know, good luck.

closed 2018-02-28 by MiczFlor