MiczFlor/RPi-Jukebox-RFID #15: RPi-Jukebox-RFID with a Pi Zero?

pjcoleman73, 2018-02-28:
Has anyone tried this project using a Raspberry Pi Zero? 
I have an unused Pi Zero and would like to avoid buying an additional pi 3 if I don't have to. I'm wondering if anyone has tried, or has any thoughts on, using a Pi Zero with a Micro USB to USB OTG hub. 

Additionally, this Pi Zero already has pins soldered in from a previous project, so I plan on adding the GPIO button functionality. 

I'm curious if anyone has any experiences or thouhgts on this before I get started. 

Thanks,
PJ

swyer222, 2018-03-02:
Yes, verified on Pi Zero W

pjcoleman73, 2018-03-05:
Excellent! Thank you!

swyer222, 2018-03-23:
In case you want to go down that RPI Zero Route, please bear in mind that the Zero does not offer audio output on the board. So you will need to output audio either over USB or (as I did) using a soundcard. I've used the Hifiberry miniamp, which allows you to drive the speakers directly. You will also need to follow these instructions: https://www.hifiberry.com/build/documentation/adding-software-volume-control/ to ensure that you can set volume levels in alsamixer. Out of the box, the volume was unbearably loud on my jukebox.

ghostdex, 2018-05-15:
Did someone already try to play music over Bluetooth (e.g. with pulseaudio) so that there is no need for a soundcard?

MiczFlor, 2018-05-15:
hi @ghostdex and welcome. I haven't tried it yet, but it would be good to get this into the manual as a "howto" section. Possibly open a new "issues" with this title and see if somebody joins the process? As you can see: the "issues" here are more like forum threads than bug reports ;) and the bluetooth issue has no thread yet. Start it, fix it and create a pull request for the documentation and possibly code. Thanks

jacktbio2, 2018-05-16:
Hello, thanks a lot for your work. It's very easy to use it and so cool. I made my jukebox with a raspberry pi zero WT and used a cheaper soundcard. https://www.aliexpress.com/item/Raspberry-Pi-pHAT-Sound-Card-I2S-interface-PCM5102-DAC-Module-24-bit-Audio-Board-With-Stereo/32742608325.html

For  installation of the homemade DAC on a Raspbian, you can use the pHAT DAC's one line installer command:

curl -sS https://get.pimoroni.com/phatdac | bash 

See the link for connection : https://www.hackster.io/shiva-siddharth/audio-dac-hat-for-pi-with-headphone-jack-and-3w-speaker-out-51bad6

Everythink work well.

Stumpf74, 2018-09-07:
Yes, I verfied on the PI Zero without W ;-)
It's works very well.

itready, 2018-09-13:
I also try to set this up with a PI Zero W. How can I connect the RFID-Reader (not USB) to the Raspberry Zero W? Type is RFID-RC522. Is there a how to? How to wire this? I totaly newby. My prototype with a PI 3 and USB RFID Reader did well...

Raspfarbend, 2018-10-08:
Short Update from me:
i currently running phoniebox 1.1.5 on a Pi Zero W with an RC522 Reader and one active USB-Speaker.

Raspfarbend, 2018-10-09:
@vw2audi

You have to set this value „Speaker“ in settings/Audio_ifacename
and in /etc/mpd.cfg search for PCM section Audio Output

I hope this helps

vw2audi, 2018-10-10:
Hi together, 
i got it. Now system is running and i wanted to use my RFID NFC Tags, but with tese it isnt working ;-(
i have here one typical Card with RFID, these works and even can read the UID.

But i also have here some Tags witch give me a Serial No. insted of UID.
Is possible to use them too?

MiczFlor, 2018-10-10:
Hi @vw2audi 
which version of Phoniebox are you running? This can be found in the file `settings/version` or in the web app under info.
There was a code issue with the reader, so you might have that problem. Please take a look here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/231#issuecomment-427675366
and your best bet would be to reinstall Phoniebox by deleting the entire folder `RPi-Jukebox-RFID` and use the one line script for the installation again.

vw2audi, 2018-10-11:
Hi @MiczFlor 
thnaks a lot for the hint. I have Version 1.1.6. is a fresh installation.
If i place the RFID nothing happend. If i place an RFID Card is working .....

But if i use my RFID Reader in my mobile i can read both chip.

marcohorstmann, 2018-10-12:
What kind of cards are it that are not working? The suggested reader here is able to read EM4100 125khz cards.

vw2audi, 2018-10-14:
Ok in these case i have found the difference.
But isnt it possible to use also other chips?

marcohorstmann, 2018-10-15:
Not as far as I know the reader and cards needs the same communication stanard. I had seen on eBay a dual frequency reader but no idea if you get it connected and working:
https://www.ebay.de/itm/Dual-Frequenz-125kHz-und-13-56MHZ-WG26-34-RFID-Lesegeraet-/172666970376

I think buy the working cards would be cheaper and less time consuming

closed 2020-01-18 by MiczFlor