MiczFlor/RPi-Jukebox-RFID #16: Delay Cron running scripts

swyer222, 2018-03-02:
Oh boy!! This took me forever to figure out - yes, I'm a raspi noob. Anyway documenting here, so no one else wastes time on this. 

Issue: daemon_rfid_reader.py worked perfectly when running through SSH manually. However, when running at reboot, it just didn't play the audio files when triggered by RFID tag. It turns out that cron ran them too early in the boot process.

Solution: Delay running the script by 60 secs.
@reboot **sleep 60 &&** mpg123 /home/pi/RPi-Jukebox-RFID/misc/startupsound.mp3
@reboot **sleep 60 &&** python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py &

tullm, 2018-03-06:
HI,

I just had a problem that the daemon started, but was closed after some time.

so I created a shell script named "chk_daemon_rfid_reader.sh"

with this in it:
_

> # check rfid_reader_daemon
> ps -ef | grep -v grep | grep python2
> # if not found - equals to 1, start it
> if [ $? -eq 1 ]
> then
> python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py &
> else
> echo "eq 0 - rfid_reader_daemon found - do nothing"

_

So then I entered this script in the crontab to check every 30 seconds for the deamon an start it if not running.
crontab:
_

> * * * * * /home/pi/RPi-Jukebox-RFID/scripts/chk_daemon_rfid_reader.sh &
> * * * * * ( sleep 30 ; /home/pi/RPi-Jukebox-RFID/scripts/chk_daemon_rfid_reader.sh & )

_

Maybe I will try to set up from scratch, but everything is running nice and smooth now. so I will leave it for now.

cheers

MiczFlor, 2018-03-08:
Hi @tullm @swyer222 , I will pull this into a tips and tricks and hacks section in the INSTALL and MANUAL files. Thanks for your input.

MiczFlor, 2018-03-26:
@tullm @swyer222 I added your input in the troubleshooting section. Thanks again!
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md#troubleshooting--faq

closed 2018-03-26 by MiczFlor