MiczFlor/RPi-Jukebox-RFID #17: Install on Ubuntu Mate for the Raspberry Pi

boden, 2018-03-02:
Here just some notes on stuff I had to do to get the software running on Ubuntu Mate for Raspberry Pi instead of Raspbian. Maybe someone else will find it helpful.

- PHP has to be installed with `sudo apt install php php-cgi` instead of the command given in the docs. This will install PHP7, which seems to work flawlessly.
- As on Mate the username is set during the install, it's usually differet from the common `pi` user on Raspbian. I hence had to change `$conf['base_path']` in `./htdocs/config.php` to match my home folder.
- I could not get auto login for my user working via `raspi-config` as recommended in the docs. As a workaround, I added autostart of the demon to `/etc/crontab` instead, which works (almost) the same way.
- Last but not least, I somehow have to run the demon with sudo to have sound. Probably just a missing group membership or something like that, but I have yet to investigate.

So far everthing works great, thanks again for sharing! :)

MiczFlor, 2018-03-08:
Thank you @boden , I plan to add a couple of tips and tricks to the INSTALL and CONFIGURE files. I will include this, too.

MiczFlor, 2018-03-26:
@boden I added your short howto here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/INSTALL.md#installing-on-ubuntu-mate-for-raspberry-pi

closed 2018-03-26 by MiczFlor