MiczFlor/RPi-Jukebox-RFID #18: Play Spotify

infomobi, 2018-03-09:
Hello,

thanks for the great work.
Will it be possible to play songs or playlists from Spotify premium?

Thanks four your help!

tullm, 2018-03-09:
Hi.
Yes this is possible. I've done it. 
I've installed mopidy which included mpd. 
I can trigger to play spoitfy content over mpd with the swipe of a RFID tag.
So install mopidy with the Spotify extension. When this is done and you are able to play Spotify over mopidy it is only some editing of the script here to get it working with RFID. I can share what I've edited.
Cheers

infomobi, 2018-03-10:
Thanks for your answer.
Would it be possible to make a little tutorial with all steps that I have to do (similar to the install guide of RPI-Jukebox-RFID)?
That would be very helpful :-)

tullm, 2018-03-10:
Hi. Don't remember it step by step, but have a look here: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/5#issuecomment-358655801 there is a link to the tutorial that I used to install it.
Good luck...

infomobi, 2018-03-10:
OK, installation is not the problem with the tutorial.
But how can I configure triggering a playlist or album with RFID?

tullm, 2018-03-11:
Give me a day or two for the answer. Will get back to you here...

tullm, 2018-03-12:
So,

if you have installed mopidy and spotify premium ist working for you you can go on like that:

I've crated a variable in the "rfid_trigger_play.sh" under /home/pi/RPi-Jukebox-RFID/scripts like that:

`CMDMUSIK1="0001234567` (the number represents the number from the rfid card.

Then i added another command (right under all the other commands)in the same script like that:
`elif [ "$CARDID" == "$CMDMUSIK1" ]`
`then`
`    # stop VLC and Start SpotifyPlaylist Musik1`
`    mpc -p 6681 stop && mpc -p 6681 clear && mpc -p 6681 load Musik1 && mpc -p 6681 volume 80 && mpc -p 6681 play && sudo pkill vlc
` <- this whole mpc stuff is one line

So when you swipe the card 0001234567 it detects the special card which triggers the command. the commant stops mpc if it is playing, clears the current playlist, loads the Playlist Musik1 from spotify, sets volume to 80%,starts playing and kills vlc. You can load all playlists here by name. Or with the us of the uri url from spotify. 

All command available are listed here: https://linux.die.net/man/1/mpc 

I also made it possible to trigger paue, next and so on by adding these command to the normal command within the script like that:

`elif [ "$CARDID" == "$CMDPAUSE" ]`
`then`
`    # pause VLC media and mpc`
`    echo "pause" | nc.openbsd -w 1 localhost 4212`
`    mpc -p 6681 toggle`

enjoy.

hope that helps.
tullm

infomobi, 2018-03-14:
Thank you, that helped a lot and now it works fine.

I made a little addition to the code to make everything a little more flexible. I can add new spotify playlists to the jukebox without editing the script. I just have to create a new audiofolder with a file named playlist.spotify. The content of this file is the name of the playlist.

Here is the code in rfid_trigger_play.sh:

```
        if [ -f $PATHDATA/../shared/audiofolders/$FOLDERNAME/playlist.spotify ]
        then
                playlistname=`cat $PATHDATA/../shared/audiofolders/$FOLDERNAME/playlist.spotify`
                sudo pkill vlc && mpc clear && sleep .1 && mpc load $playlistname && sleep .1 && mpc play
        else
```
You just have to insert it in the right place:
```
    # if a folder $FOLDERNAME exists, play content
    if [ -d $PATHDATA/../shared/audiofolders/$FOLDERNAME ]
    then
INSERT HERE
                # create an empty string for the playlist
                PLAYLIST=""
```
Again, thanks for your help.

PS: Sometimes, spotify stops working on my machine. The logfile shows that loading a playlist runs into a timeout. Do you have the same problem?

MiczFlor, 2018-03-14:
I am planning to add the `spotify` option to the script in core. However, since I have no spotifiy (...) can somebody give me a brief "install" info here for the initial steps to get the Pi ready for spotify? Or should I just refer to the link posted by @tullm ?
> To add spotify-connect I used: https://github.com/dtcooper/raspotify
> tutorial to install is on github there.

tullm, 2018-03-14:
@MiczFlor Hi, great. But the link you postet is for spotify connect, not for spotify itself.

this ist the right one:

> To add the Mopidy Music Server (for local or networkfiles, spotify and other music services) I used: https://www.mopidy.com/
tutorials:
english: https://maex.me/2017/10/raspberry-pi-as-a-music-server/
german: https://braspi.de/blogs/braspi-blog/raspberry-pi-als-musik-server-mit-mopidy
or the very good documentation from mopidy: https://docs.mopidy.com/en/latest/
   I used extensions: spotify, local-images (cover database for local or network files), iris (webinterface), local files.

If you need help or testing let me know. (also in german if neede ;-) )

@infomobi ohh, that sound very interesting. Will try it. The Foldername ist the cardnumber then?

infomobi, 2018-03-15:
The foldername in audiofolders is an alias like "Karneval" and includes the file playlist.spotify with the name of the playlist as content.
The automatically created folder CARDID in folder shortcuts points to this folder like it does with normal mp3-playlists.

@MiczFlor I can try to write the "step-by-step install" for spotify in the next days. You have to add my extension of your script to rfid_trigger_play.sh then.

princemaxwell, 2018-03-15:
@MiczFlor Hi, Spotify Connect is not needed. I have integrated spotify into my Pi by using Mopidy + MPC.
To get this working i created two options: Normal mode and Shuffle Mode.
I need one file in my audiofolder: "spotify.txt" or "spotifyshuffle.txt". Into that file i have to put the URL of the spotify playlist.

PS:
I made some improvements to the webinterface too:
-Play button is working for spotify
-Playing song is shown
-Covers integrated to the UI
-Added + and - buttons for Volume
-Added a filemanager to directly edit/delete the shortcuts and upload/edit/delete files to audiofolders
-Added direct links to the UI for editing/deleting shortcuts and audiofolders
-Some improvements for user friendlyness

Other improvements to rfid_trigger_play.sh:
-Added some association options to cards (shuffle, reboot, volume up, volume down ....)
-spotify integration
-shuffle for VLC and spotify
-automatic audiofolder creation (if shortcut points to an audiofolder which doesn't exist, it will be automaticalle created by reading the card)

MiczFlor, 2018-03-15:
hi @princemaxwell thanks for your input. If you were to share your work here or as a pull request, I would like to review and include it, if that is possible? And please don't forget to post a picture of your jukebox in the other thread to be included in the image gallery ;)

princemaxwell, 2018-03-15:
@MiczFlor sure, thats possible.
i will send you the files on evening...

infomobi, 2018-03-17:
Would you publish the files for testing here @MiczFlor and @princemaxwell?
I would love to test everything.

princemaxwell, 2018-03-17:
I have to remove some very individual changes from my code, which are only affecting my system.
And i have to add some comments...

MiczFlor, 2018-03-18:
@princemaxwell ok, thanks for the feedback. Looking forward to the code.

infomobi, 2018-04-25:
@princemaxwell Any progress here? I would love to test everything. Thanks!

MiczFlor, 2018-08-17:
Hi @nlssttzl 
Good point. This thread needs to go into the documentation for the branch with the old vlc code. 
And if nobody else ventures into the new mpd version, I might need to get a Spotify account to figure   out. It will take a few weeks. Busy with consolidating a few improvemebts and other tickets. 
But thanks for reminding me. It will come. 
All the best, micz

nlssttzl, 2018-08-18:
Actually this shouldnt be too much of a problem since all the commands still work. So via console it ia still possible to steer Spotify. I should even get easier since soy just have to worry about mpc and not vlc. 
Just have to figure out where and how to integrate it into your script and start playing once the card is being swiped. 
Too bad I’m a really bad at programming. Maybe I’ll spend 1-2 evenings figuring out and trying around. However I would be glad if someone shares their experience.

nlssttzl, 2018-08-19:
So I`m back from bouncing some ideas back and forth and I managed to get it running. Unfortunatly not with the more easy version of @infomobi but nevertheless, better than nothing. It should be possible to include the whole play/stop as well, but here finally my noob-skills are too noob.
In addition I did not manage yet to stop spotify when playing a "regular" card.
Well here's what I got:

First link a command to the card swiped in the settings/rfid_trigger_play.conf:
`cmdplayspotify="0002763033"`
Whereas the number matches the RFID-tag.

After that create in the scripts/rfid_trigger_play.sh in the section where all the case commands are (after: `case $CARDID in`

`       $cmdplayspotify)
         mpc stop ;
         mpc clear;
         mpc -p 6601 stop;
         mpc -p 6601 clear;
         mpc -p 6601 load auto && mpc -p 6601 play
        ;;
`
In that case "auto" is my spotify list. That command stops all "regular cards" and all running "spotify cards". 6601 is in my case the port which is being defined in the mopidy.conf

That's it. Not much but in a first step it's running. Next step now would be to include start/stop and to so playing spotify once a "regular card" is being swiped.

no-banker, 2018-09-18:
First thanks a lot for the great work and the great idea! My first steps are done but now I want to integrate Spotify and I read this post. 

@MiczFlor If I understood right you want to integrate the Spotify support in one of the next versions. You have an idea when you start with the implementation? I ask because If you are right now working on it I don’t start with the dirty version if the integration is already in progress. If you need help by testing this feature and if I’m able to help feel free to contact me.

MiczFlor, 2018-09-19:
Hi @no-banker 
thanks for offering support. A "working hack" is all we need at this stage for the latest phoniebox version. 
How would you approach the issue?
If you get started on this, could you have a look at http://tizonia.org/ ?
My thoughts were to use this software as part of phoniebox. It it works, I would then add it to the install script.

no-banker, 2018-09-21:
Hi @MiczFlor 

Tonight I had time to test with tizonia. Seems to be a propper solution. But you have to install several tools on top:

`pip install spotipy

sudo apt-get install pulseaudio`

I have tried to do everything without config file in tizonia by using the following command

`tizonia --spotify-user myusername --spotify-password mypwd --spotify-playlist nameofplaylist`

Afterwards you can steer with the keyboard:

Keyboard control:

   [p] skip to previous file.
   [n] skip to next file.
   [SPACE] pause playback.
   [+/-] increase/decrease volume.
   [m] mute.
   [q] quit.

These are all options for spotify which you can use:

Spotify options (Spotify Premium required):
  --spotify-user arg     Spotify user name  (not required if provided via 
                         config file).
  --spotify-password arg Spotify user password  (not required if provided via 
                         config file).
  --spotify-owner arg    The owner of the playlist  (optional: may be used in 
                         conjunction with --spotify-playlist).
  --spotify-tracks arg   Search and play from Spotify by track name.
  --spotify-artist arg   Search and play from Spotify by artist name.
  --spotify-album arg    Search and play from Spotify by album name.
  --spotify-playlist arg Search and play public playlists (owner is assumed 
                         current user, unless --spotify-owner is provided).

Hopefully you can integrate it with this information? Sorry but I don’t have the skills to write the code.

bgraeb, 2018-09-23:
From what I can see, tizonia would not help at what we want to achieve, as it has no headless mode (right now). 

I did a basic implementation with `mopidy`, but I had no time to integrate it.

This starts a playlist from Spotify:

```python
from mpd import MPDClient

client = MPDClient()
client.timeout = 200
client.idletimeout = None
client.connect("localhost", 6600)
client.add("spotify:user:d5l7k1dqq59jr1qrtdlicykiq:playlist:1zVlcCYZ4BII9D7KMjceJv")
client.play()
```

As you can see, you can run this headless. One could use `sys` to read from stdin, as far as I can remember.

You need a registration at Spotify to create a authentication token: [See here](https://github.com/mopidy/mopidy-spotify).

I need to upgrade my current installation. When I'm done, I'll take a look If I can integrate it (haven't had time to look into this project for a while - dunno how far you got).

Alwiz79, 2018-09-23:
@bgraeb this seems to be in general a pretty interesting thread you linked there.

https://github.com/mopidy/mopidy-spotify/issues/182#issuecomment-410465484

I also think that Tizonia won’t help us. In the above thread someone mentions he uses NFC cards that hold a Spotify uri to be played and passes that on to mopidy :-)
And your first trials look good.
I’ll return from holidays in around two weeks, will try to help then if it’s not solved until then

MiczFlor, 2018-09-24:
Hi @bgraeb @no-banker @Alwiz79 
looking forward to get a Spotify account to test your pull requests :)
My hope for tiziona was that it would be a neat wrapper around not only Spotify but Google Music and other services. The issue at hand: is it "truely" headless. I haven't tried it.
Please take a look also at an older thread: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/118#issuecomment-410885541
There is some information on Spotify - which, as you can imagine without an account (yet!) - I haven't tested.
Keep the good thoughts and contributions coming!
all the best, micz

Alwiz79, 2018-09-26:
@bgraeb did you install mopidy in _parallel_ to the existing jukebox project or _instead_ for this basic implementation?

I am asking because:
I installed mopidy in _parallel_. When I want to start mopidy I get an error message saying:
`Frontend (MpdFrontend) initialization error: MPD server startup failed: [Errno 98] Address already in use`
That is probably because installing in parallel is a stupid idea? Not sure, too noob for that.

anyhow it also says: "Logged in to Spotify in online mode" so..I guess the first step is done :-)

bgraeb, 2018-09-26:
The error you see is, because mpd is a server component that we launch and use. So, it starts and tries to register a listener on your system. If there is already a listener on your system on that port, you'll see this error. So, no panic.

Maybe, I'll get some time this week to do an implementation of the Spotify part.

Alwiz79, 2018-10-02:
@MiczFlor just signed up for spotify premium account. But now I have to admit, I don't know how to move on. If you need access to premium account, let me know (PM?).

bgraeb, 2018-10-03:
@Alwiz79: Next step would be, getting an [authentication token](
https://github.com/mopidy/mopidy-spotify). With that you can automate the playback of Spotify with mpd.

Then, the playback of a Spotify playlist must be integrated. From what I have seen (buying a house breaks all your plans regarding hobbies and development  :-/ ): 

- they need to be recognized as a playlist
- The app should create a config from it
- the config must be used to playback the playlist

Alwiz79, 2018-10-04:
got that authentication token, and as mentioned before: when i start mopidy it says:
"Logged in to Spotify in online mode"
..besides the error message.
I guess i simply don't understand (enough) how to control mopidy..does this thing need some kind of client? or command line input? i'm totally confused about that parallel installtion and how to move on

@bgraeb: if you think buying a house brings chaos to your life, wait for children :-)...but after a few years you will have time for your hobbies again (look at me)...anyhow i don't want to wait years for the spotify integration :-)

Alwiz79, 2018-10-05:
I found the foloowing issue:
[https://github.com/mopidy/mopidy-spotify/issues/140](https://github.com/mopidy/mopidy-spotify/issues/140)
This might be the reason, why I didn't manage to load a playlist via mpc command line

princemaxwell, 2018-10-07:
I checked Spotify integration for Version 1.1.4 after i succesfully integrated in the old VLC Version < 0.94.
The integration with mopidy was very easy for the old VLC version, but i got many problems for version 1.1.4 because it uses MPD on port 6600. Mopidy uses MPD on port 6600 by default too. I could change the port to 6601, so that two MPD servers are running.

There are only 5 steps to get this running. The syntax for MPC is then "mpc -h 0.0.0.0 -p 6601 add spotify:album:###"

Before that i told the webinterface of phoniebox, that type Spotify has to create a spotify.txt with the spotify URL in it.

My problem at the moment is, to get this spotify url from spotify.txt and start playing by using mpc on port 6601.

In the actual phoniebox code, port 6600 is hardcoded so that there is no easy way to get all the commands working on port 6601.

if someone wants a description, how to get mopidy working including mpc and has the time to get this all together in the code, i will paste my self-written manual here.

princemaxwell, 2018-10-07:
> 
> 
> I found the following issue:
> [mopidy/mopidy-spotify#140](https://github.com/mopidy/mopidy-spotify/issues/140)
> This might be the reason, why I didn't manage to load a playlist via mpc command line.
> 
> And this seems to be an approach to fix it. Need to take a deeper look at it.
> [mopidy/mopidy-spotify#188](https://github.com/mopidy/mopidy-spotify/pull/188)

yes, it's sure.
i tested it in my fully working VLC version with spotify.
Tracks, Albums are playing, but playlists aren't working anymore. In April 2018 they worked, and i have not changed anything since then...

We have to hope, that there is a workaround for mopidy-spotify...

Alwiz79, 2018-10-07:
But there seems to be a working Fix for that.
I just don’t know how to implement this (yet).

And this seems to be an approach to fix it. Need to take a deeper look at it.
mopidy/mopidy-spotify#188

Alwiz79, 2018-10-09:
can anybody who has the coding skills please take a look at the spotify integration of playlists. I'm really stuck here because of my lack of coding skills, but i urgently need spotify to  be running (birthday present of the phoniebox).....

kathateich, 2018-10-10:
The project https://github.com/fsahli/music-cards supports Spotify as well as local files. I don't know python good enough to say if there are ideas that are helpful for this Jukebox. But maybe someone else knows?!
I am interested in support for Spotify, too.

MiczFlor, 2018-10-10:
Hi @kathateich 
the code of this project has not been touched in three years, for all I know. I have to check again, but I am sure it will run into the same problems listed in this thread. The problems come from the lack of support by Spotify to their API which would allow to get info and sound from their platform into other platforms.

I feel with you guys (like @Alwiz79 put it in a few posts above) and would like to dedicate time to this. From experience I know that opening this can of worms requires an entire weekend of work. Firstly to make it work (might only take a few hours) to fixing all the triggered glitches and problems it will cause (takes three times the original time put in) to finally documenting it (another hour). And then merging it - and catching the debris of the changes for the coming days, posted in these threads here.

This is how it works and it is a good process. My hope is somebody else makes the first steps and I can focus on documentation and bug fixing :) simply because right now I don't have the time for this hike and focus on smaller walks.

princemaxwell, 2018-10-11:
> 
> 
> But there seems to be a working Fix for that.
> I just don’t know how to implement this (yet).
> 
> And this seems to be an approach to fix it. Need to take a deeper look at it.
> [mopidy/mopidy-spotify#188](https://github.com/mopidy/mopidy-spotify/pull/188)

i took a deeper look at this fix.
i might fix the playlist problems for web API and web UI but not for playing spotify playlist over MPC.

But MPC is what we need for handling with @MiczFlor Phoniebox and RFID cards.

BTW: mopidy-spotify is not needed for playing song, artists or albums with MPC.
i removed it and made a reboot. after that i tried playing albums with MPC and it just worked. Without having mopidy-spotify installed. So, thats the reason, why the fix doesn't take effect for MPC.

when i type a command like this:

"mpc clear"
"mpc add "spotify:artist:0oLWXBuoO0namCRC2oji0q"
"mpc play"

this works very fine!

the only thing, whats not working:

"mpc add "spotify:playlist:37i9dQZF1DWUVpAXiEPK8P"
error adding spotify:playlist:37i9dQZF1DWUVpAXiEPK8P: directory or file not found

i don't have an idea, how MPC is interpeting the spotify links.
and i habe no idea, what package has to be updated to understand playlists.
until april 2018 it worked fine...

Alwiz79, 2018-10-11:
@princemaxwell there are definitly users reporting, that a fix introduced by @BlackLight works,
He has an unofficial fork for mopidy-spotify.
Check the installation guide here: 
[https://github.com/mopidy/mopidy-spotify/issues/182#issuecomment-412370556](https://github.com/mopidy/mopidy-spotify/issues/182#issuecomment-412370556)

Anyhow I could not manage to load playlists. Maybe I simply used the syntax wrong, not sure.
As soon as i load a playlist with the iris frontend, i can then play/pause/stop with the mpc command line.

I tried add and load with play list name, https-link, uri format...nothing. but, as mentioned before there are people who were succesfull

princemaxwell, 2018-10-11:
I installed the fork from @BlackLight.
Mopidy-Spotify package is NOT needed for MPC command line. I removed it and MPC still worked with
MPC
-play
-add (Spotify:artist/album:###)
-clear
-etc...

Have you tried my syntax for artist link above?
Their you can test syntax.

For the moment only albums, tracks and artists  are possible for phoniebox...

Alwiz79, 2018-10-11:
@princemaxwell I'm not so quite sure.
I had the original mopidy-spotify extension installed. In addition I installed the Iris frontend to have a nice frontend to play with.
My spotify playlists worked with that (no Blacklight fork installed yet).
But command line wasnt working for playlists.

then I installed the Blacklight fork, but it still didn't work as mentioned above. So, I'm not sure that this fork is needed for the web UI (?)

I now tried the following:
mpc -h 192.168.178.58 -P mopidy -p 6681 add spotify:album:5KD3xJ04ta4XyrBaqpOwe2
mpc -h 192.168.178.58 -P mopidy -p 6681 play
It's not working here. Mpc still plays the last playlist that I loaded with the web UI Iris

princemaxwell, 2018-10-11:
You have to clear first. Otherwise you add the new album at the end of your existing list.

mpc -h 192.168.178.58 -P mopidy -p 6681 clear

Then add and play like above. That looked good...

Alwiz79, 2018-10-12:
@princemaxwell I can now confirmwhat you said :-(
after clear, add, play I can play albums, artists etc, but I cannot load a playlist.
But I'm still convinced that I am / we are doing something wrong.
See the comment here:
[https://github.com/mopidy/mopidy-spotify/issues/182#issuecomment-392593817](https://github.com/mopidy/mopidy-spotify/issues/182#issuecomment-392593817)
I interpret this as: the fix should also be working for mpc command line..

BTW: is there a difference between user created playlist, and the spotify own playlists. like the ones for one special music genre e.g.?

kathateich, 2018-10-13:
> @princemaxwell <https://github.com/princemaxwell> @MiczFlor
> ok, here is the solution that finally working for me. just a rough summary
> today, as i am going on holiday now.
>
>    1.
>
>    install mopidy in parallel to phoniebox
>    2.
>
>    configure the porst (see above)
>    3.
>
>    install the mopidy-spotify fork from Blacklight
>    mopidy/mopidy-spotify#182 (comment)
>    4.
>
>    use the following syntax:
>    mpc -h 192.168.178.58 -P mopidy -p 6681 clear
>    mpc -h 192.168.178.58 -P mopidy -p 6681 add
>    spotify:user:USER_ID:playlist:PLAYLIST_ID not
>    spotify:playlist:PLAYLIST_ID
>    mpc -h 192.168.178.58 -P mopidy -p 6681 play
>
> mopidy/mopidy-spotify#182 (comment)
>
> Now, this has to be merged to the phoniebox code, card swipe etc :-)

@alwiz
I cannot get mopidy work, i.e. when I run mpc via ssh I get error messages
that mpd refuses connection. This always appears after I used mpc add for
the first time.
Would you help me, please?
Do you run it as a service or "normal"?
How do your [http] and [mpd]-sections look like? I guess that somewhere
there is my fault.
Thanks a lot.

Alwiz79, 2018-10-13:
@kathateich sorry i can’t copy my sections here - i just left for a vacation this morning.

Anyhow, I’m running mopidy as a service. The configuration file when running as a service is on a different location (compare one of my earlier comments above).

princemaxwell, 2018-10-13:
@Alwiz79 
I got it! You will never believe, what i had to do...
I used an old client_id and client_secret from developer.spotify.com and i had to renew this on mopidy.com/authenticate/ and then playlists are working FINE.

the whole time i wasn't using the web API ;-) since now!

I now have mopidy working well and will now start merging it into Phoniebox.
Have a nice holiday...

Alwiz79, 2018-10-13:
Thats good News! Looking forward to the new Spotify-phoniebox:-)

princemaxwell, 2018-10-14:
@Alwiz79 @MiczFlor 
It's done! I have a first working version of Phoniebox with spotify support ;-)
The best is, that i deactivated MPD, because i preffered to use the MPD of Mopidy, so that not two instances are running. Mopidy MPD is listening on port 6600 by default, so the difference between the master branch and my version are very easy to handle.

I added the following features:
-spotify integration
-cover integration for spotify tracks
-AJAX integration for title, album, artist, time (elapsed and overall)
-playlist with title, album, artist, time
-fix for songs with > 60 minutes. the time now shows hours.
-advanced inc.playerStatus.php

I am testing all the functions now.
I wrote a little step-by-step guide how i got it working ;-)

Alwiz79, 2018-10-14:
That’s great news! Happy to see the result next week :-)

farbschein, 2018-10-14:
Thank you. That's great :-)

MiczFlor, 2018-10-15:
Hi @princemaxwell 
six months after you solved this already for the VLC version of Phoniebox, now *you did it again* :) And thanks to @Alwiz79 to open the hood of the car, too on this issue.
I am waiting for the pull requests and then will merge it with the develop branch for testing. The Spotifiers in the Phoniebox universe: stay tuned for testing!

princemaxwell, 2018-10-16:
@MiczFlor 
I have made a pull request. You get my installation step-by-step guide, because i hope you can handle the installerscript integration.

MiczFlor, 2018-10-16:
@princemaxwell 
I merged your work with the `develop` branch. So others can test it, too. I also created a new file for your "howto" information:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/SPOTIFY-INTEGRATION.md
Please add it in there and create a pull request.
I would like this to be with the Phoniebox community asap in the `develop` branch, also the documentation, because (in my experience) the challenge to make this universal starts only now - and will take some time. So the more input and testing, the better. (Some of the older Phonieboxers might remember the *second swipe* discussions :)

petschni, 2018-10-16:
@princemaxwell thanks for the PR - great news! What is the best way to "update" my current installation to the "develop" commit?

princemaxwell, 2018-10-16:
The develop branch is under testing!!
So if you want to test, make a backup of your SD-card and try (!) to use the document: [
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/SPOTIFY-INTEGRATION.md](url)
and please read it first and the look, what steps you need...

farbschein, 2018-10-16:
@princemaxwell  I made a first installation of develop branch but couldn't find where to change the Playlist_Folders_Path variable as mentioned in Couldn't find where to set the Playlist_Folders_Path. Could anyone help?
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/SPOTIFY-INTEGRATION.md#change-playlists_folders_path-to
Could you please help?

princemaxwell, 2018-10-16:
Look at your settings folder in your phoniebox installation:
/home/pi/RPi-Jukebox-RFID/settings/

There should be a file named "Playlists_Folders_Path"

Later we put that step into installerscript...

farbschein, 2018-10-16:
Thank you. That worked out. 
But now I'm wondering where to type in the spotify track-url (eg. spotify:track:6ZAQxzVUL20heVoowCy...)
Is the stream-url the right place? And what type should I choose?
Thank you again.

princemaxwell, 2018-10-16:
Have you updated to the develop branch?
Then you should see it when you AddNewStream.
Stream type: Spotify
Stream Url: spotify:(user:username:)album/track/playlist:###

[65 more replies]

closed 2018-11-21 by MiczFlor