MiczFlor/RPi-Jukebox-RFID #19: php5 -> php7

sigmaeo, 2018-03-10:
Great work! :-)
With actual raspbian stretch this line from installation manual doesn't work anymore:
sudo apt-get install php5-common php5-cgi php5 

PHP5 isn't available anymore as installation package.

It should be substituted simply with:
sudo apt-get install php-common php-cgi php 

This installs PHP7 and everything works as it should.

hailogugo, 2018-03-13:
Can you use the playbutton for each dictornery on the Webinterface?

sigmaeo, 2018-03-14:
The same here. I can't use them.
Never tried it before.

sigmaeo, 2018-03-14:
I looked into it. Maybe it's only a typo.
In this button there is in htdocs/index.php written /usr/bin/cvlc instead of /usr/bin/vlc:
I will try it tomorrow - if this repair this issue.

`// play folder with VLC
if(isset($urlparams['play']) && $urlparams['play'] != "" && is_dir($urlparams['play'])) {
    // kill vlc if running
    // NOTE: this is being done as sudo, because the webserver does not have the rights to kill VLC
    $exec = "/usr/bin/sudo pkill vlc > /dev/null 2>/dev/null &";
    exec($exec);
    // pipe playlist into VLC
    // NOTE: this is being done as sudo, because the webserver does not have the rights to start VLC
    $exec = "/usr/bin/sudo /usr/bin/cvlc -I rc --rc-host localhost:4212 ".$urlparams['play']." > /dev/null 2>/dev/null &";
    exec($exec);
    /* redirect to drop all the url parameters */
    header("Location: ".$conf['url_abs']);
    exit; 
}`

MiczFlor, 2018-03-14:
`cvlc` is the original command line interface for `vlc`. Not sure if the `c` is still needed today. I am planning to add the --no-video option to the script, just to be sure this will only play sound.
https://wiki.videolan.org/Video_Output/

sigmaeo, 2018-03-17:
Yes, you are right of course.
I changed the 2>/dev/null part of the cvlc play command to a logfile and got the following output:

```
[01355ef0] pulse audio output error: PulseAudio server connection failure: Connection refused
[0136eb40] core interface error: no suitable interface module
[012bd0f0] core libvlc error: interface "globalhotkeys,none" initialization failed
[0136eb40] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[0136eb40] core interface error: no suitable interface module
[012bd0f0] core libvlc error: interface "dbus,none" initialization failed
libdvdnav: Using dvdnav version 5.0.3
[0136eb40] [cli] lua interface: Listening on host "localhost:4212".
libdvdread: Encrypted DVD support unavailable.
************************************************
**                                            **
**  No css library available. See             **
**  /usr/share/doc/libdvdread4/README.css     **
**  for more information.                     **
**                                            **
************************************************
libdvdread: Couldn't find device name.
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.IFO failed
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.BUP failed
libdvdread: Can't open file VIDEO_TS.IFO.
libdvdnav: vm: failed to read VIDEO_TS.IFO
[0136eb40] [cli] lua interface error: Error loading script /usr/lib/arm-linux-gnueabihf/vlc/lua/intf/cli.luac: lua/intf/modules/host.lua:279: Interrupted.
```

sigmaeo, 2018-03-17:
I don't understand where the "piping the playlist to vlc" takes place?

The following command manually started as user pi works (without --no-video and with):
´´´
/usr/bin/sudo /usr/bin/cvlc  -I rc --rc-host localhost:4212 file:///home/pi/RPi-Jukebox-RFID/playlists/Disney_1.m3u
´´´

hailogugo, 2018-03-18:
It will be same command, if we use a RFID card.
I am not a programme but my thinking is:
kill cvlc
play folder which is mark

sigmaeo, 2018-03-18:
Sorry, I made yesterday a mistake in testing your suggestion with the "--no-video" option.
I checked it again today and now everything works (with this additional option and not without)!
:-)
It seems that this option is needed. Now the function in index.php looks like this:
```
// play folder with VLC
if(isset($urlparams['play']) && $urlparams['play'] != "" && is_dir($urlparams['play'])) {
    // kill vlc if running
    // NOTE: this is being done as sudo, because the webserver does not have the rights to kill VLC
    $exec = "/usr/bin/sudo pkill vlc > /dev/null 2>/dev/null &";
    exec($exec);

    // pipe playlist into VLC
    // NOTE: this is being done as sudo, because the webserver does not have the rights to start VLC
    $exec = "/usr/bin/sudo /usr/bin/cvlc --no-video -I rc --rc-host localhost:4212 ".$urlparams['play']." > /dev/null 2>/dev/null &";
    //$exec = "/usr/bin/sudo /usr/bin/cvlc --no-video -I rc --rc-host localhost:4212 ".$urlparams['play']." > /home/pi/vlc1.txt 2>/home/pi/vlc2.txt &";
    exec($exec);
    /* redirect to drop all the url parameters */
    header("Location: ".$conf['url_abs']);
    exit;
}
```

closed 2018-03-18 by MiczFlor