MiczFlor/RPi-Jukebox-RFID #20: Webinterface Sound Controls not working with USB Sound

svo08, 2018-03-16:
Hi,

first, thanks for that great project :-)

Im using a external USB Soundcard (Soundblaster Play).
I did manage to geht the Card working.

Everything is working fine except the volume control via webinterface.

I discoverd that i have to use different command to adjust Volume
amixer -D hw:U0x41e0x30d3 sset Speaker 10%

I was able to adjust the Control Script but i do not know what and where i have to adjust the files for the webinterface to get it to work there as well.
Think it has to be adjustet like i did it in the control script.
Can you please help me.

Thanks

svo08, 2018-03-17:
Sorry, done i did found it index.php.
I adjusted the line and volume control is working.

MiczFlor, 2018-03-18:
@svo08 can you please post your solution here, so I can include it in the code. I was thinking of adding this for HiFiBerry anyway. So this could be a good addition.

svo08, 2018-03-20:
Hi, not sure if it is possible to add the volume command to config file.
Then there will be a single place, single command to replace.

As i wrote in the post im using a different sound card (USB One).
I was able to get it work with description in the docs.
$ cat /proc/asound/modules 
There is a way to adress with amixer the HW Card.
So "amixer sset ´PCM´ 100%" was not working but adressing sound card with 
"amixer -D hw:U0x41e0x30d3 sset Speaker 100%"

So i had to replace the string amixer sset ´PCM´ with "amixer -D hw:U0x41e0x30d3 sset Speaker"
I did this in index.php in htdocs and in the control script rfid_trigger_play.sh

MiczFlor, 2018-03-26:
hm hm hm. Thinking how this could be a global setting... Any thought? You mention if should be in the config file, but that is limited to PHP. This also requires changes in the bash script (as you also point out).

MiczFlor, 2018-04-16:
@svo08 I added a new file that is serving the web interface and the bash script:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/playout_controls.sh.sample
Possibly this might help?

closed 2018-04-16 by MiczFlor