MiczFlor/RPi-Jukebox-RFID #23: Resume folder instead of restarting

Bockiii, 2018-03-19:
Hi,

first off: love this! I build a jukebox for my little son and it's great :)

I have an enhancement request: Resume a playlist/folder instead of restarting it. A restart could be done on second scan.

I would like to use this for Audiobooks, but it's annoying that it starts from the beginning everytime. I already looked into the (horribly extensive) vlc man file but couldnt find an option to "resume" a previously played playlist, so we would probably need to do a workaround for that. Something like:
Create a file within the folder that will note down progress within that folder and that gets read by the script if the folder is started again. The "new card" event could check for running vlc, get the current playstate and log that out into that file. I'll try to experiment a bit with that and may do a pull request, but maybe you have that in your backlog already :)

Bockiii, 2018-03-23:
Hi,

I researched a bit on this during the last days and will post my findings:
- VLC does have an included "resume" feature, controllable in the vlc settings. Problem is, this is resuming "per file", not per playlist. Example: Playlist with 5 entries (episodes of a play). Play the playlist until half of the second track. When you start the playlist again, it will play the full first track again (becaues that's "done" in the previous run) and then jump to the half of the second track. So, unuseable for us.
- There might be a way to build the logic yourself with interface commands (playlist, title and get_time in conjunction for example), but that would mean a probe every few seconds (probe the vlc for those three commands, save to file, probe again etc. at the start of the next playlist run, check if such a file is there and jump to that song (goto index and seek). But that is quite a lot of interface stuff with little accuracy. During testing I found that the -rc interface you use is actually the "legacy" one and there is a newer lua one that does everything the rc does plus more. You can test that ( -I luaintf --lua-intf cli ).
- Addons and Lua scripts. I found a LUA script that fits well and could be used either completely or as a base : https://addons.videolan.org/p/1154006/ . The only problem is: I could not find a way how to start a LUA script on command line. All LUA scripts have to be activated in the QT interface under "view" (at least as much as I found out). What does work though, are actual addons (dlls). Those are used, even with rc/lua interface options. I found some that did approximately the correct thing, but nothing that did exactly what I wanted.

I'm open for suggestions :)

Bockiii, 2018-03-25:
I've written my own solution. Open for discussion if the functionality should be configurable or changed.
#24

MiczFlor, 2018-03-26:
Hi @Bockiii I commented on the pull request. Thanks for the contribution. Let's move the discussion to the pull request thread. All the best, micz

closed 2018-03-26 by MiczFlor