MiczFlor/RPi-Jukebox-RFID #26: unable to register device

d7071, 2018-03-23:
after calling python2 `python2 RegisterDevice.py` I get the following:
Traceback (most recent call last):
  File "RegisterDevice.py", line 4, in <module>
    from evdev import InputDevice, list_devices
  File "/usr/local/lib/python2.7/dist-packages/evdev/__init__.py", line 9, in <module>
    from evdev.uinput import UInput, UInputError
  File "/usr/local/lib/python2.7/dist-packages/evdev/uinput.py", line 35
    def from_device(cls, *devices, filtered_types=(ecodes.EV_SYN, ecodes.EV_FF), **kwargs):

Takeit-Ge, 2018-03-23:
Exactly the same on my side on a Raspberry with Stretch. Just a plain fresh install and it gives me:

pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ python2 RegisterDevice.py
Traceback (most recent call last):
  File "RegisterDevice.py", line 4, in <module>
    from evdev import InputDevice, list_devices
  File "/usr/local/lib/python2.7/dist-packages/evdev/__init__.py", line 9, in <module>
    from evdev.uinput import UInput, UInputError
  File "/usr/local/lib/python2.7/dist-packages/evdev/uinput.py", line 35
    def from_device(cls, *devices, filtered_types=(ecodes.EV_SYN, ecodes.EV_FF), **kwargs):
                                                ^
SyntaxError: invalid syntax

d7071, 2018-03-24:
I figured out, that evdev just recenlty published Version 0.0.8. I fored python to install evdev 0.07. 
Now I got no errors but the cards won't show it's ID.

MiczFlor, 2018-03-26:
@d7071 thanks for the input. What changes did you need to do to run it on Stretch? I haven't made a migration yet to the new system, but I guess this code would benefit from an update to the latest system. If you give me some pointers, I will give it a try. Thanks, micz

d7071, 2018-03-26:
Yes I ran it on strecht
I believe I used sudo pip install evdev -version="0.0.7" , version 0.08 was released some days before.

Like I said, when I use the RFID card there is no ID in the text file.

MiczFlor, 2018-03-26:
Hi @d7071 

`sudo pip install evdev -version="0.0.7"` will assume "verbose" and gives me the error: "rsion=0.0.7 should either be a path to a local project or a VCS url beginning with svn+, git+, hg+, or bzr+"

The option `--version` shows the version and quits. So that is no help either.

Possibly you forced the version via `--install-option <options>` (Extra arguments to be supplied to the setup.py install command (use like --install-option="--install-scripts=/usr/local/bin"). 

?

MiczFlor, 2018-04-14:
Hi @d7071 
did you figure out how you specified the version of evdev on install? I think it would need to be version 0.7.0 - at least that is the version that is listed as 'latest' on github: https://github.com/gvalkov/python-evdev/releases

MiczFlor, 2018-04-14:
Please @d7071 have a quick check if this is how you made it work:
~~~
sudo pip install "evdev == 0.7.0"
~~~

closed 2018-04-16 by MiczFlor

d7071, 2018-04-19:
I didn't try anything else. After all that I used a complete new R PI and everything went just fine.