MiczFlor/RPi-Jukebox-RFID #27: Use ACR122U as RFID reader

raymonvdm, 2018-03-23:
I`m trying to get an ACR 122U working as RFID cardreader. The reader is seen on the USB port,but it looks like it`t not reporting as HID input device

```
root@testbank:/opt/RPi-Jukebox-RFID/scripts# lsusb | grep ACR
Bus 001 Device 006: ID 072f:2200 Advanced Card Systems, Ltd ACR122U
root@testbank:/opt/RPi-Jukebox-RFID/scripts# 
```
Bus 001 Device 006: ID 072f:2200 Advanced Card Systems, Ltd ACR122U
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x072f Advanced Card Systems, Ltd
  idProduct          0x2200 ACR122U
  bcdDevice            2.07
  iManufacturer           1 ACS
  iProduct                2 ACR122U PICC Interface
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           93
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass        11 Chip/SmartCard
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      ChipCard Interface Descriptor:
        bLength                54
        bDescriptorType        33
        bcdCCID              1.00
        nMaxSlotIndex           0
        bVoltageSupport         7  5.0V 3.0V 1.8V 
        dwProtocols             2  T=1
        dwDefaultClock       4000
        dwMaxiumumClock      4000
        bNumClockSupported      0
        dwDataRate          10752 bps
        dwMaxDataRate      250000 bps
        bNumDataRatesSupp.      0
        dwMaxIFSD             256
        dwSyncProtocols  00000000 
        dwMechanical     00000000 
        dwFeatures       00020040
          Auto parameter negotation made by CCID
          Short APDU level exchange
        dwMaxCCIDMsgLen       271
        bClassGetResponse      00
        bClassEnvelope         00
        wlcdLayout           none
        bPINSupport             0 
        bMaxCCIDBusySlots       1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               2
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)

```

```
Is it still possible to use this reader (since i had it laying arround)

MiczFlor, 2018-03-26:
(I leave this issue open, because I get similar questions quite often)

I can not give any advice how to use this code with this model, because I don't have it here to check if it works :). If you solve it or anybody else has ideas and pointers, feel free to post them in this issue thread.

USB RFID readers usually are set up in a way that they act "like a keyboard". The Chip ID is pushed to the RPi as if you were typing it in a keyboard and pressed "enter". This input is what the python daemon is waiting for and then it pipes the result into the bash script.

raymonvdm, 2018-03-26:
I found a work arround but it is not usable without an X session (display connected) so i gave up on it. 

https://github.com/Sam-Hall/NFC-HID-Emulate

Maybe you can  uses this code for a solution? In the meantime i ordered some MFRC522 readers just to keep the project going :-)

MiczFlor, 2018-04-16:
Hi @raymonvdm , I'll close the issue, because I can't keep up with hardware specific issues and edge cases. Hope you found a solution and got it going.

closed 2018-04-16 by MiczFlor

TheSplatterer, 2026-02-23:
Ok so I found this comment and thought I'd reply to save someone else the time. This was 99% Claude AI writing and helping me along the process, including writing this summary of the changes someone else would need to make. This was all done on the current Version 3 of phoniebox. This has fixed my install but I can't speak to whether it's correct, safe, or good for anyone else. 

## Fix for ACR122U NFC Reader with Phoniebox V3

The ACR122U is not a standard HID keyboard-style reader, so it requires some extra steps to work with Phoniebox V3.

### 1. Blacklist conflicting kernel modules

Create `/etc/modprobe.d/blacklist-nfc.conf` with:
```
blacklist nfc
blacklist pn533_usb
```

These modules grab the device before nfcpy can access it.

### 2. Disable pcscd

Even though pcscd is needed for the ACR122U in other contexts, it conflicts with nfcpy's direct USB access:
```bash
sudo systemctl stop pcscd
sudo systemctl disable pcscd
```

### 3. Run the RFID setup script and select option 6 (Generic NFCPY)
```bash
bash ~/RPi-Jukebox-RFID/installation/components/setup_rfid_reader.sh
```

### 4. Fix the USB initialisation issue in generic_nfcpy.py

The ACR122U consistently fails to initialise on first connection attempt, requiring a USB reset first. Edit `src/jukebox/components/rfid/hardware/generic_nfcpy/generic_nfcpy.py` and replace:
```python
self.clf = nfc.ContactlessFrontend()
self.clf.open(device_path)
```

With:
```python
self.clf = nfc.ContactlessFrontend()
import time
import subprocess
subprocess.run(['sudo', 'usbreset', '072f:2200'], capture_output=True)
time.sleep(2)
for attempt in range(10):
    try:
        if self.clf.open(device_path):
            break
    except Exception as e:
        self._logger.warning(f"Reader not ready, retrying in 3s (attempt {attempt+1}/10): {e}")
    time.sleep(3)
```

This resets the USB device before attempting to connect, then retries up to 10 times if it still fails. Without this, the reader reliably fails on startup and each failed attempt puts the device into a worse USB state.

### 5. Install usbutils if not already present
```bash
sudo apt install usbutils
```

---

Hopefully this saves someone else the same debugging journey!