MiczFlor/RPi-Jukebox-RFID #28: trouble adding user "pi" to samba

nealhmakwana, 2018-03-31:
awesome tutorial! thanks for sharing!
i followed the steps of going into the flie "/etc/samba/smb.conf"
1. i fixed the workgroup and the wins support
2. i added the path found in the document
3. i went to a new line in the file and added user: pi and password: raspberry
4. i saved and exited
5. in the terminal, i entered sudo smbpasswd -a pi, but got the error "command not found"

MiczFlor, 2018-04-03:
Hi @nealhmakwana 

thanks for joining the community. Not sure if this will solve your problem, but please try to install
~~~
sudo apt-get install samba-common-bin
~~~
Please let me know if that fixes the issue. If so, I need to change the manual to include this. However, strange that this is the first time I hear of this issue and never experienced it myself.
What operating system are you using?

nealhmakwana, 2018-04-04:
i realized i forgot to set a home group for the raspberry pi on windows, this fixed the issue, thanks for the response!

thanks
neal makwana

closed 2018-04-09 by MiczFlor