MiczFlor/RPi-Jukebox-RFID #35: latestID.txt file not being created

hoodvisions, 2018-04-15:
Hi, I went through the manuals a few times now and I cannot find what I may have missed. Right now, when I swipe an RFID tag/card all it does is "typing" the ID into the shell (754852442 command not found), but there is no latestID.txt file being created in the shared folder at all. I guess it's something obvious, but my "limited" Linux skills are preventing me from spotting the issue...
Hope someone can help me? Thanks in advance!

MiczFlor, 2018-04-15:
Hi @hoodvisions 

there could be a number of reasons why this isn't working. First question: does the web app work? Can you play your audio files through the web interface?

Secondly, here a list of events that needs to happen in the right order:

1. the reader is [registered properly](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/CONFIGURE.md#register-your-usb-device-for-the-jukebox) so that the python daemon can find it. Note: even if that hasn't happened, the reader will write the number when you swipe a card. So this could be an error (link to the documentation above)
2. the script `daemon_rfid_reader.py` is [launched on boot](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/CONFIGURE.md#auto-start-the-jukebox). Another possible error: if that doesn't work, the swipe will create the numbers in the command line or whereever your cursor is, but the ID is not passed on to the bash script 
3. You need to create the file `rfid_trigger_play.sh` fron the file `rfid_trigger_play.sh.sample` [as described here](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/CONFIGURE.md#copy-the-media-player-script)
4. the script `rfid_trigger_play.sh` is called by `daemon_rfid_reader.py`. This is the script that will then write the `latestID.txt` file. You can try if this script works if you for example type `./rfid_trigger_play.sh -c=1234567890` in the directory `scripts`. This should create the file. with the content `1234567890`.

I know this does not help you. It might give you some pointers to try again, though. Once you did all the installation steps and none of the configuration steps, the reader seems to work, but the system isn't configured to take the card ID and do something with it. If you solve your issue, please tell us here how you did it, because that might help me to improve the documentation.

hoodvisions, 2018-04-15:
Thanks a lot! This will help me a lot, I guess - I will go through it tomorrow morning :) Again, thanks!

MiczFlor, 2018-04-16:
Hm. Not sure how I can diagnose this from the distance. What I am writing now is a bit like swimming in a muddy lake...

Can you post the results of this command:

`cat /home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt`

And then, please start the jukebox, ideally in command line mode, not Raspbian windows manager. Unplug the RFID reader, replug it - if that's what makes the Pi register it, then type the following:

`python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py &`

Swipe a card and see if this will result in any text:

`cat /home/pi/RPi-Jukebox-RFID/shared/latestID.txt`

What would that tell us? That the setup works, the reader is registered and once replugged will create the latestID.txt file. Then you would need to search around why the device is not registered on boot...

hoodvisions, 2018-04-16:
Thanks so much for looking into this. After testing I found out it is actually even "worse". The Barcode Reader ONLY works when I reboot with it unplugged and only plug it in AFTER the reboot. I realized it doesn't matter if I re-plug it while the system is running, it won't show up until I reboot with the device unplugged. The command "cat /home/pi/RPi-Jukebox-RFID/scripts/deviceName.txt" will spill out "Barcode Reader", but unpluggin and replugging still won't work.

I guess I'll have to find out why the device isn't registered on boot elsewhere as apparently it is not an issue associated with the Jukebox RFID stuff :(

Thanks a lot though!

hoodvisions, 2018-04-16:
I am 90% certain it is a power issue. I tried
`dmesg`
and the device will not show up when booted while plugged in which tells me the kernel doesn't even recognize it. However also it will show an "undervoltage error" message only when the device is plugged in and won't show it when it's unplugged, so I assume it must be an issue with the powerbank not providing enough power to the Pi3 with the USB Audio and Reader plugged in...darn. I will close this later today when I have nailed it down to that issue. Thanks again for helping!

hoodvisions, 2018-04-16:
Yes this is the issue. I will have to figure which power bank WILL work :)

closed 2018-04-16 by hoodvisions

MiczFlor, 2018-04-16:
@hoodvisions lesson learned: one person's failure is good knowledge for others. I added this to the troubleshooting section:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/b58fe6a11de021f012c6ede28f938d43ab00245d