MiczFlor/RPi-Jukebox-RFID #37: Let the jukebox run with root mounted read-only

ralle12345, 2018-04-16:
To avoid corruption due to sudden power loss and to reduce write wear on the sd card it would be good to have the system run on a read-only mounted root.

This is not a feature request - I'm already working on this.

MiczFlor, 2018-04-16:
Hi @ralle12345 
thanks for the input, I'll leave this thread open for your suggestions and discussions.

ralle12345, 2018-04-17:
hey @MiczFlor

I started digging around to find out which directories need to be writeable for the jukebox to work. 
It turns out it's just the playlists dir and the shared dir.  My first solution was to mount a different partition of the SD card to media/ and link those directories to subdirectories there like in the tree output below.

```
pi@Jukebox:~/RPi-Jukebox-RFID $ tree -d
.
|
├── media
│   ├── playlists
│   └── shared
│       ├── audiofolders
│       │   ├── audio1
│       │   └── audio2
│       ├── lost+found
│       └── shortcuts
├── playlists -> media/playlists
└── shared -> media/shared
```

While this solution works fine playing the audio files, a git status reports that the directories below shared have been deleted (since shared is now a symlink). 

There is a workaround for this, but I don't know if it would be advisable to go that route: mount the r/w partition to shared and use a tmpfs for the playlists. PRO: easy to do, CON: If the jukebox runs for a while without reboot, the tmpfs could become quite big. As I don't have a jukebox running in real life scenario yet, I don't have a clue how big the playlists dir can become. 

What do you think? Can you think of another solution?

hailogugo, 2018-04-20:
That's nice to read. This was my idea to. I wait for your answers. If I am ready to work on my Version 2 of Jukebox, i will do add it in my Howto.

I find this Threas in german how to cufigurate the read only function

[General Link Howto Read Only](https://kofler.info/raspbian-lite-fuer-den-read-only-betrieb/)

ralle12345, 2018-04-20:
I don't like the idea of the playlist folder which doesn't contain user servicable parts inside the shared folder as well. But how about the idea of moving it into a tmpfs? How big is the folder on a real-world jukebox?

Maybe we could install a cronjob to junk files not accessed for a certain amount of time to keep the tmpfs tidy?

MiczFlor, 2018-04-20:
The playlist folder is not big and only contains text files and no more than the numer of audiofolders on the jukebox. I will look into the possiblity to drop the playlist folder alltogether - but will need to run some tests.

ralle12345, 2018-04-24:
Considering that the playlist folder ist not that big I think I'll go for a tmpfs for it and have it cleaned from old cruft by systemd. I hope to have the pull request early next week.

MiczFlor, 2018-04-24:
Hi @ralle12345 please read this before you start :)

To get rid of the playlist altogether, you need to change only a few lines in this shell script:
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/rfid_trigger_play.sh.sample

At the bottom of the file, the last lines, try to replace this:
~~~
    then
        # write playlist to file using the same name as the folder with ending .m3u
        # wrap $PLAYLIST string in "" to keep line breaks
        find "$PATHDATA/../shared/audiofolders/$FOLDERNAME" -type f | sort -n > "$PATHDATA/../playlists/$FOLDERNAME.m3u"

        # first kill any possible running vlc process => stop playing audio
        sudo pkill vlc

        # now start the command line version of vlc loading the playlist
        # start as a background process (command &) - otherwise the input only works once the playlist finished
        (cvlc --no-video -I rc --rc-host localhost:4212 "$PATHDATA/../playlists/$FOLDERNAME.m3u" &)
        
        # NOTE TO SELF: can we get rid off writing a playlist if we play the folder as we do in the index.php?
        # Currently problem with whitespaces in folder name.
        # Also, keep in mind the use of stream URLs or YouTube in text files
        # (cvlc --no-video -I rc --rc-host localhost:4212 "$PATHDATA/../shared/audiofolders/$FOLDERNAME" > /dev/null 2>/dev/null &)
    fi
fi
~~~
with this
~~~
    then
        # first kill any possible running vlc process => stop playing audio
        sudo pkill vlc

        # now start the command line version of vlc loading the playlist
        # start as a background process (command &) - otherwise the input only works once the playlist finished
         (cvlc --no-video -I rc --rc-host localhost:4212 "$PATHDATA/../shared/audiofolders/$FOLDERNAME" > /dev/null 2>/dev/null &)
    fi
fi
~~~
And because you are working on this, you will also be testing if this solution works :) It will get rid off the playlist creation altogether. But I am not sure it works with webradio URLs or YouTube URLs in text files in the playout folder. That would be awesome if you could test it, too.

Good luck with the work. Make sure you catch all paths to the `shared` folder.

ralle12345, 2018-04-24:
Hi @MiczFlor 

I tested your changes. Local files work fine, MP3 links work as well. 

```
$ cat ../shared/audiofolders/audio3/stream.txt 
http://swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
```
When I put a YouTube URL inside the same file vlc won't play it with neither the above code nor the latest from git.

When I start the vlc commandline generated by the new script, this is that I get:
````
$ /usr/bin/vlc -I dummy --no-video -I rc --rc-host localhost:4212 /home/pi/RPi-Jukebox-RFID/scripts/../shared/audiofolders/audio3
VLC media player 2.2.6 Umbrella (revision 2.2.6-0-g1aae78981c)
[010b5938] pulse audio output error: PulseAudio server connection failure: Connection refused
[010c5160] core interface error: no suitable interface module
[0101e400] core libvlc error: interface "globalhotkeys,none" initialization failed
[010c5160] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[010c5160] core interface error: no suitable interface module
[0101e400] core libvlc error: interface "dbus,none" initialization failed
libdvdnav: Using dvdnav version 5.0.3
[010d0d10] [cli] lua interface: Listening on host "localhost:4212".
libdvdread: Encrypted DVD support unavailable.
************************************************
**                                            **
**  No css library available. See             **
**  /usr/share/doc/libdvdread4/README.css     **
**  for more information.                     **
**                                            **
************************************************
libdvdread: Couldn't find device name.
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.IFO failed
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.BUP failed
libdvdread: Can't open file VIDEO_TS.IFO.
libdvdnav: vm: failed to read VIDEO_TS.IFO
[010b89c8] core playlist: stopping playback
````
vlc just sits there until it is interrupted.  

When I change the commandline to include the file the output changes to show that at least vlc has recognized the youtube link:

````
$ /usr/bin/vlc -I dummy --no-video -I rc --rc-host localhost:4212 /home/pi/RPi-Jukebox-RFID/scripts/../shared/audiofolders/audio3/youtube.txt 
VLC media player 2.2.6 Umbrella (revision 2.2.6-0-g1aae78981c)
[0064c938] pulse audio output error: PulseAudio server connection failure: Connection refused
[00667910] core interface error: no suitable interface module
[005b5400] core libvlc error: interface "globalhotkeys,none" initialization failed
[00667910] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
[00667910] core interface error: no suitable interface module
[005b5400] core libvlc error: interface "dbus,none" initialization failed
[b2004930] es demux error: cannot peek
[b2004930] es demux error: cannot peek
[00667910] [cli] lua interface: Listening on host "localhost:4212".
[b2008038] core demux error: option sub-original-fps does not exist
[b2000510] core input error: no suitable demux module for `file/subtitle:///home/pi/RPi-Jukebox-RFID/scripts/../shared/audiofolders/audio3//youtube.txt'
[0064fa30] core playlist: stopping playback
[b3d41e58] lua demux error: Couldn't process youtube video URL, please check for updates to this script
[0064fa30] core playlist: stopping playback
[0066bec0] http access error: error: HTTP/1.1 403 Forbidden
[0066bec0] http access error: error: HTTP/1.1 403 Forbidden
[b200eb20] core input error: open of `https://r4---sn-oxujvavbox-jbol.googlevideo.com/videoplayback?itag=22&mn=sn-oxujvavbox-jbol%2Csn-5hne6nsk&mm=31%2C29&mt=1524599963&c=WEB&pl=15&dur=385.404&ms=au%2Crdu&initcwndbps=992500&mv=m&sparams=dur%2Cei%2Cid%2Cinitcwndbps%2Cip%2Cipbits%2Citag%2Clmt%2Cmime%2Cmm%2Cmn%2Cms%2Cmv%2Cpl%2Cratebypass%2Crequiressl%2Csource%2Cexpire&source=youtube&fexp=23724337&ratebypass=yes&lmt=1517631893790149&requiressl=yes&key=yt6&ip=89.1.30.128&ei=xIzfWv7_Jo3E1gLOrqGQAQ&expire=1524621604&mime=video%2Fmp4&id=o-ALq9T8o1gJU1gJqAV8pgaWe70dnfivtzFoe3QHWXLIs0&fvip=2&ipbits=0&signature=0D94AB35250C757CF43EB5253D8329BD993D61BD.C9A9C0B578B69989838AC24A1040FF5D660F37101A' failed
[b200eb20] core input error: Your input can't be opened
[b200eb20] core input error: VLC is unable to open the MRL 'https://r4---sn-oxujvavbox-jbol.googlevideo.com/videoplayback?itag=22&mn=sn-oxujvavbox-jbol%2Csn-5hne6nsk&mm=31%2C29&mt=1524599963&c=WEB&pl=15&dur=385.404&ms=au%2Crdu&initcwndbps=992500&mv=m&sparams=dur%2Cei%2Cid%2Cinitcwndbps%2Cip%2Cipbits%2Citag%2Clmt%2Cmime%2Cmm%2Cmn%2Cms%2Cmv%2Cpl%2Cratebypass%2Crequiressl%2Csource%2Cexpire&source=youtube&fexp=23724337&ratebypass=yes&lmt=1517631893790149&requiressl=yes&key=yt6&ip=89.1.30.128&ei=xIzfWv7_Jo3E1gLOrqGQAQ&expire=1524621604&mime=video%2Fmp4&id=o-ALq9T8o1gJU1gJqAV8pgaWe70dnfivtzFoe3QHWXLIs0&fvip=2&ipbits=0&signature=0D94AB35250C757CF43EB5253D8329BD993D61BD.C9A9C0B578B69989838AC24A1040FF5D660F37101A'. Check the log for details.
````
The same error is generated when using the old script, so it doesn't seem to be caused by the changes.

This is with the latest vlc (revision 2.2.6-0-g1aae78981c) from the debian repos.

To avoid misconceptions: The stream.txt was moved out of the directory before testing the youtube.txt.

Luegengladiator, 2018-05-07:
Hi all,
for bigger MP3 Collections mounting is a good way. At the moment I'm playing around with 20 Files and it is annoying. In the working Version, it shoud be possible to use a NAS share or something else for the Audiofiles. It's not a good Idea to store Radio dramas or Audiobooks like "Harry Potter" or "The Hitchhiker's Guide to the Galaxy" on the SD Card. In Germany almost every Router (most of them FritzBox) has a NAS Feature in it...

MiczFlor, 2018-05-09:
Hi @Luegengladiator @ralle12345 
I think a good, hands on documentation on how to set up the Phoniebox with a NAS share would be good. I would not want this to be part of the overall install process. It would complicate matters and I rather see this as an expert's addition. The original core of this repo should be as simple as possible.

MiczFlor, 2018-06-11:
Hi @ralle12345 
do you have a status update on this? The code base has moved on and we might need to discuss what needs to be done to support your quest.
All the best, micz

ralle12345, 2018-06-12:
Hi @MiczFlor 
I got distracted by other projects :(. I will git pull these days and test my draft on a new SD card. After that I'll open a PR for it.

MiczFlor, 2018-06-12:
No rush, just wanted to check in, if you are still in the orbit. We don't have a deadline to meet :)

closed 2018-12-25 by MiczFlor