MiczFlor/RPi-Jukebox-RFID #38: Reading RFID tags fails

boxboo, 2018-04-18:
Hello! 

I followed the installation guide which worked very well for me and I did not get a single error message during the whole installation and configuration process. So to begin with a big THANK YOU to MiczFlor. My kids can't wait using the Jukebox. 

The webserver, external soundcard, samba etc. are working fine and also the connected devices seem to work properly. Music is also playing when using the webserver. IDs are recognized and files are created on the samba share... 

The only (and essential) issue though is that the "translation" of RFID tags into music playback does not work. Like if there is something wrong in the listening process / the setup of the python script. When holding an RFID tag at the device it shows me the ID on the console and on top of that  an error message "-bash : 0005347171: command not found" (In German: "-bash : 0005347171: Kommando nicht gefunden). However I have followed the setup carefully and can confirm the right device (rfid-reader) is being listened to.

Any idea what else I could check?

Many thanks for your help (would be much appreciated!)

Best
Jan

boxboo, 2018-04-18:
I solved my issue which was due to use of Windows WordPad when editing the shortcuts file. It has put a break which was the reason for not working,,, Now all works fine. Excellent! Maybe worth mentioning the trap in the documentation?

MiczFlor, 2018-04-18:
You are right, and so I did:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/8c31dd8bc0f80f574edd93617365c3cbb689b6f1

axelslade, 2018-06-11:
Hi all,

this is a really awesome project. I am so excited to build this Jukebox - but I cannot solve one issue. 
I have installed everything. The WebApp lets me play music and I can hear it. The RFID readers creates the shortcut, lists the latestID. the playout_controls.sh exists.
But when I swipe the card,  no sounds gets played.

In the webapp (after a reload) I see that the folder was selected and it shows the songs which should be playing, but the player does not start (no sound).

Any idea?

MiczFlor, 2018-06-11:
Hi @axelslade 
good to have you on board, let's see if we can get this ship to leave the harbour. Always difficult to diagnose from a distance, but, hey let's try.
* What OS (jessie or stretch) and what hardware?
* Did you take the code via git or did you download the ZIP from the releases?
* What version of Phoniebox are you running? Type `cat /home/pi/RPi-Jukebox-RFID/settings/version`
* When you swipe, are you still in the windows manager Raspbian or did you switch in the autologin / console mode as described here? https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/INSTALL-stretch.md#autologin-after-boot 
* Are you using an external soundcard or the jack or HDMI? Where do you hear the sound coming from when using the web app?
* Can you play the songs from the command line - once you know the ID (as shown in latestID.txt which seems to work - with the command like this inside the scripts folder:
`./rfid_trigger_play.sh --cardid=1234567890` (substitute 1234567890 with cardID)
* Can you play the folder with the command `./rfid_trigger_play.sh --dir=foldername` inside the scripts folder?
That's all I can think off for now. Let's hear what you got. Thanks, micz

axelslade, 2018-06-12:
Hi @MiczFlor - I could not wait and reinstalled the whole pack (also on stretch - before it was jessie). Now it is working perfectly fine... 

Sorry that we cannot analyze, but on the other hand, you can close this thread again!

Thank you

closed 2018-06-13 by MiczFlor