MiczFlor/RPi-Jukebox-RFID #43: Stop on RFID removal

Luegengladiator, 2018-05-03:
Is it possible to stop playback on rfid-remove instead of additional rfid to stop?

MiczFlor, 2018-05-04:
The USB card reader are triggered only once, on entering the RFID card. So the short answer is "no", the long answer would be: "once you figured it out, please share your solution here". It would make a nice feature! Thanks, m

closed 2018-05-04 by MiczFlor

Luegengladiator, 2018-05-04:
Sorry, my fault. Had to read more about RFID first.
But managed it more with a physical hack. From Wikipedia I learned:

Shielding effectiveness depends on the frequency being used. Low-frequency LowFID tags, like those used in implantable devices for humans and pets, are relatively resistant to shielding though thick metal foil will prevent most reads. High frequency HighFID tags (13.56 MHz—smart cards and access badges) are sensitive to shielding and are difficult to read when within a few centimetres of a metal surface. UHF Ultra-HighFID tags (pallets and cartons) are difficult to read when placed within a few millimetres of a metal surface, although their read range is actually increased when they are spaced 2–4 cm from a metal surface due to positive reinforcement of the reflected wave and the incident wave at the tag.

So wraped the "Stop-RFID" in a one-layer Aluminium foil and stuck a magnet on the top of it. laying both on the RFID-Reader. A second magnet on the "Play-RFID" lifts the "Stop-RFID" out of Range due to the Range-reduction of the foil. The reader now reads the nonshielded RFID. Removing the "Play-RFID" drops the shielded "Stop-RFID" in range stoping playback....

MiczFlor, 2018-07-02:
Hi @Luegengladiator 
there is an interesting hack in the issues section:

https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/62

Possibly you want to try if this works for you. It will probably not be part of the "default" codebase, but if it works, I will include it in the documentation.