MiczFlor/RPi-Jukebox-RFID #44: VLC Alternatives like mpv

Luegengladiator, 2018-05-04:
Is it possible to get a different playback tool?
As a commandline tool mpv looks more efficient than vlc.
goodies: --save-position-on-quit can be set via config or commandline. 
Maybe this makes #30 obsolete?

MiczFlor, 2018-05-07:
Hi @Luegengladiator 
I am pondering what would be a good alternative. mpd in the background might be good, but from experience, I know that the kids jukebox is full of mp3 files with poor metadata.
The two main reasons I started with and stuck to VLC was: a) it plays almost everything (incuding YouTube) b) I know it well.
If you are working on an alternative playout, I would be interested to hear and possibly this could be a settings issue where - on install - the user can pick their system that works best for them. If you integrate spotify, then VLC is of no help anyway...
In the control bash scripts, most of the controls could even co-exist peacefully. No harm done if mpd and VLC are stopped even though only one is running.
All the best, micz

Luegengladiator, 2018-05-07:
working on that for a while now. Using mpv (not mpd) works like a charm since yesterday.
I created a fork and will upload all my settings. 
At the Moment I'm working on a reset option for Files and one for m3u. If I get this working I'll create a pull request.
Thursday is a Holiday here so I hope to get this working until Weekend.

MiczFlor, 2018-05-07:
Hi @Luegengladiator 
Curious to see your solution. Take your time to make it bullet proof and possibly add the option to select in a settings file. I added a folder `settings` which contains nothing yet. The idea would be to have text files with strings in there which are simple to read from bash or php (or pyhton in the future?) so that all changes can be made there. See also this: https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/42 discussion.
Enjoy the Feiertag on Thursday. All the best, m

MiczFlor, 2018-05-07:
Quick update: just read through the installation routine for mpv on RPi. I hope you will document this well that it can be part of an install routine that is newbie friendly :)

Luegengladiator, 2018-05-07:
Hi @MiczFlor 
the installation of MPV is the same as VLC. There will be no OnScreenDisplay and no Status on the Commandline. I worked on the config-split on https://github.com/Luegengladiator/RPi-Jukebox-RFID/tree/cleanup_scripts_from_config as a first project. Now its possible to link the VLC and MPV playoutcontrols from settings/configuration folder to the original playoutcontrols.sh. 
Next thing is to get a Installer for the hole Jukebox maybe with Installeroptions vlc/mpv

Luegengladiator, 2018-05-29:
Pull Request #50 https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/50

closed 2018-05-29 by Luegengladiator