MiczFlor/RPi-Jukebox-RFID #46: Commands via script do not work

bgraeb, 2018-05-13:
Hey there,

I'm running the project (on an older release) and I have the issue, that the pause-button by rfid is not working. It pauses successfully, when I use the Webinterface button, or type the command `echo "pause" | nc.openbsd -w 1 localhost 4212` in a ssh session. But using a rfid-card does not work.

`
pi@jukebox:/etc $ sudo systemctl status rfid-reader.service
● rfid-reader.service - RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/rfid-reader.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-05-13 20:17:35 CEST; 7s ago
 Main PID: 1369 (python2)
   CGroup: /system.slice/rfid-reader.service
           └─1369 /usr/bin/python2 /home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py

May 13 20:17:35 jukebox systemd[1]: Started RFID-Reader Service.
May 13 20:17:42 jukebox python2[1369]: CARDID = 1028061683
May 13 20:17:42 jukebox python2[1369]: pausing <- added this, to verify that the if-statement triggered
May 13 20:17:42 jukebox python2[1369]: VLC media player 2.2.6 Umbrella
May 13 20:17:42 jukebox python2[1369]: Command Line Interface initialized. Type `help' for help.
pi@jukebox:/etc $
`

Using `wget http://jukebox/index.php?player=pause` works inside ssh also, but still not inside the script. Any ideas, how this could be?

bgraeb, 2018-05-13:
Alright, fixed the problem by rtfm - obviously, things work strange if you use crontab AND systemctl at the same time to auto run the service -.-

closed 2018-05-13 by bgraeb

MiczFlor, 2018-05-14:
Hi @bgraeb , you were not alone, because I WTFM (wrote the f******* manual) and had that mistake in the install script, too. Just changed that a few days ago. Good luck with the jukebox and I'd be honoured to get some pics of your work here https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/5 to include in [the gallery](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/GALLERY.md)