MiczFlor/RPi-Jukebox-RFID #47: Resume last played track (especially for audiobooks)

Neponator, 2018-05-17:
Hey,

I tried for the last week to implement a set of scripts that allows to resume the last played track. I came pretty far and the theory already works, however my lack of python / bash scripting prevents me from making the final step - I count on your help here.

At first my approach (code see below): I read in another forum that the VLC that the RPi-Jukebox uses offers a XML file, that shows all information on the playing track (like artist, title, current position, current track number of the playlist). I took those information and saved it into separate text files - the artist and title for a display (that I plan for the future) as well as the current track position and playlist number. Since the "download" of this xml is pretty slow (1-3 seconds), I only use it when pressing the button for "pause" or "shutdown".
Current track position and playlist number I use in combination with the VLC remote functions "goto x" and "seek x" (seek is already integrated in the code with seek 0 as a possibility to start the track from beginning). "goto x" works similarly, it makes vlc jump to the specific item on the playlist and plays it. 
To make it work, the playlist cannot be temporary like in the original project, so I changed the path in the rfid_trigger_play.sh to RPi-Jukebox-RFID\shared\playlists\ (First I chose the Folder containing the mp3s itself, however that did not work as all the files in the folder are used to create the playlist, including the m3u's. So I added the subfolder. At the sametime I exported the path of the m3u to a textfile to use after reboot.
Now I have all I need, the playlist and the text files containing current position and current track.

I then created a bash-script that runs after boot of the pi, starts the VLC, loads the playlist, jumps to the current track and plays a specific position. My script works perfectly when I run it in the console, however if I integrate it into /etc/rc.local  it does not run at all, if I add a service in the /etc/systemd/system/ it plays the playlist, but does not jump to track and position. Again, it works when run from console with "bash lastplayed.sh". Is someone seeing my mistake here? I lack the skills to see my error. Even though I learned to program in C++ like 15 years ago, I forgot most of it and started to learn python only a week ago - so  most of my code is stolen or is the result for trial and error - if you have any recommendations for simplyfing it, let me know. 

Now to my code: 

1. To make VLC produce the xml change in the rfid_trigger_play.sh the line

`cvlc --no-video --network-caching=10000 -I rc --rc-host localhost:4212 "$PLAYLISTPATH" &`

to 

`cvlc --no-video --network-caching=10000 -I rc --extraintf=http --http-password 123 --rc-host localhost:4212 "$PLAYLISTPATH" &`

2. When already in rfid_trigger_play.sh also change the playlist from temporary to permanent:

instead of

`PLAYLISTPATH="/tmp/$FOLDERNAME.m3u"`

set

`PLAYLISTPATH="$PATHDATA/../shared/playlists/$FOLDERNAME.m3u"` (not sure if you need to create the folder "playlists" in ./shared first or if the script does it for you - if you want to be sure, create it first with `mkdir playlists`.

Additionally we want to export the path of the playlist to a textfile, to be able to remember the last used playlist at start of the pi:

Still in rfid_trigger_play.sh add 
`echo "$PATHDATA/../shared/playlists/$FOLDERNAME.m3u" > PATHDATA/../shared/lastplayed.txt`
**after**
`find` "$PATHDATA/../shared/audiofolders/$FOLDERNAME" -type f | sort -n > `"$PLAYLISTPATH"`

2. This is my script to exploit the VLC xml:

`from bs4 import BeautifulSoup
import urllib.request

def get_currentSongDetails():

    password_mgr = urllib.request.HTTPPasswordMgrWithDefaultRealm()

    top_level_url = "http://127.0.0.1:8080/requests/status.xml"
    password_mgr.add_password(None, top_level_url, '', '123')

    handler = urllib.request.HTTPBasicAuthHandler(password_mgr)

    opener = urllib.request.build_opener(handler)

    u = opener.open(top_level_url)

    soup = BeautifulSoup(u, 'html.parser')

    currents = []

    for info in soup.select('info'):            # Selects the info category
        if info['name'] == 'title':             # Selects the title inside the info category
            currents.append(info.text)          # Writes title into the currents-array

    for playid in soup.select('currentplid'):   # Selects the current playlist-number of the track
            currents.append(playid.text)        # Writes the playlist-number in the currents-array

    for current in soup.select('time'):           #Selects the current position in the track
        currents.append(current.text)           # Writes the current position into current-array

    return (currents)

currentSongDetails = get_currentSongDetails()

#with open ("../shared/latestSong.txt", "w") as text_file:     #for putting on a display - later!
#  for (cur) in currentSongDetails[0]:
#     text_file.write('{}'.format(cur))

with open ("../shared/latestPlid.txt", "w") as text_file2:
   for (cur) in currentSongDetails[1]:
    print('{}'.format(cur), file=text_file2)

with open ("../shared/latesttime.txt", "w") as text_file3:
   for (cur) in currentSongDetails[2]:
    print('{}'.format(cur), file=text_file3)
`

Relevant output are the the textfiles _latestPlid.txt_ and _latesttime.txt_ that contain the last played playlist-item and the track position.

3. Now my script that should run while/after booting the pi:

```
#!/bin/bash

sudo pkill vlc #necessary?
VLCPLAYS=$(</home/pi/RPi-Jukebox-RFID/shared/lastplayed.txt)
cvlc --no-video --network-caching=10000 -I rc --extraintf=http --http-password 123 --rc-host localhost:4212 "$VLCPLAYS" &

sleep 2 

PLID=$(</home/pi/RPi-Jukebox-RFID/shared/latestPlid.txt)
echo "goto $PLID" | nc.openbsd -w 1 localhost 4212

SONGTIME=$(</home/pi/RPi-Jukebox-RFID/shared/latesttime.txt)
echo "seek $SONGTIME" | nc.openbsd -w 1 localhost 4212
```

This script works perfectly when run in the console via `bash lastplayed.sh`, but does not run properly when started automatically after boot.

All textfiles are accessible (chmod 777, just to be sure) and contain data.

The service I use to start the script lies in /etc/systemd/system/ and looks like this:

`
[Unit]
Description=Resume Last Song Service
After=network.target iptables.service firewalld.service rfid-reader.service

[Service]
WorkingDirectory=/home/pi/RPi-Jukebox-RFID
ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/lastplayed.sh
KillMode=process
SendSIGKILL=no

[Install]
WantedBy=multi-user.target
`

I'm (again) at my wits end. Someone sees why the hell my script works to resume the last playlist, but not the track/position, but does work when run from console?

MiczFlor, 2018-05-19:
Hi @Neponator

you started a good feature here, I haven't found the time to really test what you are aiming at. Still, some feedback (to keep your motivation up! We all want your feature!). When I have more time, I will join in the effort.

* yes, if you want to keep the playlists intact after a reboot you can not use the `tmp` folder. So changing the path as you did is necessary.
* there is a `$` missing here? `echo "$PATHDATA/../shared/playlists/$FOLDERNAME.m3u" > PATHDATA/../shared/lastplayed.txt` the second PATHDATA should be $PATHDATA?
* the best thing I can come up with without getting into the code and test it: 

a) try to give each action in your script more time by adding sleep. Just for testing. 

b) the bash `rfid_trigger_play.sh` creates a new playlist from the folder contents when it is run. Possibly that confuses the script? Timing wise? Try with commenting out the playlist creation?

c) `sudo` everything and then see if that does the trick... :/

Neponator, 2018-05-19:
You're my hero - a) already solved it - it's not only the time between calling the playlist and the track (min sleep 5) but also the time between calling track (goto x) and calling position (seek x) in the track, that requires at a sleep 2, otherwise the requests confuse each other. Resume running now! Many thanks @MiczFlor ! I updated the code in post above. Will keep the issue open to create a manual that is easier to follow :-D

MiczFlor, 2018-05-20:
Hi @Neponator
Glad it works and yes, please keep it open and a bit of documentation would be great. 
I am a big fan of pull requests. They write you into the history of the phoniebox code base. And they open another thread to discuss suggestions and changes, see here for an example:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/42
From your github profile, I can't see how savy you are with git :) You might to read a bit
* https://help.github.com/articles/adding-a-file-to-a-repository/
* https://help.github.com/articles/creating-a-pull-request/
But it's worth the effort and better practice for all of us.
If you prefer, obviously, I will add your script and changes to the repo myself.
Thanks, m

MiczFlor, 2018-08-16:
I am closing this ticket, because of the launch of version 1.0 on master. If you have any issues, please try first if they are still happening on the latest version 1.0. If so, open another ticket, please. 

* Use this thread for update issues [if you are running `stretch`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/138) 

**The version 1.0 has not been tested on `jessie`**

* Read this thread [if you are running `jessie`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/137) 

Version 1.0 is a major improvement (which will be full of bugs, for sure :) and I want to thank all the contributors and Phoniebox lovers with their input, pictures, bug fixes and suggestions.

closed 2018-08-16 by MiczFlor