MiczFlor/RPi-Jukebox-RFID #51: Web App does not render as it should

TM4G1, 2018-05-30:
Hi @MiczFlor,

as my other issue is closed now, I am tackling now one on my jukebox that I initially set up with a version prior to the current version (0.9.4).

The web app somehow renders not very nicely in my browser on PC / Smartphone (see screenshot below). I already performed a GIT fetch, but this did not help. The version file got updated to 0.9.4 through the fetch. I also ran the upgrade commands that you posted.

Would you have an idea or advice how I could get the web app properly running?

[image]

Thanks for your feedback!
Tom

MiczFlor, 2018-05-30:
Hi Tom @TM4G1 

If you look into the source view of the web app, you should see the links to css and js files like this one:
~~~
<link rel="stylesheet" href="_assets/bootstrap-3/css/bootstrap.min.css">
~~~
In line 269 of this script: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/htdocs/index.php the `$conf['base_url']` is passed on to create the header (I am passing on this variable in case at some point there will be a subfolder and scripts within that subfolder need to have a relative link to the css files).
~~~
html_bootstrap3_createHeader("en","RPi Jukebox",$conf['base_url']);
~~~
The variable is set in the file https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/htdocs/config.php.sample

And the best thought I can come up with right now: I forgot to put into the manual to make a copy of this config file...?
~~~
cp config.php.sample config.php
sudo chown pi:www-data config.php
~~~
?

TM4G1, 2018-05-31:
Hi @MiczFlor,

this fixed the style =) great, thanks! Still I got an issue with the functions on the webapp. Most playback / volume control buttons do not work. I marked on the below screenshots the buttons in red, which are not working. The pause and play button in the main control are working fine.

[image]

Thanks for all your support and best regards
Tom

MiczFlor, 2018-05-31:
A hunch: are you using GPIO buttons? (not that I know why that would make a difference, but that is the definition of 'hunch')

Things to check would be:
* is the web server in the `sudoer` list and can execute as sudo?
* does the script `scripts/playout_controls.sh` have the right settings and can be executed by the web server?
* in this script, have you tried a different audio interface in the line `DEVICE=PCM`?

TM4G1, 2018-05-31:
No I am not using GPIO buttons.

I doublechecked, that in `sudo nano /etc/sudoers` the following line `www-data ALL=(ALL) NOPASSWD: ALL ` is appended at the bottom.

How can I check, if the `scripts/playout_controls.sh` have the right settings and can be executed by the web server? I compared this with my 0.9.4 jukebox's script. Notepad++ compare checks says it is the same content in it.

I tried to use the `MASTER` audio interface as well, but this did not help. I now set it back to PCM. 

Best regards
Tom

fuggly, 2018-06-01:
same here.. volume controls and play buttons don't work.. or at least I don't know how I should recognise whether they work or not. If I play one of the songs in my library, it.. well... does nothing, at last nothing I can hear. Trying to make it louder only ends up with a volume set to 0%.

an ls -lisa on the play_controls.sh results in 

1835681 4 -rwxrwxr-x 1 pi pi 2363 Jun  1 14:37 playout_controls.sh

other than that... I tried put DEVICE to MASTER but that hasn't really worked either, so I reverted back to PCM. 

What more could I try?

Best regards and thanks for this cool project!

TM4G1, 2018-06-03:
Unfortunately still the same for me.

[image]

Not sure, if this information would give a hint: When I started to play a song via RFID trigger the song starts. If I after that click on play via the web app the playback actually stops. So something is happening on the button click being the trigger.

MiczFlor, 2018-06-03:
Hi @TM4G1 
I think there are two things at play now :) the new issue with the play button that stops, it actually pauses. A new pull request that is an improvement but still needs a little improvement:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/52
if this file exists on your jukebox now, then you have the latest git core version - with the problem you describe regarding the play button. 
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/htdocs/inc.vlcStatus.php

Back to your problem from before the last pull request. Please give me some more information, becaus I really am in the dark and need to reproduce your problem.

* What Raspberry are you using?
* What OS (jessie or stretch)?
* What sound card? built in? HDMI? USB? Which make?
* Can you change the volume with the RFID chips? (see here for more info: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/MANUAL.md#jukebox-controls-using-rfid-cards )

Please run the following script in the terminal and post the results (no need for a screenshot, text is fine):
~~~
cd /home/pi/RPi-Jukebox-RFID/scripts/
echo "BASH FILES:"
ls -l scripts/*.sh
echo "HTDOCS FILES:"
ls -l htdocs/
echo "AMIXER CONTROLS:"
amixer controls
~~~

Thanks

TM4G1, 2018-06-03:
Hi @MiczFlor ,

> #52
if this file exists on your jukebox now, then you have the latest git core version - with the problem you describe regarding the play button.
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/htdocs/inc.vlcStatus.php

No I do not have this file in. Did also perform a git fetch and it did not get fetched.

> What Raspberry are you using?

Raspberry PI 3

> What OS (jessie or stretch)?

Jessie

> What sound card? built in? HDMI? USB? Which make?

Built-in, but I tried to set it up with the proposed usb soundcard. This unfortunately did not boot along with the PI (only on plugging it when the PI was booted up). I gave up on it as I found the sound is good enough without the card.

> Can you change the volume with the RFID chips? (see here for more info: 

Yes. I have a 75% and 100% volume card which work fine!

Here the results (if someone else also tries, you need to navigate one folder up for execution of ls.)

```pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ cd /home/pi/RPi-Jukebox-RFID/  
pi@raspberrypi:~/RPi-Jukebox-RFID $ echo "BASH FILES:"  
BASH FILES:  
pi@raspberrypi:~/RPi-Jukebox-RFID $ ls -l scripts/*.sh  
-rwxr-xr-x 1 pi pi 2363 Jun  3 07:57 scripts/playout_controls.sh  
-rwxr-xr-x 1 pi pi 9589 Jun  3 19:09 scripts/rfid_trigger_play.sh  
pi@raspberrypi:~/RPi-Jukebox-RFID $ echo "HTDOCS FILES:"  
HTDOCS FILES:  
pi@raspberrypi:~/RPi-Jukebox-RFID $ ls -l htdocs/  
total 96  
-rwxrwxr-x 1 pi www-data   510 May 29 17:02 ajax.refresh_id.php  
drwxrwxr-x 6 pi www-data  4096 May 16 11:40 _assets  
-rwxrwxr-x 1 pi www-data  9253 May 29 17:02 cardEdit.php  
-rwxrwxr-x 1 pi www-data  8928 May 29 17:02 cardRegisterNew.php  
-rwxrwxr-x 1 pi www-data   198 May 31 07:10 config.php  
-rwxrwxr-x 1 pi www-data   198 May 29 16:57 config.php.backup.0.9.3  
-rwxrwxr-x 1 pi www-data   198 May 29 17:02 config.php.sample  
-rwxrwxr-x 1 pi www-data  3729 May 29 17:02 func.php  
-rwxrwxr-x 1 pi www-data   719 May 29 17:02 inc.controlPlayer.php  
-rwxrwxr-x 1 pi www-data   332 May 29 17:02 inc.controlVolumeUpDown.php  
-rwxrwxr-x 1 pi www-data  5360 May 29 17:02 inc.formCardEdit.php  
-rwxrwxr-x 1 pi www-data  1270 May 29 17:02 inc.navigation.php  
-rwxrwxr-x 1 pi www-data  1233 May 29 17:02 inc.processAddNewStream.php  
-rwxrwxr-x 1 pi www-data   795 May 29 17:02 inc.volumeSelect.php  
-rwxrwxr-x 1 pi www-data 13704 May 29 17:02 index.php  
-rwxrwxr-x 1 pi www-data  2294 May 29 17:02 test.php  
pi@raspberrypi:~/RPi-Jukebox-RFID $ echo "AMIXER CONTROLS:"  
AMIXER CONTROLS:  
pi@raspberrypi:~/RPi-Jukebox-RFID $ amixer controls  
numid=3,iface=MIXER,name='PCM Playback Route'  
numid=2,iface=MIXER,name='PCM Playback Switch'  
numid=1,iface=MIXER,name='PCM Playback Volume'  
numid=5,iface=PCM,name='IEC958 Playback Con Mask'  
numid=4,iface=PCM,name='IEC958 Playback Default'  
pi@raspberrypi:~/RPi-Jukebox-RFID $  
```

MiczFlor, 2018-06-04:
Hi @TM4G1 
could you try if this suggestion might fix your problem?
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/54
That would lift a heavy weight off my chest :)

TM4G1, 2018-06-06:
Hi @MiczFlor 

unfortunately it does not start the playback from the available audio section still, but the reloading seems to be fixed with this. I can click now on play on one of the available audios and the page just is reloading for a short period, but still no playback afterwards. The upper play and pause button work perfectly fine. Meaning I start with swiping the rfid and then I can pause and play the currently playing song. If I select one from the available audio section is also shows up under the playback section, but even when I then click the main play button --> no playback. Really weird.

I may consider to run through the complete installation again as it worked for the second jukebox that I build. Maybe the try to set up the broken usb audio card has to do with it... not sure so.

Thanks
Tom

MiczFlor, 2018-06-08:
Hi @TM4G1 
I think the issue should be fixed now with the latest release. Please red the upgrade info, because it brings a few changes PLUS the issue described elsewhere: the if statement requires "" around string vars in some shells.

closed 2018-06-08 by MiczFlor

TM4G1, 2018-06-10:
Hi @MiczFlor 

I upgraded to 0.95 based on the upgrade information and rebooted. Unfortunately still the same issue. I may just set up everything from scratch. 

Best regards
Tom