MiczFlor/RPi-Jukebox-RFID #53: New idea: Sleeptimer function command? Turn off after ideling 30 minutes?

TM4G1, 2018-06-03:
Hi @MiczFlor ,

my wife came up with the idea of having a sleeptimer command. E. g. swipe a RFID card, which makes the raspberry shut down after 45 minutes to not consume the whole powerbank overnight when accidently left turned on.

May in general it would also be good to have an option to automatic turn off the pi, if there is no playback for 30 minutes.

What do you think about this?

Best regards
Tom

MiczFlor, 2018-06-03:
Nice idea. I‘ll work on the timer first, because I know how to do that. 

The auto shutdown... not so sure how to do it :) I will research and add this when I know how to as an optional setting, because it‘s not perfect for all use cases. But a great feature for many. 

Thanks for the input. 

All the best, micz

itkevin, 2018-06-04:
Been working on a auto sutdown script for my Pi Zero. Wanted to do a Pull Request whenever I have tested it for a few days. Maybe it would help. https://github.com/itkevin/RPi-Jukebox-RFID/blob/master/scripts/shutdown_on_idle.py

Luegengladiator, 2018-06-05:
You can use Linux commant "at" to do that:
at now + 10min /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c stop/shutdown/pause
at 23:59 /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c mute

please don't reinvent the wheel

Neponator, 2018-06-05:
> You can use Linux commant "at" to do that:
> at now + 10min /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c stop/shutdown/pause
> at 23:59 /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c mute

I was thinking of having a shutdown timer that is running right from the boot with like 45 min until shutdown, but every push of a button/use of RFID resets that timer. You see an easy way to implement that?

Luegengladiator, 2018-06-06:
A "easy" way is to use atq to list your planed jobs and delete the one you want with atrm. After that add a new one with the new time. 
Maybe you can use https://stackoverflow.com/questions/4304631/rescheduling-an-at-job?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa
to modify your job

MiczFlor, 2018-06-06:
Hi @Luegengladiator 

* sleeptimer: I was planning to do the sleeptimer with "at" - and link that to a chip, so the future event of pkill vlc will be set on swipe
* autoshutdown: I am not sure how to handle that. @itkevin script is neat and could be run as a separate serverd - so it is "optional" to those who want it. And I want it to be optional, because I don't want it on my install ;)

How to do the autoshutdown in bash is unclear to me. The autoshutdown would be:

`if nothing was swiped for an hour, sudo shutdown`

But I do not want to use crontab to check in every five minutes. Any ideas?

Luegengladiator, 2018-06-06:
Looking at the code of @itkevin the Idea is to shut down if load < x (where x is something you can define).
Let's assume the player runs with a defined List of files. For example 20 songs so it's nearly 60 minutes to run. Let's assume during the playback the load of the machine is > 0,8. After playback ends the player stops and the load drops to 0,2. 
My Idea is now to use a rfid to start a command "at -d shutdown -h 5" because -d (batch) executes the command if load < atd invocation. 
From the man page https://linux.die.net/man/8/atd we can define exactly which load is ok and which is "idle".
BUT! That is only possible, if the RPi runs only the Phonie. Every additional Program, Update and action on the systems ends in System load keeping the Phonie up and running until load drops below the given value.

Autoshutdown on nothing swiped is a strange feature. In your example the system goes down if you listen to you favorite Radio broadcast for more than 1 hour...

MiczFlor, 2018-06-06:
"Autoshutdown on nothing swiped is a strange feature." ... you are right, I really didn't think this through.

I suggest to do a "quick fix" by supplying two features:

a. swipe and the audio stops after 30 minutes
b. swipe and the box shuts down after 30 minutes

That might catch 90% of the use cases (i.e. "Ok, ok, you go to bed and listen some more. But once this stops, it's really sleepy time!")

MiczFlor, 2018-06-08:
Hi @TM4G1 @Luegengladiator 
I merged a couple of changes. With the last commit, I implemented the timer function. Please give it a try. There are a couple of hoops to go through, you can see the update information to see details.

When the box is running, you can stop playout in 1 minute with the command
~~~
./playout_controls.sh -c=playerstopafter -v=1
~~~
Shutdown after 1 minute with the command
~~~
./playout_controls.sh -c=shutdownafter -v=1
~~~

There is a new file for the card configuration in the `settings` folder, at the bottom you can see the setting if IDs for 5, 15, 30 and / or 60 minutes.

Works for me, hope it works for you. Do not forget to install `at`
~~~
sudo apt-get install at
~~~

Luegengladiator, 2018-06-09:
Code looks good. Manual execution of the "at" part works for me too.
Was a little tricky to find the correct place in the  if-Block. Maybe it's better to use a case at that point.
I'll try the new Version from Tuesday on.

MiczFlor, 2018-06-11:
HI @Luegengladiator 
thanks for the peer review, I will close the issue for now, because the complex solution I will only start if another ticket will be opened especially for that. I believe the latest commits should satisfy most needs...
Commits: https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/adb614ea9f8b371bca8980ec3d38a1ab81866ccd https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/c2577db150e865f8591fc8371fabbbf8070d5b1b

zxa, 2018-06-18:
Sorry for reviving that topic but maybe someone is interested in my solution. As i'm using a powerbank in my phoniebox i want that powersave feature.

It uses the "at" command and i tried to not interfere with the sleep timer function using different "at" queues. It also handles the case if the box is in pause mode:
https://github.com/zxa/RPi-Jukebox-RFID/tree/shutdown-after-idle

zxa, 2018-06-19:
> Something tells me it could be slightly more elegant

As i'm only an interested amateur, i can live with that. Sorry for hurting the eyes of anybody who knows what he is doing while coding. ;-)
I'll doing a stress test to the code in the next days, as the paused VLC idle tracking is prone to failure.
I'm thinking about of moving the
`[ "$COMMAND" == "shutdownidle" ]`
part into the watchdog-idle.sh, as it for my feeling (let's talk about feelings!) doesn't belong in to the playout_controls. What's your opinion?

zxa, 2018-06-19:
> My idea: you create a pull request for this AND add the following near the top in https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/playout_controls.sh [...]

I have that already in my code at the moment, so only one file is needed for setting the feature on/off and adjusting the idle time for your own needs:

```
##############################################
# Idle time after the RPi will be shut down. 0=turn off feature.
# 1. create a default if file does not exist
if [ ! -f $PATHDATA/../settings/Idle_Time_Before_Shutdown ]; then
    echo "0" > $PATHDATA/../settings/Idle_Time_Before_Shutdown
fi
# 2. then|or read value from file
IDLETIME=`cat $PATHDATA/../settings/Idle_Time_Before_Shutdown`
```

and i check with
`if (( $IDLETIME > 0 ))`

(OK, i forgot that in the pause section).

I don't know why we should bring in a second config file for that feature, it doesn't feel elegant ;-)

zxa, 2018-06-28:
Hi @MiczFlor,

i created a new solution which uses the vlc rc interface. It leaves the playout_controls.sh untouched and consists only of the service and this file:

https://github.com/zxa/RPi-Jukebox-RFID/blob/idle-shutdown/scripts/idle-watchdog.sh  

It works for me under stretch, but as it relies on the output "state playing" from the rc interface it should be checked if the output under jessie is the same.

MiczFlor, 2018-07-03:
The sleeptimer is part of this pull request:
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/67
It's not yet part of the documentation, will do that in the coming days

closed 2018-07-03 by MiczFlor