MiczFlor/RPi-Jukebox-RFID #54: playout_controls.sh - use "=" instead of "=="

PhilTonic, 2018-06-04:
took me some time to discover this. from shell i could run the script without errors, but launched within python via check_call it throws this error:
volumeup: unexpected operator
Unknown COMMAND volumeup VALUE

i've read that only few shells support == in if-constructions, so i had to change 
every occurance of $COMMAND == "xyz" to $COMMAND = "xyz"

regards,
phil

MiczFlor, 2018-06-04:
Hi @PhilTonic 
nice catch. And a good proof that "never change a running system" is not always true when it's not only one system that needs to run... (aka "works for me"...)

I will look into this in the coming days.

MiczFlor, 2018-06-08:
This was fixed with the latest 0.9.5 release

closed 2018-06-08 by MiczFlor