MiczFlor/RPi-Jukebox-RFID #57: Volume Up/Down not working

zxa, 2018-06-14:
I've installed the script on a RPi Zero W with Raspbian stretch. I've noticed that the Volume Up/Down feature seems to be broken:

./playout_controls.sh -c=volumedown
30 30
expr: syntax error

It seem to be not possible to do math with the output of that grep line:

       # read volume in percent
        VOLPERCENT=`amixer sget \'$DEVICE\' | grep -Po '(?<=\[)[^]]*(?=%])'`
        echo $VOLPERCENT
        # increase by $VOLSTEP
        VOLPERCENT=`expr ${VOLPERCENT} - ${VOLSTEP}

I haven't found a fix (i experimented with let, declare -i ....).
Is this code working for anyone?

MiczFlor, 2018-06-14:
Hi @zxa 
it is working for me on two devices, (Pi1 and Pi3) with Raspbian, via cards and via webapp. I will take a look at this. 

This is happening for volumedown AND volumeup?
And can you check if $DEVICE is set? (try adding 'echo')

in your code extract there is an important ``` missing, at the end of the line:
~~~
VOLPERCENT=`expr ${VOLPERCENT} + ${VOLSTEP}` 
~~~
might that have to do with it?

zxa, 2018-06-14:
Hi @MiczFlor,

it happens with up and down. The missing \` is just an copy paste error, it's in the code.
But I managed to fix the problem:

`  amixer sget Master`

gives me:

> Simple mixer control 'Master',0
>   Capabilities: volume
>   Playback channels: Front Left - Front Right
>   Capture channels: Front Left - Front Right
>   Limits: 0 - 255
>   Front Left: 82 [32%]
>   Front Right: 82 [32%]

(I'm using a pHAT DAC, which has only a "Master", not a PCM-Fader)
The grep makes out of this

> 32 32

and the calculation with $VOLPERCENT fails. I guess that "amixer sget PCM" only return only one percentage value. So it is a stereo fader problem.

The fix is to tell grep to only get the first matching line by adding -m 1

  VOLPERCENT=`amixer sget \'$DEVICE\' | grep -Po -m 1 '(?<=\[)[^]]*(?=%])'`

This works now. The embarrassing thing is, that it took me only a minute to spot the problem today while i was fooled yesterday for hours by the double "echo $VOLPERCENT" in the code in thinking that the output of "32 32" is ok.

I think this could be included in the code as it should work with mono faders like PCM as well.

MiczFlor, 2018-06-15:
Hi @zxa

Nice catch! To become part of the inglorious Phoniebox contributor crowd: why not create a pull request and enter the hall of fame? :)
All the best, micz

MiczFlor, 2018-06-15:
+1 :) and don‘t forget to share pics of your box in the feature thread. Seeing all these sparkling inspirations is my biggest motivation  @zxa

closed 2018-06-22 by MiczFlor