MiczFlor/RPi-Jukebox-RFID #60: No update on Web App

Daigel2002, 2018-06-19:
Hi, I just implemented the scripts and the web server for configuration starts well. But any update in the web app and also the RFID are not updating so there is no interaction. What could be the reason?
Regards
Christian

MiczFlor, 2018-06-22:
Hm. If I understand you correctly, you can see the web interface for the Phoniebox. But nothing else works. Neither pressing buttons in the webapp nor using the RFID cards has any effect?
I don't know what OS you are using, so the following links assume it is Stretch
* Make sure you have registered your RFID reader https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/CONFIGURE-stretch.md#register-your-usb-device-for-the-jukebox
* Make sure the web server is in the sudoer list https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/INSTALL-stretch.md#lighttpd-web-server-for-web-app
* Try different iFace names for the audio playout https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/CONFIGURE-stretch.md#create-settings-for-audio-playout

That's the best I can come up with for now. Please post your progress here.

Thanks and good luck!

Daigel2002, 2018-06-24:
Hi, thank you for you response, sorry, my question was very unspecific.
I realized, that any update of the RFID-reader cannot be written back, that is how it looks like.
When using "sudo systemctl status rfid-reader", I receive a positive feedback about the reader-status, but also:
"python2[390]: cp: cannot stat '/home/pi/RPi-Jukebox-RFID/../settings/rfid_trigger_play.conf.sample': No such file or directory"
despite the fact, that the file(s) are available. 
Are there permissions missing?
Thank you
Regards
Daigel2002

MiczFlor, 2018-06-25:
Hi @Daigel2002 

thanks for the update. I have actually no idea where this line could come from. The only places where the file `rfid_trigger_play.conf.sample` is mentioned in the repo is either in the documentation or in install scripts or, the only other place, inside `rfid_trigger_play.sh` like this:
~~~
# create the configuration file from sample - if it does not exist
if [ ! -f $PATHDATA/../settings/rfid_trigger_play.conf ]; then
    cp $PATHDATA/../settings/rfid_trigger_play.conf.sample $PATHDATA/../settings/rfid_trigger_play.conf
    # change the read/write so that later this might also be editable through the web app
    sudo chown -R pi:www-data $PATHDATA/../settings/rfid_trigger_play.conf
    sudo chmod -R 775 $PATHDATA/../settings/rfid_trigger_play.conf
fi
~~~
So where does this line you are quoting come from? When does it show on the screen? When you test the status of the daemon?

One thing I notice: the path in the line is not correct, I believe. It says
~~~
/home/pi/RPi-Jukebox-RFID/../settings/rfid_trigger_play.conf.sample
~~~
Where it should say
~~~
/home/pi/RPi-Jukebox-RFID/settings/rfid_trigger_play.conf.sample
~~~

MiczFlor, 2018-07-03:
This thread has been idle for some time, closing it.

closed 2018-07-03 by MiczFlor