MiczFlor/RPi-Jukebox-RFID #65: Support of EAN Scanner

Luegengladiator, 2018-07-02:
EAN Codes are much simpler to create and are almost everywhere (CD, Tape, aso.)

Luegengladiator, 2018-07-02:
Created a Project in Fork.
Pull Request when done

MiczFlor, 2018-07-03:
* Would this require an optical reader / camera? 
* Would this allow to use the existing codes on products?

Luegengladiator, 2018-07-03:
Yes this is the Idea behind that. 
Use the existing EAN-Codes and yes a camera/ean-scanner is needed.

Luegengladiator, 2018-07-09:
Worked out of the box.
Register EAN Scanner with the RegisterDevice.py
There are 2 Prerequisites:
1) The Scanner has to send Carriage Return after reading
2) The Scanner has to work as USB-Keyboard as STDIN
most of the Barcodescanner can be configured. Please see the device Handbook.

Luegengladiator, 2018-07-09:
@MiczFlor 
finished this one.
All my RFID-Cards now get additional EAN Codes on the back. So I can run 2 Jukeboxes with different input devices with only one definition. (cloning the Raspberry SD)

closed 2018-07-09 by Luegengladiator