MiczFlor/RPi-Jukebox-RFID #66: Use APT as installer

Luegengladiator, 2018-07-02:
Simplify the Installation process with requirements and default Installer Pakages for RPi

Luegengladiator, 2018-07-02:
Created Project in Fork.
Pull Request when done

MiczFlor, 2018-07-03:
I wouldn't mind if this exists, but I am not sure what that would entail for future development. Also, there are so many "manual" steps (Wifi, RFID reader type, samba config, ssh password) that I am not sure what the apt-get would look like. But, hey, go ahead, I love to learn something new.

Luegengladiator, 2018-07-03:
I think configuration is a more or less manual part. We can ask a few things during installation like "which player" or "do you use EAN or RFID" or something like that.
But there are many prerequisites to run without problems. So I think it's nice to install the Phonie with apt-get install RPi-Jukebox-RFID and all the requirements (samba, tmux, at, ssh, ...) are installed automatically....

MiczFlor, 2018-07-04:
Welcome @Luegengladiator as our packet manager :+1: :)

Luegengladiator, 2018-07-04:
I think i'll need time on that. 
Usually I'm building RPMs and I have to figure out how to build an apt-package

Luegengladiator, 2018-07-09:
Only nice to have

closed 2018-07-09 by Luegengladiator