MiczFlor/RPi-Jukebox-RFID #69: 404 Error on Web App Login

capirex65, 2018-07-03:
Hallo zusammen. Meine Hardware nach der MiczFlors Anleitung läuft und ich teste fleißig. Aktuell noch über den HDMI Ausgang an meinen Verstärker, aber die Boxen kommen erst diese Woche. Leider geht das Web Interface nicht wenn ich die IP des PI ins Iphone eingebe. Ich bekommen nur einen Fehler 404. Was ja eigentlich auf einen falschen Link hinweist.

Hätte Ihr einen Tipp, was ich noch prüfen könnte? Kann es mit einem Fehler der Samba Konfiguration zusammen hängen?

MiczFlor, 2018-07-03:
Hallo, schön, dass du dich in den Code eintauchst.

Schwer zu sagen aus der Distanz was da geht und was da nicht geht. Deshalb ein paar Fragen:

* Kannst du dich über Samba mit dem Rechner verbinden (also nicht über den Browser)
* Ist der Pi am Wifi Netz auf dem auch das Handy / Laptop / Tablet ist?
* Wenn du am Pi sitzt (mit Mouse und Keyboard eingestöpselt) bist du dann auch im Netz? Wenn ja, kopiere mal hier in den thread was du siehst, wenn du im terminal `ifconfig` eingibst.
* Hast du schon RFID chips, die die Phoniebox steuern (abspielen und/oder Vol ändern)?

capirex65, 2018-07-03:
1. den ersten Punkt weis ich leider nicht so recht zu testen aber ich kann sowohl vom Mac als auch vom Windows PC per remote zugreifen. Und im Mac habe ich den PI als Laufwerk eingebunden also per smb:// ip... 
2. ja ich seh die IP im Router
3. Achtung ich habe noch keine statische IP vergeben, um so wenig Fehler wie moeglich zu machen. Aber die Adresse 192.168.1.3 stimmt.
4. ja habe 4 CD "von Hand" vom Mac kopiert und die Files angelegt. Die Musik wird auch direkt gespielt. Soweit ist eigentlich alles OK

Danke schon einmal fuer deine Hilfe 

th0      Link encap:Ethernet  Hardware Adresse b8:27:eb:cd:16:b2  
          inet6-Adresse: fe80::d43e:d611:1927:bdaa/64 Gültigkeitsbereich:Verbindung
          UP BROADCAST MULTICAST  MTU:1500  Metrik:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Lokale Schleife  
          inet Adresse:127.0.0.1  Maske:255.0.0.0
          inet6-Adresse: ::1/128 Gültigkeitsbereich:Maschine
          UP LOOPBACK RUNNING  MTU:65536  Metrik:1
          RX packets:30350 errors:0 dropped:0 overruns:0 frame:0
          TX packets:30350 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1 
          RX bytes:191646357 (182.7 MiB)  TX bytes:191646357 (182.7 MiB)

wlan0     Link encap:Ethernet  Hardware Adresse b8:27:eb:98:43:e7  
          inet Adresse:192.168.1.3  Bcast:192.168.1.255  Maske:255.255.255.0
          inet6-Adresse: 2003:ed:43c7:2700:37e0:4ea6:e18b:3bc6/64 Gültigkeitsbereich:Global
          inet6-Adresse: fe80::16e5:e7f0:44b8:a08e/64 Gültigkeitsbereich:Verbindung
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metrik:1
          RX packets:21328 errors:0 dropped:4608 overruns:0 frame:0
          TX packets:19570 errors:0 dropped:0 overruns:0 carrier:0
          Kollisionen:0 Sendewarteschlangenlänge:1000 
          RX bytes:4140709 (3.9 MiB)  TX bytes:20859659 (19.8 MiB)

capirex65, 2018-07-03:
Liegt mein Fehler hier in der Smb.conf? ueberschneidet sich die definition oben nicht mit der unten fuer die juekbox? 

[homes]
   comment = Home Directories
   browseable = no

# By default, the home directories are exported read-only. Change the
# next parameter to 'no' if you want to be able to write to them.
   read only = yes

[pi_jukebox]
        comment= Pi Jukebox
        path= /home/pi/RPi-Jukebox-RFID/shared
        browseable=yes
        writeable=yes
        only guest=no
        create mask=0777
        directory mask=0777
        public=no

user :pi
password:raspberry

MiczFlor, 2018-07-04:
Mit `samba` sollte das nichts zu tun haben. Was passiert denn, wenn du den Pi mit Monitor und Keyboard verbunden hast und da im Browser `http://localhost/` eingibst?

capirex65, 2018-07-04:
Ich glaube wir kommen der Sache naeher:

wget localhost
--2018-07-04 21:40:54--  http://localhost/
Auflösen des Hostnamen »localhost (localhost)«... ::1, 127.0.0.1
Verbindungsaufbau zu localhost (localhost)|::1|:80... fehlgeschlagen: Verbindungsaufbau abgelehnt.
Verbindungsaufbau zu localhost (localhost)|127.0.0.1|:80... verbunden.
HTTP-Anforderung gesendet, warte auf Antwort... 404 Not Found
2018-07-04 21:40:54 FEHLER 404: Not Found.

MiczFlor, 2018-07-05:
Es sieht so aus, als ob der webserver nicht läuft. Wenn du noch keine Daten auf dem Pi hast, vielleicht einfach noch mal neuinstallieren? Es gibt dazu ja auch den Ordner `installscripts:

https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/scripts/installscripts

Da solltest du je nach OS jessie oder stretch das "default" install script wählen.

capirex65, 2018-07-08:
Laeuft  :D Danke nochmal fuer die vielen Tipps!!!! Haette ich die scripts mal eher gesehen. Damit wirds ja auch fuer den nicht Linux Programmierer fast idiotensicher.

MiczFlor, 2018-07-09:
Hi @capirex65 
Freut mich, dass es läuft und bitte, bitte, bitte vergiss nicht ein, zwei Fotos von deiner Box hier zu posten:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/5
Die Bilder motivieren mich dieses Projekt weiter zu betreuen und zu betreiben. Danke dir und viel Spass mit der Arbeit und ich freue mich auf Verbesserungsvorschläge!

micz

closed 2018-07-09 by MiczFlor