MiczFlor/RPi-Jukebox-RFID #70: Web app: control buttons too small

zxa, 2018-07-03:
On my Android device it's hard to hit the right button for the player control because they are pretty small (and i'm in the believe to not have abnormal fat fingers ;-).

Is this different on Apple devices?

As i'm working on the Web App at the moment (adding a settings page) i could fix this.

zxa, 2018-07-04:
Hi @MiczFlor,

i'll have look at that. This is how btn-lg looks like for me:
[image]

But before i will upload the new settings page for comments and public testing. It's optimized it for my device and it would be interesting how the Apple likes it:

[image]

MiczFlor, 2018-07-04:
Hi @zxa 
looks great. Can you move this to a separate settings page, like 
~~~
http://localhost/cardRegisterNew.php
~~~
?

The timed shutdown would be great to have at the top of the index page, alongside the player and volume change.

Other than that I think the front page is for playing and the settings would only be required 0.5% of the time you pull out your phone, no?

zxa, 2018-07-04:
Hi@MiczFlor,

it's already a separate page. I have time clean up the code a bit an then i'll upload it. 

And then we need someone who writes a graphical web app configuration tool, as for example i've never used the sleep timer and need more the volume limiter and button disabler ;-)

zxa, 2018-07-04:
Hi @MiczFlor, 

it's all here now:

https://github.com/zxa/RPi-Jukebox-RFID/tree/web-app-settings-page

Should i do a pull request or do you take it from there for testing?

MiczFlor, 2018-07-05:
Please create a pull request. Thanks

closed 2018-07-06 by MiczFlor