MiczFlor/RPi-Jukebox-RFID #74: order audio by type

Luegengladiator, 2018-07-10:
From the code it's possible to order the type of audio in subfolders:

find . -type d | grep shared
./shared/audiofolders
./shared/audiofolders/weblinks
./shared/audiofolders/weblinks/ZZZ-YouTube-Link1
./shared/audiofolders/weblinks/ZZZ-YouTube-Link2
./shared/audiofolders/audiobooks
./shared/audiofolders/audiobooks/Audiobook1
./shared/audiofolders/audiobooks/Audiobook2
./shared/audiofolders/audiobooks/Audiobook3
./shared/audiofolders/podcasts
./shared/audiofolders/podcasts/ZZZ-Podcast-1
./shared/audiofolders/podcasts/ZZZ-Podcast-2
./shared/audiofolders/music
./shared/audiofolders/music/music1
./shared/audiofolders/livestreams
./shared/audiofolders/livestreams/ZZZ-Webstream1

you have to create the same folderstructure in playlists

find . -type d | grep playlists
./playlists
./playlists/audiobooks
./playlists/livestream
./playlists/podcasts
./playlists/music
./playlists/webstream

to store the m3u. 

Now it's possible to recreate or keep the m3u by type of audio.
You want to recreate it on podcast or store it in audiobooks.
There is no need to change the code. Only remember to create the shortcut with the specific subfolder:

cat 5000112547726
audiobooks/Audiobook1

cat 4045145288001
music/music1

zxa, 2018-07-10:
Hi @Luegengladiator 

out of curiosity: what problem are you tackling with this?

Luegengladiator, 2018-07-10:
There was a discussion to use m3u to provide podcast (rebuild every run) and store position of played context inside the m3u. 
Therefore we have the need to decide which content is played. 
I described a way to do this an easy way.
It's only a step on the way to get the m3u handling working

Luegengladiator, 2018-07-10:
stored positions are direct linked to the mp3 folder. 
Recreation of m3u on the folder maybe ends up in broken stored positions: 
Example:
your m3u is created on the audiolist file1 file3 file4
Stored position is now track 2 second 125.
you add file2 to the folder recreate the m3u. stored position is now a different audiofile.
So you have to delete (or recalculate) the stored positions if you recreate the m3u which is automatically done if you store the position inside the m3u.

zxa, 2018-07-10:
Does this only apply to mpg123/conplay or does vlc also store positions (i don't see any change in the m3u with vlc).

Luegengladiator, 2018-07-10:
at the moment this is a mpg123 issue only. 
Is there a way to store position with vlc?

Luegengladiator, 2018-07-10:
you are right. I used mpg123 to get the storing feature and its nice to use. 
Closed the Pull request because i have to do a rebase on my fork.

There is a "new" feature of playing web podcasts which needs to rebuild the m3u every time it is triggert, because you don't know the content of the cast. @MiczFlor mentioned it in the pull request.

I have to figure out whether mpg123 is able to play webstreams. But for me it's was a minor issue so i tried to fix the "bigger" problems like playlist replay or track replay in a nice way and one solution is to store streams in a dedicated folder ;)

zxa, 2018-07-10:
Yes, but there's also the podcast.txt or livestream.txt in the folder you can check for.

MiczFlor, 2018-07-10:
Hi @Luegengladiator @zxa 

good to see a little crowd in this thread :) the implications of this decision are not trivial, so I appreciate a variety of opinions.

A couple of thoughts from my side
* I would like to keep this as simple as possible. The more meta info we need about the "type of content" the more twists and jumps we need to make at a later stage.
* If we need to treat some content differently, what's missing now to do so? We already have the information for podcasts inside the podcast folder (e.g. the podcast.txt file).
* Would it be better to think about a new player alongside VLC? What about `mpd` to play content and playlists?

From what I understand for the time being, we need to store filename and time(position) in file for all folders that contain only audio files (== !*.txt file)?

I think audio books is a big feature that would really benefit the code base if it was included in a good way.

Luegengladiator, 2018-07-10:
Ok I see the point. I'll do my rebase to have a look at the new features. After that I'll give podcasts and Webstreams a try. Didn't have a look at that with mpg123 but from a brief overview it should work.
Let's do it step by step so we have the chance to get all the benefits. 
I'll get my new hardware (pi zero w) today so I can crosscheck #15 and do the rebase in one step. 
If this is ok we can close this one.

MiczFlor, 2018-07-10:
Looking at the users again, I came to think that my daughter

* always wants to start her fairy tales from the beginning
* not necessarily listens to the end but often swipes something else half way through
(I can't remember how I did this with tapes when I was a kid. Vinyl would definitely start from scratch - no pun intended)

Over to grown ups: as an avid podcast user, I really like the fact that all episodes on my podcast app store where I am. I would not want to start from the beginning. And the same is true for audio books, of course.

So if I would share the same Phoniebox with my daughter and listen to my audio books, wouldn't I be the one to want the "extra feature" to store the position on stop or shutdown. And I would possibly either do that using the web app or I could also see a "grown up" card that saves the position and stops on swipe.

What I am wondering, following this: even if we separate the content in "audio file playlist" versus "something else" and can save the position alongside the "audio file playlist", we might still have the problem that the content type does not describe the desired behaviour "on stop", because the desire differs from user to user.

I am sure we will find a way to store track name and position, but then the related question is (as described above) is this "always" happening? Or is there a special card to trigger that special behaviour?

MiczFlor, 2018-07-10:
@Luegengladiator let's leave this thread open for a few days. Not sure how we best describe the roadmap in github. 

Possibly we can start using projects? https://github.com/MiczFlor/RPi-Jukebox-RFID/projects Does anybody have experience with this?

But it would be good to keep this open until we have all we need to make a good roadmap.

Since you are itching :) I think the pull request for the bar scanner might be floating to the top?

zxa, 2018-07-11:
@MiczFlor @Luegengladiator  i also made the experience, that kids wants to listen to their audio books from the beginning. I would make that as an configurable option for every audio folder, which can be set via the app (just put a file into that folder with some text maybe, default is "off")

But the more important project in my eyes is to settle on a player. I've checked MPD with the MPC command line interface and it seem to me the best option:
- only up to 10 % CPU usage on a Zero W while playing (VLC uses 30-40 %)
- plays various types of files (like VLC, as mpg123 only plays mp3)
- has a clean and easy to "grep" status output with more information than VLC

Downsides:
- has to be configured before usage
- playing Youtube requires some extra work (i would use youtube-dl for that with the advantage of having the Youtube audio offline, but even online use seems to be possible, but i haven't looked deeper)

I could tinker a first demo with mpd.

I haven't used "projects" before, but maybe it's worth to try out for that. Maybe we should also have a "dev" branch where we can put more experimental features.

Luegengladiator, 2018-07-11:
I'm fine with that.
But please keep in mind: 
kids will grow up and there is a time they don't want to hear small audio dramas. 
There will be a time they want to hear audiobooks with hours and hours of playtime. 
And than it's not an option to restart every time. 
Example: Harry Potter and the Philosopher's Stone 8 h 47 min 0 s
Harry Potter and the Deathly Hallows 23 h 59 Min. 

So I think at least on this use case resume play (even days later with playing other playlist in the meantime) is a must have.

zxa, 2018-07-11:
Yes, i also think that it is a good and needed feature. MPD doesn't behave other than VLC (i forgot that in my list) regarding storing positions. So we need a solution like i described above, a problem which i think is solvable. I prefer mpd because it solves a problem we can't influence: the high cpu load of vlc. And it plays a great variety of audio formats as like vlc, so we don't loose anything there.

(I'm selfish here, having my powerbank and OGG-files in mind ;-) )

Luegengladiator, 2018-07-11:
mpd has different features i like. 
There are many clients (web, mobile, desktop) to control the daemon and to administrate the audio-repository. So rfid or ean are optional features.

On my way home I talked to a colleague and he mentioned "ogg123" as a lightweight ogg player and "file" to decide which player is used. 

If we get youtube, webstream and podcast running, we maybe have a very light solution to run mpeg and ogg audio files?

MiczFlor, 2018-07-12:
While we are still brainstorming... I would like to avoid mixing players, possibly mpd is a good candidate?

I am also a big fan of CMUS https://cmus.github.io/ which also comes with a command line UI: https://github.com/cmus/cmus/blob/master/Doc/cmus-remote.txt 

Playing YouTube files is a nice feature, but I am not sure if it is a must have. I would much rather have a longer list of supported audio formats like ogg, mp3, flac, aiff, wav, wma (I know, I know...), aac

Other player candidates I came across in a quick search:

* Audacious https://www.systutorials.com/docs/linux/man/1-audtty/

Luegengladiator, 2018-07-12:
At the moment Systemsounds like Startup and Shutdown are played with mpg123. All the Files are played with vlc. So there is already a mix. 
To figure out what the player should provide we must define what are the features needed.
I got following:
1) Audioformats: ogg, mp3, flac, aiff, wav, wma, aac                 high priority
2) low resource utilization to get the most out of Powerbank     high priority
3) Commandlineinterface                                                           high priority
4) fast startup to avoid impatiens                                         medium priority
5) resumeplay per Playlist to support long Tracks/Lists       medium priority

MiczFlor, 2018-07-13:
Audacious seems to do a lot of things via command line that we need:
~~~
$ audtool help
Vital information:
   current-song                       - returns current song title
   current-song-filename              - returns current song filename
   current-song-length                - returns current song length
   current-song-length-seconds        - returns current song length in seconds
   current-song-length-frames         - returns current song length in frames
   current-song-output-length         - returns current song output length
   current-song-output-length-seconds - returns current song output length in seconds
   current-song-output-length-frames  - returns current song output length in frames
   current-song-bitrate               - returns current song bitrate in bits per second
   current-song-bitrate-kbps          - returns current song bitrate in kilobits per second
   current-song-frequency             - returns current song frequency in hertz
   current-song-frequency-khz         - returns current song frequency in kilohertz
   current-song-channels              - returns current song channels
   current-song-tuple-data            - returns the value of a tuple field for the current song
   current-song-info                  - returns current song bitrate, frequency and channels

Playlist manipulation:
   playlist-advance                   - go to the next song in the playlist
   playlist-auto-advance-status       - returns the status of playlist auto-advance
   playlist-auto-advance-toggle       - toggles playlist auto-advance
   playlist-reverse                   - go to the previous song in the playlist
   playlist-addurl                    - adds a URL to the playlist
   playlist-insurl                    - inserts a URL at specified position in the playlist
   playlist-addurl-to-new-playlist    - adds a URL to the newly created playlist
   playlist-delete                    - deletes a song from the playlist
   playlist-length                    - returns the total length of the playlist
   playlist-song                      - returns the title of a song in the playlist
   playlist-song-filename             - returns the filename of a song in the playlist
   playlist-song-length               - returns the length of a song in the playlist
   playlist-song-length-seconds       - returns the length of a song in the playlist in seconds
   playlist-song-length-frames        - returns the length of a song in the playlist in frames
   playlist-display                   - returns the entire playlist
   playlist-position                  - returns the position in the playlist
   playlist-jump                      - jumps to a position in the playlist
   playlist-clear                     - clears the playlist
   playlist-repeat-status             - returns the status of playlist repeat
   playlist-repeat-toggle             - toggles playlist repeat
   playlist-shuffle-status            - returns the status of playlist shuffle
   playlist-shuffle-toggle            - toggles playlist shuffle
   playlist-stop-after-status         - queries if stopping after current song
   playlist-stop-after-toggle         - toggles if stopping after current song
   playlist-tuple-data                - returns the value of a tuple field for a song in the playlist

More playlist manipulation:
   number-of-playlists                - returns the number of open playlists
   current-playlist                   - returns the number of the active playlist
   set-current-playlist               - activates the specified playlist
   current-playlist-name              - returns the title of the active playlist
   set-current-playlist-name          - sets the title of the active playlist
   new-playlist                       - creates a new playlist
   delete-current-playlist            - removes the active playlist
   play-current-playlist              - starts or resumes playing the active playlist

Playqueue manipulation:
   playqueue-add                      - adds a song to the playqueue
   playqueue-remove                   - removes a song from the playqueue
   playqueue-is-queued                - returns OK if a song is queued
   playqueue-get-queue-position       - returns the playqueue position of a song in the given position in the playlist
   playqueue-get-list-position        - returns the playlist position of a song in the given position in the playqueue
   playqueue-length                   - returns the length of the playqueue
   playqueue-display                  - returns a list of currently-queued songs
   playqueue-clear                    - clears the playqueue

Playback manipulation:
   playback-play                      - starts/unpauses song playback
   playback-pause                     - (un)pauses song playback
[... 13 lines cut ...]
Equalizer manipulation:
   equalizer-activate                 - activates/deactivates the equalizer
   equalizer-get                      - gets the equalizer settings
   equalizer-set                      - sets the equalizer settings
   equalizer-get-preamp               - gets the equalizer pre-amplification
   equalizer-set-preamp               - sets the equalizer pre-amplification
   equalizer-get-band                 - gets the equalizer value in specified band
   equalizer-set-band                 - sets the equalizer value in the specified band

Miscellaneous:
   mainwin-show                       - shows/hides the main window
[... cut ...]
   preferences-show                   - shows/hides the preferences window
   about-show                         - shows/hides the about window
   version                            - shows Audacious version
   shutdown                           - shuts down Audacious

Help system:
   list-handlers                      - shows handlers list
   help                               - shows handlers list

Handlers may be prefixed with `--' (GNU-style long-options) or not, your choice.
Report bugs to http://redmine.audacious-media-player.org/
~~~

zxa, 2018-07-13:
I couldn't get a beep out of it as i tested it:

$ audacious -H
ERROR dbus-server.cc:808 [dbus_server_init]: D-Bus error: Cannot autolaunch D-Bus without X11 $DISPLAY
$ audtool
D-Bus error: Cannot autolaunch D-Bus without X11 $DISPLAY

I haven't found a solution for that in a quick search. 9 years ago it was the way to go:
http://blog.spiralofhope.com/232/headless-audacious-audtool-howto.html

As a headless mode which works only with X11 seems weird, i'm assuming i messed something up.

Luegengladiator, 2018-07-27:
with #79 i think its useful to order a big amount of Music and/or audiofiles in Subfolders different to the RFID. MPD has a own Database with all the Information in it. 
For example: 
I ordered my music files alphabetic by artist in Subfolders A - Z. 
I have a Playlist folder where a playlist "links" the audiofiles in the subfolders.

closed 2018-08-02 by MiczFlor