MiczFlor/RPi-Jukebox-RFID #76: Docs: listing simple controls for USB sound device

marians, 2018-07-11:
First, thanks for all the work that went into Phoniebox! I just built one as a birthday gift for my daughter and can't wait to see her test it out.

---

I am using the latest Raspbian stretch.

```
$ uname -a
Linux raspberrypi 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch
```

My goal was to use a USB sound device instead of the Raspberry's built-in one. I got an output, but at first I could not set the volume.

My `/home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name` file had the value `PCM` in it, so I checked the docs at

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/docs/CONFIGURE-stretch.md#create-settings-for-audio-playout

and saw the comment about listing the available iface names using `amixer controls`. For me, this results in

```
$ amixer controls
numid=3,iface=MIXER,name='Mic Playback Switch'
numid=4,iface=MIXER,name='Mic Playback Volume'
numid=7,iface=MIXER,name='Mic Capture Switch'
numid=8,iface=MIXER,name='Mic Capture Volume'
numid=9,iface=MIXER,name='Auto Gain Control'
numid=5,iface=MIXER,name='Speaker Playback Switch'
numid=6,iface=MIXER,name='Speaker Playback Volume'
numid=2,iface=PCM,name='Capture Channel Map'
numid=1,iface=PCM,name='Playback Channel Map'
```

Unfortunately, it is not really obvious, which names of these would be usable as the iface name. I tried using the value `MIXER`, without success.

I ended up checking the `amixer` help and found the `amixer scontrols` command, which lists these results:

```
Simple mixer control 'Speaker',0
Simple mixer control 'Mic',0
Simple mixer control 'Auto Gain Control',0
```

So I tried the value `Speaker` in the file `/home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name` and it worked immediately.

Not sure if this is specific to the versions I used or for the sound device. However I think instead of `amixer controls` the docs should recommend `amixer scontrols`.

Another unrelated note: I recognise no difference in quality between the 6 EUR USB sound device and the Raspberry 3 Model B+ built-in one. Both are prone to CPU noise.

MiczFlor, 2018-07-12:
Hi @marians 

I have the same issue with the USB devices. The original links and information still comes from my very first research into this, I admit... Possibly time for an update and a proper benchmarking.

I did not know about `scontrols`, thanks for that. I added your docs tweak:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/63f1f3f027af18f96c6918d1e0ba34d799408c1c

All the best, micz

closed 2018-07-12 by MiczFlor

marians, 2018-07-12:
Nice, thanks!