MiczFlor/RPi-Jukebox-RFID #78: Headless installation/configuration via SSH

marians, 2018-07-12:
As I don't have an extra keyboard around, I was looking for a way to set up the Phoniebox via SSH. Which of course also saves the need for a monitor and a mouse.

The following worked for me with Raspian stretch. Maybe you find the time to add a description for this in the right place(s):

1. Flash your SD card with the Raspian image
2. Eject the card and insert it again. This should get the `boot` partition mounted.
3. In the boot partition, create a new empty file called `ssh`. This will enable the SSH server later.
4. Create another file in the same place called `wpa_supplicant.conf`. Set the file content according to the following example:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
network={
    ssid="YOUR_NETWORK_NAME"
    psk="YOUR_PASSWORD"
    key_mgmt=WPA-PSK
}
```

Note: This works for WPA-secured wifi networks, which should be the vast majority.

5. Save the file
6. Unmount and eject the card, insert it into the Raspy, boot.
7. Find out the IP address of the raspberry. Most Wifi routers have a user interface that lists all devices in the network with the IP address they got assigned.
8. Connect via ssh with username `pi` and password `raspberry`.

Source:
- https://howchoo.com/g/ndy1zte2yjn/how-to-set-up-wifi-on-your-raspberry-pi-without-ethernet
- https://howchoo.com/g/ote0ywmzywj/how-to-enable-ssh-on-raspbian-without-a-screen

MiczFlor, 2018-07-12:
HI @marians 

Thanks once more :) I added your prose almost identically:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/d62293bde0aa9d62b6cb8aa66486c65d3b020f11

closed 2018-07-12 by MiczFlor

marians, 2018-07-12:
Cool!