MiczFlor/RPi-Jukebox-RFID #79: mpd as player with resume play

zxa, 2018-07-14:
I experimented with mpd in the last days and it showed to be a good replacement for vlc. As i was running vlc "daemon-like" with playlist juggling before, i had to make only small adjustments. And then it escalated a little bit and i added resume play to it.

### mpd and mpc

mpd can be controlled like vlc with an rc interface (nc localhost 6600). The available commands can be found here: https://www.musicpd.org/doc/protocol/command_reference.html
I also use mpc, the command line client. It has less commands available but it's easier to write "mpc play" than "echo play | nc localhost 6600". So it could be completely replaced.

### Resume play

I added a new script, resume_play.sh, which is called from playout_controls with the option -c=savepos everytime when there is a stop, shutdown or load-new-playlist command. It saves the filename of the current track and time to a file named lastplayed.dat in to the corresponding folder ONLY if there's already a file named lastplayed.dat. This is done to control resume play for different audio folders how you desire.
This can be set in the web app, where i added a status marker to the folders in "Available Audio" and a control button (expand the accordion). This button simply writes or deletes the lastplayed.dat file in the folder. There is no button for livestreams as it doesn't make sense.
To resume the old position the script is called with -c=resume and looks for the lastplayed.dat and tries to find the audio file indicated there in the already loaded playlist (this is done for podcast support). If it is found, it resumes, otherwise it starts the playlist from the beginning.
One issue i haven't found a solution (yet?) is, if you resume playing in the middle of the playlist and you listen until the end, next time you start that playlist it will start from the same position in the middle of the playlist. This is because there's no event i could use to call the resume_play.sh if the playlist ends. But maybe mpd offers something, i haven't looked very deep into the possibilities.

### Setup

If you want to try out, it's maybe better to backup the files i've touched:

cp ~/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh  	~/RPi-Jukebox-RFID/scripts/rfid_trigger_play.sh.vlc
cp ~/RPi-Jukebox-RFID/scripts/playout_controls.sh	~/RPi-Jukebox-RFID/scripts/playout_controls.sh.vlc
cp ~/RPi-Jukebox-RFID/scripts/idle-watchdog.sh		~/RPi-Jukebox-RFID/scripts/idle-watchdog.sh.vlc
cp ~/RPi-Jukebox-RFID/htdocs/inc.controlPlayer.php	~/RPi-Jukebox-RFID/htdocs/inc.controlPlayer.php.vlc
cp ~/RPi-Jukebox-RFID/htdocs/inc.header.php		~/RPi-Jukebox-RFID/htdocs/inc.header.php.vlc
cp ~/RPi-Jukebox-RFID/htdocs/inc.vlcStatus.php		~/RPi-Jukebox-RFID/htdocs/inc.vlcStatus.php.vlc
cp ~/RPi-Jukebox-RFID/htdocs/index.php			~/RPi-Jukebox-RFID/htdocs/index.php.vlc

Install mpd and mpc:

`sudo apt install mpd mpc`

Configure mpd:

`sudo nano /etc/mpd.conf`

Find these options and change to following:

music_directory                "/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
playlist_directory             "/tmp"
user "root"

and maybe 

mixer_control   "yourAudioIfaceNameHere"
i had to change it to Master, but as it defaults to PCM, it should work for the most without change.

### Regressions

There's no youtube support yet, but i can be added with youtube-dl.
Another thing my test user uncovered is that you can't skip from the first track to the last track of the playlist via "previous track". This behaviour is different than in vlc.

MiczFlor, 2018-07-16:
Hi @zxa 
thanks for the quick work. I got it to work, but have to start from scratch to write the exact install procedure. There was a `mpc update` at some point.

On the Raspberry Pi 3 mpd uses 3% CPU power.
VLC uses 15% CPU power, which is 5 times more. Not sure what that means for battery power. But it should make a difference.

What's the difference on your machine?

If we do the switch to `mpd` I guess that this should be with a version jump to 1.0 

I find it difficult at the moment to see a configurable version that supports `mpd` `vlc` and `mpg123`

In my heart though, I would like to have a master branch which does not kill running Phonieboxes on `git pull` because it will continue to run `vlc` while it adds the option to switch to an alternative player (requiring the apt-get install and mpd config, of course)

What are your thoughts?

MiczFlor, 2018-07-16:
Hi @zxa 

one other big advantage of `mpd` comes to mind: it might be possible to drop running many things as `root`. Currently this is the case because (from what I remember) the web app required it and there was something about VLC. However, `mpd` might make it easier to move away from *running as root* which is a nice feature of Raspian, I admit :)

Then the Phoniebox might extend its reach also to other platforms. Even without the RFID feature, the current web app is a treat and good enough for a standalone?

zxa, 2018-07-16:
Hi @MiczFlor 

> There was a mpc update at some point.

You are right, i forgot about that. There are two ways to solve that:
- uncomment the line `auto_update "yes"` in mpd.conf
- or we use absolute pathes in the playlists, but as they have to start with file:// i was to lazy to do that.
I updated my comment above.

> On the Raspberry Pi 3 mpd uses 3% CPU power. VLC uses 15% CPU power, which is 5 times more. Not sure what that means for battery power. 

I don't know either. On the Zero W the CPU load is approx three times higher. The main difference is that at the beginning of a playlist VLC produces an awfuf stuttering sound for a few seconds maybe because the CPU of the Zero W is at 100% in that time. I tested that on two boxes (pictures coming soon ;-) ).

> I find it difficult at the moment to see a configurable version that supports mpd vlc and mpg123

I wouldn't do a mpg123 version. It has the same CPU load on my machine and lacks a lot of features of mpd, especially the codec support. So, in my opinion, we gain nothing with mpg123.

VLC has only one advantage left, the Youtube support (which isn't usable on Zero Ws). mpd can't do that and i would use youtube-dl to retain some kind of Youtube support. Although there seem to be the possibility to stream Youtube via youtube-dl and ffmppeg to mpd, i wouldn't do that as youtube-dl needs up to 10 seconds to start on a Raspberry and i imagine that the needed live converting with ffmpeg will set your Pi in to flames ;-)
Thatswhy i think it's the better way to download the Youtube audio with youtube-dl and have it offline on your pi. Or is there a use case for a Phoniebox where you need a Youtube live stream?

If there's no major show stopper discovered, i would drop vlc in the future. The rc-interface is to limited in my eyes (for example sometimes my test user likes to listen to a song over and over again, and a "single repeat" feature would be nice. There's this command in the rc-interface of vlc, but there is no output if you have enabled it or not. mpd's status output is way more talkative.)

> In my heart though, I would like to have a master branch which does not kill running Phonieboxes on git pull because it will continue to run vlc while it adds the option to switch to an alternative player (requiring the apt-get install and mpd config, of course)

I don't know if people read the release notes before upgrades, if they do upgrades at all. Maybe you see more in the traffic graph. Is it possible on Github to set an other branch as the main branch (not "master") from which the readme.md is shown? Then we could retire the master branch and start a new one. But i'm not that into git.

> one other big advantage of mpd comes to mind: it might be possible to drop running many things as root.

Maybe. It seem to work well as the user "mpd", only the resume feature is broken then (i guess it's because it requires to delete old playlists, but i want to change this anyway).

Luegengladiator, 2018-07-23:
Good work. I'll give it a try now on my new RPi3. If I'll get it to work I'll test in on the RPi1B+ too. 
I think there can be a feedback in 2 or 3 days.
Because of the Client/Server architecture of mpd there are different Clients for Windows, Linux, Android and Apple. Nice if you want to shut down the childrens phonie from the mobile device...

zxa, 2018-07-26:
Hi @Luegengladiator

Which mpd client are you using? Maybe it's something for the documentation if someone wants to use an external client.

Luegengladiator, 2018-07-26:
Hi @zxa 
I'm using gmpc but thats only a personal choice. 
There is a wide variety of clients. Please have a look here: https://www.musicpd.org/clients/
MPD is the Server and MPC or GMPC is the client communicating with the Server locally or remote. 
Communication is on Port 6600 but to communicate with the Server the Server has to listen on the Network Interface (10.*.*.*, 192.168.*.*  or 172.16-32.*.*). NOT on localhost or 127.0.0.1

Luegengladiator, 2018-07-26:
I did some testing now on the mpd Version and it is BEATUIFUL! 
I really like it.
Thanks a lot for that!

At the moment I'm thinking about linking the phonie to my Stereo having RFIDs for the kids and the background Client gmpc on PC Devices and MPDroid on Mobile Devices for parents. 
On "audiofolders" I mounted my NAS Music-folder and on "audiofolders/audiobooks" i mounted NAS audiobook-folder. So it's simple to define Parent Playlists with all the Music from the Client and "hard coded" RFID Playlists for the Kids. All running on one RPi 3. To support the subfolders, I did some path-recalculation for the temp m3u. I'll bring it to git if the pull-request #94 is finished.

Luegengladiator, 2018-08-14:
@zxa 
I played around with a special RFID to restart the actual Playing List, not only the actual playing track.
I did it in playout_control with:
    playlistreset)
        mpc play 1
        ;; 

and linked it at the moment to a special RFID but will link it to a gpio in future. 
Do you want to add it to the code?

MiczFlor, 2018-08-16:
I am closing this ticket, because of the launch of version 1.0 on master. If you have any issues, please try first if they are still happening on the latest version 1.0. If so, open another ticket, please. 

* Use this thread for update issues [if you are running `stretch`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/138) 

**The version 1.0 has not been tested on `jessie`**

* Read this thread [if you are running `jessie`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/137) 

Version 1.0 is a major improvement (which will be full of bugs, for sure :) and I want to thank all the contributors and Phoniebox lovers with their input, pictures, bug fixes and suggestions.

closed 2018-08-16 by MiczFlor