MiczFlor/RPi-Jukebox-RFID #83: bug with mpd player: playernext works from web app but not with RFID chip

MiczFlor, 2018-07-18:
Hi @zxa 
I need some help here. Testing the `mpd` version, I am running into a strange problem. Using the web app, I can skip forward and backward in the playlist, no problem. When trying the same with the RFID chip, skipping to the next track plays approx. half a second of the next track, then the playout stops.

To narrow down the problem, I tried to trigger the process from the command line. Below you see that when I call `playout_controls.sh -c=playernext` mpc tells me a file name, that it starts playing and the information - as expected.
~~~
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ ./playout_controls.sh -c=playernext
froschk1/Froschkoenig-04-MONO.mp3
[playing] #4/10   0:00/5:10 (0%)
volume: 78%   repeat: off   random: off   single: off   consume: off
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $
~~~
When I call `rfid_trigger_play.sh -c=0008021289` which then calls (correctly) `/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playernext` it starts the same process. Then it throws a second line of player information and at the same time stops the playout.
~~~
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $ ./rfid_trigger_play.sh -c=0008021289
from rfid_trigger_play.sh calling /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=playernext
froschk1/Froschkoenig-05-MONO.mp3
[playing] #5/10   0:00/8:10 (0%)
volume: 78%   repeat: off   random: off   single: off   consume: off
volume: 78%   repeat: off   random: off   single: off   consume: off
pi@raspberrypi:~/RPi-Jukebox-RFID/scripts $
~~~
I tried a couple of things to narrow down the issue

* makes no difference if I call `playout_controls.sh -c=playernext` with sudo or without
* I added `sleep 3` before and after `mpc next` in `playout_controls.sh`. This is when it's getting weird. The resulting behaviour:
* * `sleep 3`: before the skip, there is an expected 3 second delay
* * `mpc next`: throws the same information as if I call `./playout_controls.sh -c=playernext` which is desired behaviour
* * `sleep 3`: another three seconds pass and then...
* * ... the additional line comes up and the playout stops.

Can you reproduce this error?

MiczFlor, 2018-07-18:
Ah, found it. Towards the end of `rfid_trigger_play.sh` come the commands
~~~

    # Save position (to catch playing and paused audio) for resume and clear the playlist -> audio off
    # Is has to be sudo as daemon_rfid_reader.py doesn't call this script with sudo
    # and this produces an error while saving lastplayed.dat

    sudo $PATHDATA/playout_controls.sh -c=playlistclear
	
    # Before we create a new playlist, we remove the old one from the folder.
    # It's a workaround for resume playing as mpd doesn't know how its current playlist is named,
    # so we only want the current playlist in the "mpc lsplaylists" output.

    mpc lsplaylists | \
    while read i
    do
        mpc rm "$i"
    done
~~~
they need to be **inside** the `if [ "$FOLDERNAME" ]; then` check. Fixed it.

closed 2018-07-18 by MiczFlor

zxa, 2018-07-18:
Yes, i've never used a special card. The code could be even in the next "if" block starting with 

`if [ -d "$PATHDATA/../shared/audiofolders/$FOLDERNAME" ]`

I feared it's something weird, like the playlist sorting (aka battle of the birds):

Web app puts "Alle meine Entchen" at the first position in the playlist, if i trigger the same folder with RFID-card, number one goes to "Alle Vöglein sind schon da".