MiczFlor/RPi-Jukebox-RFID #85: Podcast *.xml

Bartman83, 2018-07-19:
I tried to add a podcast-link to my jukebox: https://feeds.br.de/betthupferl/feed.xml
Unfortunately it doesn´t work. I found out that my jukebox have a problem with *.xml, because it works with *.podcast (https://www1.wdr.de/kinder/radio/kiraka/hoeren/podcast/kinderhoerspiel-podcast-108.podcast) fine.

Maybe somebody can help me?!

Greets

MiczFlor, 2018-07-19:
HI @Bartman83 

nice catch. Thank you for the links to start with :) and regarding the grabbing of the file URLs, I have to spend some time on that issue (which I won't be able to in the coming two weeks). Thing is: it has nothing to do with the file ending. The way the podcasts are grabbed is by looking into the `enclosure` tag of the feed and grabbing the string inside the `url=` attribute.

I had a quick look using the regular expression in my script. Here the one that works, you can try that in the command line of the RPi. First: get the feed and pipe it through the regular expression:
~~~
wget -q -O - "https://www1.wdr.de/kinder/radio/kiraka/hoeren/podcast/kinderhoerspiel-podcast-108.podcast" | sed -n 's/.*enclosure.*url="\([^"]*\)" .*/\1/p' > "kinderhoerspiel-podcast-108.podcast.m3u"
~~~
Then you can take a look at the file URLs in the original feed by looking at the resulting file `kinderhoerspiel-podcast-108.podcast.m3u `:
~~~
cat kinderhoerspiel-podcast-108.podcast.m3u 
~~~
In the original feed (and I am writing this in detail, because somebody else might improve the expression in my absence :) the `enclosure` tag looks like this:
~~~
<enclosure url="https://wdrmedien-a.akamaihd.net/medp/podcast/weltweit/fsk0/168/1682380/kirakakinderhoerspielimwdr_2018-07-07_waldorfschulemehralsnurdennamentanzenradiogeschichte_kiraka.mp3" length="60587156" type="audio/mpeg"/>
~~~
In the feed that does not work, the enclosure tag looks like this:
~~~
<enclosure length="4960768" type="audio/mpeg" url="https://cdn-storage.br.de/iLCpbHJGNL9zu6i6NL97bmWH_-bg/_-TS/_AFf52xG9U1S/180719_1955_Betthupferl_Missy-und-der-Drache-Pit-Peperoni---Fremde-.mp3"/>
~~~
Kind of the same, but the `url` attribute is further down the line. This should not be a problem. Let's find out. Instead of `url`, let's try to get all the `length` values, which are right after the `enclosure` tag. This works for me, try it:
~~~
wget -q -O - "https://feeds.br.de/betthupferl/feed.xml" | sed -n 's/.*enclosure.*length="\([^"]*\)" .*/\1/p' > "betthupferl-feed.xml.m3u"
cat betthupferl-feed.xml.m3u
~~~
Trying the same with the second attribute `type` also works:
~~~
wget -q -O - "https://feeds.br.de/betthupferl/feed.xml" | sed -n 's/.*enclosure.*type="\([^"]*\)" .*/\1/p' > "betthupferl-feed.xml.m3u"
cat betthupferl-feed.xml.m3u
~~~
So it does not seem to be the position in the tag, but might be a special char that needs escaping in the URL. The regex should grab anything but `"`. However, it seems to stall somewhere within the URL string. Anyone any idea?

Good night and good luck :)

zxa, 2018-07-19:
It's the single space character before the dot which is the problem. The regex matches for everything between url=" and " (" plus a space). 
`'s/.*enclosure.*url="\([^"]*\)" .*/\1/p'`
With removing that space, it may also match for cases like .mp3"/>

MiczFlor, 2018-07-20:
thanks to @zxa for reminding us that nothing can mean something or - in other words - nothing can make such a difference.
I merged the pull requests.

@Bartman83 here is what you need to do:

If you got the code base via `git clone` you need to go to the Phoniebox root dir and type `git pull` and make sure the shell scripts are executable. Something like this:
~~~
cd /home/pi/RPi-Jukebox-RFID/
git pull
cd scripts
chmod +x *.sh
~~~

If you downloaded the ZIP, you need to get the updated file via `wget`. Now you must know: are you using the `mpd` branch `pr80-mpd-as-audio-player` or the `master` branch using the `VLC` player. If you are unsure, try `master` first.

This is how you do it for `master`:
~~~
cd /home/pi/RPi-Jukebox-RFID/scripts/
wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/rfid_trigger_play.sh
chmod +x rfid_trigger_play.sh
~~~
This is how you do it for `pr80-mpd-as-audio-player/scripts`
~~~
cd /home/pi/RPi-Jukebox-RFID/scripts/
https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/pr80-mpd-as-audio-player/scripts/rfid_trigger_play.sh
chmod +x rfid_trigger_play.sh
~~~

closed 2018-07-20 by MiczFlor