MiczFlor/RPi-Jukebox-RFID #91: add audio recording?

andimanuel, 2018-07-22:
Hey there, I just got the basic box running (btw, thanks for the awesome project!) and want to add audio recording.
I'm a noob at Linux, but so far I've managed to record a set duration (30s for testing) via a key-fob with these small copy/paste additions:

in rfid_trigger_play.conf
`RECORD="0016007223"`
in rfid_trigger_play.sh
`elif [ "$CARDID" == "$RECORD" ]
then
    # record audio
    $PATHDATA/playout_controls.sh -c=record`

in playout_controls.sh
```
elif [ "$COMMAND" == "record" ]
then
    # kill vlc, then record mono ogg to folder recordings
    sudo pkill vlc
    arecord -f S16_LE -c1 -r44100 -d 30 | oggenc -b 128 - -o $PATHDATA/../shared/audiofolders/recordings/`date +%Y%m%d%H%M%S`.ogg
```
The files can then be played with the card linked to "recordings".

I'd like to be able to stop the recording by using the key fob again, what's the best way to check whether I'm recording right now? Check for the arecord process and kill it if it's running, or work with a temp file?

I'll also have to add a LED that's on while recording. Will try to figure that out later...

MiczFlor, 2018-07-23:
Hi @andimanuel,

Nice idea and a good start. What‘s the use case?

My thoughts regarding your feature

* you might want to start a kill process parallel to the recording. In case you or somebody forgets to stop the recording, your sd card will fill up. Take a look at the sleeptimer function for inspiration. 
* if the modes are only „recording on“ vs „recording off“ it would be best to use the process - if you can. And not a temp file. The toggle would then be: if card { if recording on } { else}.
* could it happen that the Pi runs out of battery and after boot will continue recording undetected? If so, the recording should be killed on boot, just in case. 

I am on holidays. When i am back, i am happy to beta test your first pull request :)

andimanuel, 2018-07-27:
Thanks for the feedback, my use case is just having a simple way of recording voice & music, either for myself or my key user. ;)  He's probably too young for now, but I'm seeing the phoniebox kind of as a modern cassette recorder, so recording is the one part that was missing.

Here's an improvement on the playout_controls.sh:
```
...
elif [ "$COMMAND" == "record" ]
then
    # if arecord isn't running yet, kill vlc, then record mono ogg to directory recordings
    if [ -z "$(ps -A | grep arecord)" ]; then
        sudo pkill vlc
        arecord -f S16_LE -c1 -r44100 -d 300 | oggenc -b 128 - -o $PATHDATA/../shared/audiofolders/recordings/`date +%Y%m%d%H%M%S`.ogg &
    else  #stop recording
        sudo pkill arecord
    fi
...
```
On the parallel kill process - arecord has a duration set by -d. I guess that should work the same way?
It's set to 5 mins now, this max. duration should maybe be accessible as a variable in the settings?
I'll have to look into how the pull request thing works... :)

MiczFlor, 2018-07-28:
Looking forward to your pull request :) In the README file I linked this small howto: https://www.digitalocean.com/community/tutorials/how-to-create-a-pull-request-on-github

MiczFlor, 2018-09-19:
Hi,
I am closing this issue, because it has been silent for over two weeks. If it's not resolved, please open a new issue. Thanks.
All the best, micz

closed 2018-09-19 by MiczFlor