MiczFlor/RPi-Jukebox-RFID #92: auto shutdown, without function

webjochen, 2018-07-23:
I did a new clean installation, with STRETCH.
all functions work, GPIO-Buttons, web app(volume up,down,play, stop ...), USB Sound
but the auto shutdown when idle for longer than x minutes, is without function
nothing happens after the set 10 min

zxa, 2018-07-23:
Hi @webjochen 
have you installed the jukebox software from ZIP or from git? If you used git, have you used the master branch?
What does the web app says If your box is idling for a couple of minutes? 
Have you enabled and started the idle-watchdog.service?

So many questions ;-)

webjochen, 2018-07-24:
I have it installed over the script "stretch-install-default-01.sh"
I installed it in the home directory and started with the following command:
`./stretch-install-default-01.sh`

webjochen, 2018-07-24:
`sudo apt-get install at`
That was the solution
now the time in the web app runs backwards
and the sleep timer works too

closed 2018-07-24 by webjochen