MiczFlor/RPi-Jukebox-RFID #95: Call rfid_trigger_play instead of mpc directly

Luegengladiator, 2018-07-24:
playerstop:
        $PATHDATA/resume_play.sh -c=savepos && mpc stop
    playerstopafter:
        echo "mpc stop" | at -q s now + $VALUE minute

both call a mpc stop to reduce the direct mpc calls and reduce redundancies playstopafter should look like: 
echo "$PATHDATA/playout_controls.sh -c=playerstop" | at -q s now + $VALUE minute

MiczFlor, 2018-07-26:
Did you change this in your pull request?

Luegengladiator, 2018-07-26:
No I didn't. 
I saw this during adding subdirectory support

zxa, 2018-07-28:
I haven't understood the playstopafter feature yet. What is a use case? Thatswhy i haven't supported it in the web app and made only a sloppy adaption for mpd.

Luegengladiator, 2018-08-14:
I think the "playstopafter" is needed for the timer feature. 
I didn't check all mpc calls. But I think it's useful to minimize the direct mpc-calls.
If there is a change on the control it's not easy to fix if there are same calls all over the place.

If I understand the example right, the timebased stop will not update the savepos. So your resume_play is the perfect example for this issue.

closed 2019-03-08 by MiczFlor