MiczFlor/RPi-Jukebox-RFID #97: while shutdown the jukebox - dropping url parameters failed

Raspfarbend, 2018-07-26:
Hi,
in Line 248 (/htdocs/inc.header.php) there is the excution of the shutdown command.
After this there is no connection anymore and the browser can't redirect to index.php without url parameters. Thats necessary because a browser reload by the user fires the shutdown command a second time.

**/htdocs/inc.header.php**
// shutdown the jukebox
if(isset($urlparams['shutdown']) && $urlparams['shutdown'] == "true") {
    $exec = "/usr/bin/sudo ".$conf['scripts_abs']."/playout_controls.sh -c=shutdown";
    if($debug == "true") { 
        print "Command: ".$exec; 
    } else { 
        **exec($exec);**
        /* redirect to drop all the url parameters */ 
        header("Location: ".$conf['url_abs']);
        exit; 
    }
}

MiczFlor, 2018-07-26:
Hi @Raspfarbend 
true. I never bothered. But also felt it‘s not elegant. Do you habe a better idea? Please pull request your innovation. Thanks

MiczFlor, 2018-08-02:
This should fix it: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/108
But now I am unsure if this is in master and the mpd branch?

PR #108 (2018-08-02): Execute shutdown and reboot in background
This gives enough time to drop the url parameters.
Closes #97
commit 5bd8eb93 (2018-08-02): Execute shutdown and reboot in background

This gives enough time to drop the url parameters.
Closes #97
closed 2018-08-02 by MiczFlor