MiczFlor/RPi-Jukebox-RFID #102: Long Starting Time vs. Buffering

BigSlikTobi, 2018-07-30:
Hi
thank you for this great tutorial! I really enjoyed building a Jukebox for my daughters second birthday!

I have an issue regarding the starting time of my box. After swiping the RFID card it takes a pretty long time until the box starts playing (about 20 sec.). This is a pretty hard time for a 2 year old girl :). I tried to manage this by changing the buffering time at the scrips by changing 
cvlc --no-video --network-caching=10000 -I rc --rc-host localhost:4212 "$VLCPLAYS" &
to a lower count but than the songs starts to stumble...
Do you have any idea how this can be solved.

I am using a USB - Stick to save the songs on.

Best regs
Tobi

MiczFlor, 2018-07-31:
Hi @BigSlikTobi 
nice to hear the box works and your daughter can handle it. The two seconds delay might trigger her to pull the plug and throw it against the wall, I beleive you. Cause and effect must be very close with kids :)
A few things come to mind
* what RPi are you using? The older ones are a bit slower. 
* I have no experience with USB, so not sure what that does. The buffering should not habe an effect, though, because it is strictly for networks. 
* A bit of superstition: at times it feels as if a bew RPi needs some time to settle in. Possibly there is some indexing going on in the background. That‘s not science but experience. 

Another thing you might want to try: use the new branch which used MPD instead of VLC. Read more here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/79
This will become standard by the end of August.

Caliandroid, 2018-08-14:
@BigSlikTobi: we use the 1st raspbi B Rev.2 with 700Mhz single-core and besides the boot time (40sec with current VLC version and 25sec with the upcoming mpd version) everything reacts quickly without lags (maybe 2sec until the music starts to play after rfid card has been swiped).

We are using an older Class 10 SD card (30MB/s) and no additional USB key, so i guess your lag comes from the usb key. 

If you haven't installed the raspbian itself on the usb key, you could try to remove it and put some songs on the internal sd card to see if it works faster like our phoniebox.

MiczFlor, 2018-08-16:
I am closing this ticket, because of the launch of version 1.0 on master. If you have any issues, please try first if they are still happening on the latest version 1.0. If so, open another ticket, please. 

* Use this thread for update issues [if you are running `stretch`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/138) 

**The version 1.0 has not been tested on `jessie`**

* Read this thread [if you are running `jessie`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/137) 

Version 1.0 is a major improvement (which will be full of bugs, for sure :) and I want to thank all the contributors and Phoniebox lovers with their input, pictures, bug fixes and suggestions.

closed 2018-08-16 by MiczFlor