MiczFlor/RPi-Jukebox-RFID #107: YouTube audio download

zxa, 2018-08-02:
I've added with #106 a possibility to download YouTube audio with youtube-dl (https://rg3.github.io/youtube-dl/). It's a replacement for the YouTube-streaming capability of VLC which mpd doesn't have.

To use that feature you have to install youtube-dl as described here: https://rg3.github.io/youtube-dl/download.html

So for example:

```
sudo curl -L https://yt-dl.org/downloads/latest/youtube-dl -o /usr/local/bin/youtube-dl
sudo chmod a+rx /usr/local/bin/youtube-dl
```
There's also a raspbian-package, but it's quite old.

Additionally, ffmpeg is needed to convert the audio to mp3:

`sudo apt install ffmpeg`

I integrated the download possibility to the "Register new card ID" page of the web app. Like stream audio, you may create a new folder. There, the audio is downloaded to and the last swiped RFID will linked to it. Additionally, if you specify an existing folder, the audio is put there without linking a RFID (because it's assumed that an existing folder has already a RFID linked to).

The downloading and conversion take a long time up to some minutes. As i have failed to get one of the dozens code snippets out there for "real time command line output in php" to work, youtube-dl is simply called in the background and you may check youtube-dl.log in the /shared folder if something went wrong.

MiczFlor, 2018-08-02:
Changes are merged to the MPD branch. I leave it open for now until this is part the documentation.

MiczFlor, 2018-08-16:
I am closing this ticket, because of the launch of version 1.0 on master. If you have any issues, please try first if they are still happening on the latest version 1.0. If so, open another ticket, please. 

* Use this thread for update issues [if you are running `stretch`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/138) 

**The version 1.0 has not been tested on `jessie`**

* Read this thread [if you are running `jessie`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/137) 

Version 1.0 is a major improvement (which will be full of bugs, for sure :) and I want to thank all the contributors and Phoniebox lovers with their input, pictures, bug fixes and suggestions.

closed 2018-08-16 by MiczFlor