MiczFlor/RPi-Jukebox-RFID #115: sleeptimer

harryliebteuch, 2018-08-04:
A sleeptimer function over fixed times as "parents RFID" or via the web interface would be great, in the forum there is already a script by tobi79

https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=337902#post337902

Raspfarbend, 2018-08-05:
@harryliebteuch,
I‘ve got the same idea.
But an Raspi don‘t have a Real-Time clock (RTC) onboard. Currently i‘am waiting for my RTC to be shipped and than i would code a sleeptimer function - Lullaby-Feature

zxa, 2018-08-05:
@Raspfarbend 
why don't you use network time?

Raspfarbend, 2018-08-05:
Hi @zxa ,
I think of offline-usage
Traveling by car etc.

Caliandroid, 2018-08-21:
I'm no Python coder but wouldn't it be possbile with threads and something like wait() or sleeep?

The accuracy is not perfect but should do its job as a sleeptimer.

At least this would be an easy task in Java (my base).

MiczFlor, 2018-08-21:
Not sure if this answers the question, but there is this inside `playout_controls.sh`:
~~~
shutdownafter)
        # remove shutdown times if existent
        for i in `sudo atq -q t | awk '{print $1}'`;do sudo atrm $i;done
        # -c=shutdownafter -v=0 is to remove the shutdown timer
        if [ $VALUE -gt 0 ];
        then
            # shutdown pi after $VALUE minutes
            echo "$PATHDATA/playout_controls.sh -c=shutdownsilent" | at -q t now + $VALUE minute
        fi
~~~
And it can be triggered with an RFID card in `rfid_trigger_play.sh` (adding this also to a GPIO button should be trivial):
~~~
$SHUTDOWNAFTER5)
            # shutdown after -v minutes
            $PATHDATA/playout_controls.sh -c=shutdownafter -v=5
            ;;
~~~
There are a few options, so you can add an RFID in the settings folder in `rfid_trigger_play.conf`(https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/settings/rfid_trigger_play.conf.sample)

It can also be set and managed through the web app - if you go to settings, there is auto shutdown and sleep timer.

winterrace, 2018-08-31:
The new shutdownafter feature in the playout_controls.sh can only be one part of the solution. It won't take long for the kids to notice that a restart of the box is sufficient to start again.

The script mentioned above was once made by me and also checks the remaining time quota even beyond restarts.
However, I don't recommend to use that script with the new (mpd-based) release, I have and plan to adopt this during the next weeks (I'll build a second box anyway). Also, pratice showed that there's still some kind of bug or side effect contained. After booting up, it sometimes does not grant quota, even if there was enough left or new quota would be due. I still have to locate the exact effect.

closed 2019-03-22 by MiczFlor