MiczFlor/RPi-Jukebox-RFID #118: Dissussion: Code Freeze for pr80-mpd ?

Raspfarbend, 2018-08-06:
Hi @MiczFlor ,
how about plans for next stable-release ?

MiczFlor, 2018-08-06:
Hi @Raspfarbend 

thanks for opening this discussion. I was on holidays for the past two weeks and merged / commented from my mobile. So my thought was: "I'll sneak a few things in, clean the UI and then release"
The move to MPD is a major change and deserves a major version number: 1.0 - I think. And thanks to all the contributors in the past two weeks, especially @zxa and @Luegengladiator and all the others.

Given your discussion thread, I need to add:
* I never used github to manage a team. If there are good ways to assign *issues* to a roadmap, please let me know
* Besides the code freeze, i wonder about alpha testing. Who is in on this job?
* I want to open a new branch with the latest VLC version for jessie and stretch and let it sit 
* the new release I want to limit testing to stretch

Things I have in mind, but am not sure if they are *must haves* for 1.0:
* I started an **install script** here: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/pr80-mpd-as-audio-player/scripts/installscripts/stretch-install-default-02.sh but didn't have a lot of time working on it. Inspired by https://github.com/tizonia/tizonia-openmax-il#debian--ubuntu--raspbian -  a one liner to copy paste into the command line
* Move the gallery and install scripts to the **wiki on github**
* Include **Spotify** (possibly via including tizonia?)
* Make the **dir for audiofiles a variable** so it will be easier to move your audio files to other dirs or a USB stick
* Build a **web UI for file management** (upload, delete, move) to allow management through the browser for the ssh illiterate (those who might use the Phoniebox they got given as a gift - but never built it themselves - there are quite a few out there)
* The **seek forward and backward** as suggested by @harryliebteuch https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/113
* @Luegengladiator wanted to work on a **barcode reader**, which I think would be neat. Use case: my father wants one so that my mother can play their classic CDs from the Phoniebox directly by scanning the code from the CD
* @andimanuel did a **record** tweak, which I think is a nice thing to have - and a low hanging fruit: https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/91

So, what else should go in, who will work on what and how do we manage the team effort?

zxa, 2018-08-06:
My 2 cents on some points:

- **Install script:** That's great, we need also an **update script** which could be called by the web app, so not so tech savvy people may keep there boxes up to date. This also touches the topic of alpha testing as with an easy update possibility we could do[ bananaware releases ](https://en.wikipedia.org/wiki/Perpetual_beta) without the burden to be "perfect". I think that's the most important task, as all the other features may come via update later.
- **Branches:** I think beside the retired vlc branch we should have a "release" and a "dev" branch (like the pr80, where we can play).
- **Spotify:** This is a complicated topic. When i tested audio players, i also considered mobidy as it works like mpd and has spotify support. Sounds perfect, but: the project seems to be near death (last release 18 month ago) which is even worse as the spotify support (same for tizonia) relies on the libspotify which was written by spotify, but they ended the support for it some time ago. So more and more features of libspotify [stop to work from one day to the other](https://github.com/tizonia/tizonia-openmax-il/issues/490). I don't know why these projects don't use clients for spotify connect [like this one](https://github.com/Spotifyd/spotifyd) and control it via the spotify web api, but i guess there is a good reason for it (like "doesn't work, you naif!"). What is the impact on our project: if we add spotify support, there may be issues in the not so far future and we're dependent on how fast the problem is fixed on the spotify client we use.
- **Resume play:** I don't understand the issue with that.

Another thing to be simplified for the non-tech box owners could be the special cards management.

Luegengladiator, 2018-08-07:
**Branches:** I like the Idea with a dedicated development branch. We are using a Consolidation- and Testbranches at work too. Maybe it's not useful here but we should think about?
**Projects:** We should think about Projects to share work. 
**Hotspot:** I like the Idea. So it's easy do start up and configure later.
**Variables:** There should be as much as possible in configurable variables. 
**Phonie as a gift:** The more we support it, the more it will be used.
**Barcode reader:** Tested with my Datalogic it's exactly working like the rfid-reader. All you need is already there. 
**Raspbian:** Stretch is the actual release. But Buster release will be first half 2019 so let's take care to support different Operating Systems. Btw. Phonie is working on nearly all Linux Distributions without bigger problems. 

I'm concerning about the Phonie. There is a commercial Product doing nearly the same: https://tonies.com/
Is there a Problem with patents?

MiczFlor, 2018-08-16:
I am closing this ticket, because of the launch of version 1.0 on master. If you have any issues, please try first if they are still happening on the latest version 1.0. If so, open another ticket, please. 

* Use this thread for update issues [if you are running `stretch`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/138) 

**The version 1.0 has not been tested on `jessie`**

* Read this thread [if you are running `jessie`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/137) 

Version 1.0 is a major improvement (which will be full of bugs, for sure :) and I want to thank all the contributors and Phoniebox lovers with their input, pictures, bug fixes and suggestions.

closed 2018-08-16 by MiczFlor