MiczFlor/RPi-Jukebox-RFID #122: New install script: testing and input needed!!!

MiczFlor, 2018-08-09:
While we are working on the new branch `pr80-mpd-as-audio-player` a new installation routine is in the making. You can see the script here: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/pr80-mpd-as-audio-player/scripts/installscripts/stretch-install-default-02.sh

The new install will also look for an existing install and ask what elements to re-use (e.g. GPIO scripts, audio folders, etc.)

The one line install should look something like this:
~~~
wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/pr80-mpd-as-audio-player/scripts/installscripts/stretch-install-default-02.sh; chmod +x stretch-install-default-02.sh; sudo ./stretch-install-default-02.sh
~~~
Needless to say: there are bound to be problems once this branch goes live. 

So I would appreciate feedback and possible tests (having said that, you would need to edit the current version to work with the branch `pr80-mpd-as-audio-player` and not `master`.

I want this to work at least for "out of the box" installs before we move the entire master to the new `MPD` driven playout - which has a lot of neat little improvements (like 'resume play' for audiobooks which can be set individually) besides massive battery saving.

The script is kind of half finished. But the sooner the testing starts the better. One of the features that will be included once the install works: "fallback to hotspot", meaning: if you are on the road, you can connect to the Phoniebox directly, it will start a WiFi hotspot. When at home, it will connect to your WiFi as usual.

So please, knock yourself out to test this.

gusmax777, 2018-08-10:
Hi, 
I have installed the script but now the audio playback does not work anymore. The web-gui show 0% volume and I am unable to change this value. On the command line I am able to play sound with vlc but with mpd it is not working. Any idea?

Cheers, Thomas

MiczFlor, 2018-08-10:
Hi @gusmax777 
thanks for jumping into the deep end of the pool :)
Firstly, you should see if you can jump back to your working VLC install by switching back to the main branch:
~~~
git checkout master
~~~
When you say `mpd` is not working, does that mean it doesn't work on your system or that it doesn't work on Phoniebox?
When you went through the install script, did you 
* set PCM as the audio iFace?
* stick to the default audio file install?
What do you get when you type
~~~
sudo cat /etc/mpd.conf | grep mixer_control
sudo cat /etc/mpd.conf | grep music_directory
~~~

gusmax777, 2018-08-10:
Hi, 

And about mpc... The phoniebox itself works. But I cannot start the playback of any file. 
So it is kind of both. Mpc does not work on Phoniebox and the system via cmd. Maybe I will try a fresh install in the evening. 

Cheers, Thomas

MiczFlor, 2018-08-10:
Can you manually replace 
~~~
%DIRaudioFolders%
~~~
with 
~~~
/home/pi/RPi-Jukebox-RFID/shared/audiofolders
~~~
inside `/etc/mpd.conf` (or whatever folder you are using for your folders containing audio files)?

Then do a reboot...

gusmax777, 2018-08-10:
Thanks Micz, 

That made it working again. Will do some more tests in the evening. 

Cheers, Thomas

Caliandroid, 2018-08-13:
Had the latest vlc installed, RaspberryPi 1 B Rev2.
The installation script was quite easy to follow.

Here is my feedback:

1. Received the question to overwrite gpio-buttons.py -> Answered with "yes"
[image] 

GPIO buttons are working correctly

2. Playback was not startin g in the web app, so i checked the audio folder inside mpd.conf = correct
So i applied the rights like described in your post:
```
sudo chown pi:www-data /home/pi/RPi-Jukebox-RFID/settings
sudo chmod 775 /home/pi/RPi-Jukebox-RFID/settings
```
This solved the problem

3.  My Pi is booting much faster now (25sec from first rfid initilization beep till startup sound, 40sec with the VLC version). 

4. Persistent problem:

If a music folder (started by rfid) is playing and the rfid-reader scans another card, the new folder is loaded but stays in PAUSE mode. To resume playing i can press my gpio PAUSE-button or scan this card a second time. 

If i pause the playback and scan another card, the new music folder starts playing on the fly without problems, so i guess with every new card a pause/play (=identical) command gets executed.

Thanks a lot!

Caliandroid, 2018-08-13:
Had the latest vlc installed, RaspberryPi 1 B Rev2.
The installation script was quite easy to follow.

Here is my feedback:

1. Received the question to overwrite gpio-buttons.py -> Answered with "yes"
[image] 

GPIO buttons are working correctly

2. Playback was not startin g in the web app, so i checked the audio folder inside mpd.conf = correct
So i applied the rights like described in your post:
```
sudo chown pi:www-data /home/pi/RPi-Jukebox-RFID/settings
sudo chmod 775 /home/pi/RPi-Jukebox-RFID/settings
```
This solved the problem

3.  My Pi is booting much faster now (25sec from first rfid initilization beep till startup sound, 40sec with the VLC version). 

4. Persistent problem:

If a music folder (started by rfid) is playing and the rfid-reader scans another card, the new folder is loaded but stays in PAUSE mode. To resume playing i can press my gpio PAUSE-button or scan this card a second time. 

If i pause the playback and scan another card, the new music folder starts playing on the fly without problems, so i guess with every new card scanned during playback a pause/play command gets executed instead of loading the music from the rfid-linked folder

Thanks a lot!

MiczFlor, 2018-08-14:
Hi @Caliandroid 
thanks for the thorough testing. Based on your findings, I just pushed another fix
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/39464e90e2ee07a8f91c1a7d2e5cd96ab85a1a7b
If you go into the Phoniebox dir and type `git pull` you should get the latest changes. Please let me know if it works now.
The behaviour now should be: 
* if the same is pressed in the web app or swiped, it should stop if playing or paused
* and start if stopped. No matter where you start the playing process. 
* Only issue: the web app does not reload when using the RFID card, so the player controls will remain the same.

And thanks again for testing!!!!

Caliandroid, 2018-08-14:
You're welcome and i have to thank you for all the effort. 

Results:  

If folder A is playing and i swipe card B, folder B starts directly to play but if i swipe again card A, folder A starts from the very beginning and does not resume from the last position.

Resume play doesn't work too if i press the GPIO pause button before swiping card B

MiczFlor, 2018-08-14:
@Caliandroid 
I have to change the GPIO script. I totally forgot aboit that :)
Which means you will possibly need to alter your code, too. 

But swipe and web app should work identically. Can you try some more? There was a change in the format of the file tracking position. Possibly the resume only works the second time round.

Caliandroid, 2018-08-14:
> Possibly the resume only works the second time round.

No i'm sorry, but the behavior does not get better.

1.  Swipe card A, playback of folder A starts.
Swipe card B, playback of folder A stops but nothing else (other behaviour like before after the latest git pull..)

2. Swipe card A, playback of folder A starts.
Press pause gpio button, playback stops,
         1. If i swipe card A again, playback does not continue
         2. If i swipe card B, playback of folder A continues on the current position of the played track. And if i swipe card B again, playback stops and if i swipe card B a third time, folder B starts playing from the very beginning.

Caliandroid, 2018-08-14:
Another problem found (without modifications)

I've received a lastplayed.dat file in folder B that contains the data from folder A (folder B was playing and i swiped card A).

and something else:

` 1.1 IF new folder given ("$Latest_Folder_Played" != "$FOLDER")
`
returns sometimes false while switching between folder A and B so there must be a constellation where $latest_folder_played or $folder are not updated correctly.

MiczFlor, 2018-08-15:
Hi @Caliandroid 
please, you are doing such a great job testing, help us once more. There is a newer version out not. `git pull` should get you the code. Keep up the good work!

Caliandroid, 2018-08-15:
Hi @MiczFlor : 2hours ago i was able to pull the changes but it became even worse and some cards did not play anymore and no position was saved. I was not able to check the logs and did everything over my smartphone.

I've seen that you changed file access rights + folder write procedure for lastplayed.dat in your latest commit 15 min ago, so i updated the repository once again and my first RFID tests worked just fine after i de- and reactivated the resume play to delete the existing lastplayed.dat files. 

- Pure RFID usage: OK
- Pure web-app usage: OK
- Mixed usage: OK
- Reboot while playing a track: OK  (position saved)

I'll keep this branch and get back at you in case of problems.

**Another minor bug that i've found:** (i don't know if it exists only in the pr80 branch):

File order of tracks inside a folder is sometimes A-Z then a-z:
Example: 
1. Carmen
2. Dora
3. Zora
4. calvin
5. dina

and sometimes it's Aa-Zz
Example:
1. Carmen
2. calvin
3. Dora
4. Dina
5. Zora

As a result, the track order is not always identical while using the "NEXT" command.
I don't know yet what triggers the different order logic.

I'll try to send you some screenshots

MiczFlor, 2018-08-15:
Hi @Caliandroid 
I made another update. Please pull again. I also teleased version 1.0 on master. 
It worked for me and the pr80 branch got too big. You should be able to switch to master branch an pull. Your configs shpuld be save.

MiczFlor, 2018-08-15:
And please start another issue regarding the ordering. Thanks!

Caliandroid, 2018-08-15:
Done - Thank you

MiczFlor, 2018-08-16:
I am closing this ticket, because of the launch of version 1.0 on master. If you have any issues, please try first if they are still happening on the latest version 1.0. If so, open another ticket, please. 

* Use this thread for update issues [if you are running `stretch`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/138) 

**The version 1.0 has not been tested on `jessie`**

* Read this thread [if you are running `jessie`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/137) 

Version 1.0 is a major improvement (which will be full of bugs, for sure :) and I want to thank all the contributors and Phoniebox lovers with their input, pictures, bug fixes and suggestions.

closed 2018-08-16 by MiczFlor