MiczFlor/RPi-Jukebox-RFID #126: Handling of playlists

gusmax777, 2018-08-10:
Hi, 
First of all: Great project!!! I am working on a xmas present to my 4 year old with it :-)

The whole setup works really well, the only thing I am missing is a way to play playlists within the interface. On the command line with VLC they are working. Why do I need this: 

 I got a directory with many subdirectorys. Each sub hold a few MP3 files that belong to the same topic and are just another part. I have produced a playlist that holds all the single files. But I am unable to get in running in the web-interface? 

I have to explain that I have not received the RFID reader and the cards. So web-gui and cmd are the only way to get in working at the moment. 

Cheers, Thomas

MiczFlor, 2018-08-10:
Hi @gusmax777 
welcome on board :)
We are currently migrating the entire code to work with MPD instead of VLC. Have a look here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/79
And there is an install script for the current state (which already working well):
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/122
Do you think you could give this a try so we can add your wishes to the newer code base?

gusmax777, 2018-08-10:
Sure, 

Thanks for the hint. Do I need a fresh install, or can I let it run "over" it?

Cheers, Thomas

gusmax777, 2018-08-10:
Hi Micz, 

I do still have no luck with the playlist. My plan was to have one playlist with all MP3 Files to shuffle thru it. I could already produce the file via: 

find /home/pi/RPi-Jukebox-RFID/shared/audiofolders/* -iname *.mp3 -type f | shuf | head -n 200 > /home/pi/RPi-Jukebox-RFID/shared/audiofolders/playlist/mp3_shuffle.m3u

But if I place the file into a singe directory it does not play. Even the file holds complete path information like: 

/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Conni 13 - und der Liebesbrief/._Das geheimnisvolle Liebestreffen.mp3
/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Conni 09 - Im Krankenhaus und lernt Tanzen/Der Unfall.mp3
/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Conni 21 - schläft im Kindergarten-Conni geht in den Zoo/10  Conni - Zebras und Elefanten.mp3

Do you have an idea to get this going?

Cheers, Thomas

MiczFlor, 2018-08-10:
Hi @gusmax777 

first question: does the Phoniebox play files which you have placed in folders which are inside the audiofolders directory `/home/pi/RPi-Jukebox-RFID/shared/audiofolders/`
?

If so...
does this line
~~~
sudo cat /etc/mpd.conf | grep path
~~~
produce (amongst other outputs) this line:
~~~
#save_absolute_paths_in_playlists	"no"
~~~

If so... you need relative paths in your playlist. Not
~~~
/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Conni 09 - Im Krankenhaus und lernt Tanzen/Der Unfall.mp3
/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Conni 21 - schläft im Kindergarten-Conni geht in den Zoo/10 Conni - Zebras und Elefanten.mp3
~~~
But
~~~
Conni 09 - Im Krankenhaus und lernt Tanzen/Der Unfall.mp3
Conni 21 - schläft im Kindergarten-Conni geht in den Zoo/10 Conni - Zebras und Elefanten.mp3
~~~
Try making one manually, like the one here, and save it in the dir you have in the conf (should be:)
~~~
playlist_directory		"/tmp"
~~~
Then try playing it:
~~~
mpc clear
mpc update
mpc load <playlist_name>
mpc play
~~~
Does that work?

zxa, 2018-08-10:
What also works is full path with file:// in front of it (yes, three slashes):
```
file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/Conni 09 - Im Krankenhaus und lernt Tanzen/Der Unfall.mp3
file:///home/pi/RPi-Jukebox-RFID/shared/audiofolders/Conni 21 - schläft im Kindergarten-Conni geht in den Zoo/10 Conni - Zebras und Elefanten.mp3
```

MiczFlor, 2018-08-14:
Hi @gusmax777 
try to update the code with `git pull`. I made changes and it might fix the issue:
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/39464e90e2ee07a8f91c1a7d2e5cd96ab85a1a7b

gusmax777, 2018-08-14:
Hi @MiczFlor 

Maybe I am to stupid. What is the exact command? I only get : 

fatal: Kein Git-Repository (oder irgendein Elternverzeichnis): .git

What is the complete command?

Cheers, Thomas

MiczFlor, 2018-08-15:
Hi @gusmax777 
there is another version, since I last pinged you. Please try by typing:
~~~
cd /home/pi/RPi-Jukebox-RFID/
git branch
~~~
This should return something like this:
~~~
  master
* pr80-mpd-as-audio-player
~~~
Note: the * must be in front of pr80...
Then do a git pull:
~~~
git pull
~~~
Which should upate the code base to the latest version.

MiczFlor, 2018-08-16:
I am closing this ticket, because of the launch of version 1.0 on master. If you have any issues, please try first if they are still happening on the latest version 1.0. If so, open another ticket, please. 

* Use this thread for update issues [if you are running `stretch`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/138) 

**The version 1.0 has not been tested on `jessie`**

* Read this thread [if you are running `jessie`](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/137) 

Version 1.0 is a major improvement (which will be full of bugs, for sure :) and I want to thank all the contributors and Phoniebox lovers with their input, pictures, bug fixes and suggestions.

closed 2018-08-16 by MiczFlor