MiczFlor/RPi-Jukebox-RFID #128: USB-Soundkarte nachträglich einrichten

Shrike0815, 2018-08-12:
Hallo,

ich habe meine Phoniebox komplett nach Anleitung aufgebaut, installiert und konfiguriert, bis auf die USB-Soundkarte. Diese wurde erst 1 Woche später geliefert, daher hatte ich alles ohne sie eingerichtet.

Nun bekomme ich die Soundkarte nicht ans laufen. Alle Funktionen der Box laufen einwandfrei, nur kommt kein Sound raus. Web-Interface läuft einwandfrei, RFID auch usw..
Boxen habe ich an einem anderen Gerät getestet, diese laufen auch.
Wenn eine Karte auf den Leser gelegt wird, wird auch das passende Album gestartet. Zumindestens kann ich das in dem Web-Interface sehen dass er es abspielt.

Gekauft habe ich komplett nach der Vorgabe der Anleitung.

Muss man die USB-Soundkarte einrichten oder einen Treiber installieren?

Vielen Dank schon mal für die Hilfe!!

Andco7, 2018-08-12:
Hi,

hast du das neue Installscript verwendet oder das alte? (MPD oder VLC).
Micz hat schon an anderer Stelle für das neue Installscript Hilfe angeboten:
[https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/122#issuecomment-412079005]

webjochen, 2018-08-12:
Ich meinte ich musste noch folgendes bei mir ändern obwohl ich Stretch verwende

USB audio card on Jessie

> Check if the device is recognised:
> 
> $ cat /proc/asound/modules 
>  0 snd_bcm2835
>  1 snd_usb_audio
> This shows that device 1 is the usb audio card.
> 
> Open the sound configuration file.
> 
> $ sudo nano /usr/share/alsa/alsa.conf
> Now replace the lines near the end of the file:
> 
> defaults.ctl.card 0
> defaults.pcm.card 0
> with:
> 
> defaults.ctl.card 1
> defaults.pcm.card 1
> Now the audio card for the system is set to 1. Reboot the RPi:
> 
> $ sudo reboot

webjochen, 2018-08-12:
das hier hast du überprüft ?

> settings/Audio_iFace_Name
> 
> This is a file containing a string, by default PCM.
> 
> Inside settings/Audio_iFace_Name is the iFace name of the sound card. By default for the RPi this would be PCM. But this does not work for every setup. If you are using phatbeat as a DAC for example, you need to change the content of Audio_iFace_Name from PCM to Master or Speaker. Other external sound cards might use different interface names. To see if PCM could work for you, type amixer sget PCM. To list all available iFace names, type amixer scontrols.

Andco7, 2018-08-12:
@Shrike0815  
die Veränderungen inder alsa.conf waren genau das, was @webjochen gerade gepostet hat.

Shrike0815, 2018-08-13:
Es waren beide Lösungen nötig.

Die settings in Audio_iFace_Name hatte ich vorher schon einmal ausprobiert, das hatte aber nicht geholfen.

Als ich nun die vorgeschlagenen Änderungen in der alsa.conf vorgenommen habe hat es auch noch nicht funktioniert.
Aber nach der Änderung in alsa.conf habe ich nochmal mittels "Mixer stet PCM" die Funktion geprüft und dort kam "amixer: Unable to find simple control 'PCM',0".
Also habe ich mittels "amixer scontrols" geschaut was da ist und nun erschien "Simple mixer control 'Speaker',0"
Also in Audio_iFace_Name "PCM" auf "Speaker" geändert, wieder rebooted und siehe da, es kam Musik.

Ich hoffe ich habe die nötigen Schritte verständlich zusammengefasst.

Vielen Dank an euch alle!! 
Ich werde Fotos meiner Umsetzung posten sobald ich damit aus der Papp-Box ausgezogen bin.

closed 2018-08-13 by Shrike0815

MiczFlor, 2018-08-13:
Danke @Shrike0815 @Andco7 @webjochen für die Infos. Ich nehme das dann in die Troubleshooting Sektion auf.

matthesdr, 2018-12-09:
Guten Abend zusammen,

ich habe meine Phoniebox per One-Line-Installation installiert und bei dem Setup für die Audio-Settings erst mal die Defaults gelassen. Im Nachgang habe ich dann meine USB-Soundkarte angeschlossen und Stolper jetzt über die Einrichtung. Ich habe die oben beschriebene Vorgehensweise befolgt und bekomme Audio nichts ans laufen. Im Web-Interface kann ich auch die Lautstärke nicht einstellen. Der eingestellte Wert ist "-1". Hat vielleicht noch jemand eine Idee, was ich noch tun kann?

Folgende Informationen kann ich liefern:

~ $ cat /proc/asound/modules
 0 snd_bcm2835
 1 snd_usb_audio

$ amixer scontrols
Simple mixer control 'Speaker',0
Simple mixer control 'Mic',0
Simple mixer control 'Auto Gain Control',0

**Inhalt der asound.conf**
```
pcm.!default {
    type hw
    card 1
}
ctl.!default {
    type hw
    card 1
}
```

**Inhalt der alsa.conf:**
defaults.ctl.card 1
defaults.pcm.card 1
defaults.pcm.device 0

LG

derKai, 2019-06-20:
Hallo,

falls nochmal jemand an dieser Stelle nach einer Lösung sucht. Der entsprechende Rückgabewert von 'amixer scontrols' (in deinem Fall sehr wahrscheinlich 'Speaker') muss unter 

/home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name

eingetragen werden.

Viele Grüße